adafruit/circuitpython #5405: `_bleio` does not accept `data` as a kwarg in `start_advertising`

SomTambe, 2021-09-29:
### CircuitPython version

```python
Adafruit CircuitPython v 7.0.0
```

### Code/REPL

```python
import _bleio

adp = _bleio.adapter
msg = bytes('abcde', 'utf-8')
adp.start_advertising(data=msg, # msg is of bytes type
                      connectable=True, 
                      anonymous=True, 
                      timeout=30, 
                      interval=0.1, 
                      tx_power=4)
```

### Behavior

Traceback (most recent call last):
  File "\<stdin\>", line 5, in \<module\>
TypeError: function missing 1 required positional arguments

### Description

It works when I pass `msg` as a normal argument:
```python
adp.start_advertising(msg,
                       connectable=True,
                       anonymous=True,
                       timeout=30,
                       interval=0.1,
                       tx_power=4)
Traceback (most recent call last):
  File "<stdin>", line 5, in <module>
_bleio.BluetoothError: Unknown system firmware error: 0009 
```
Should have worked in the previous case also, especially when the [docs](https://circuitpython.readthedocs.io/en/7.0.x/shared-bindings/_bleio/index.html#bleio.Adapter.start_advertising) show it as a kwarg.

PS: If you could also help me resolve the `_bleio.BluetoothError: Unknown system firmware error: 0009`, that would be great! I tried sending a blank string payload, it worked, but it is not working if I send any other string. What should I do?

### Additional information

Board: Adafruit nRF52840 Bluefruit Feather Sense

dhalbert, 2021-09-29:
> PS: If you could also help me resolve the `_bleio.BluetoothError: Unknown system firmware error: 0009`, that would be great! I tried sending a blank string payload, it worked, but it is not working if I send any other string. What should I do?

The data you're trying to send is an ill-formed advertisement. An advertisement contains chunks of data in a specific format, with type and length bytes, etc. If you use the https://github.com/adafruit/Adafruit_CircuitPython_BLE library, it will help you build up these advertisements.

The keyword arg thing is another problem, which we'll look at in more detail. We may open a more general issue for that.

dhalbert, 2021-09-29:
Closing in favor of #5407.

closed 2021-09-29 by dhalbert