adafruit/circuitpython #5406: Danish Keyboard

Kildeager48, 2021-09-29:
Hi
I am really loving my Raspberry pi pico, but currently i need to switch to the US keyboard(or other languages, US is just my default when i do this), so when i want to show scripts to my friends they need to change the keyboard language on their pc, which just really ruins the "magic" of it.
I would be more than happy to make the new setting, but i have never tried to do it before, so i dont really know how to, never the less i will try.
-Søren

Neradoc, 2021-09-29:
I believe what you want is to look here for international layouts:
https://github.com/Neradoc/Circuitpython_Keyboard_Layouts

Kildeager48, 2021-09-29:
Thx so much for the help, i wouldn’t have been able to find this if it weren’t for you, again thx

https://github.com/Neradoc/Circuitpython_Keyboard_Layouts

closed 2021-09-29 by dhalbert