adafruit/circuitpython #5408: Support more modules on esp32c3

microDev1, 2021-09-29:
A list of modules that need to be modified to work on `esp32c3`:

- [ ] ALARM
- [ ] ANALOGIO
- [ ] AUDIOBUSIO
- [ ] COUNTIO
- [ ] FREQUENCYIO
- [ ] IMAGECAPTURE
- [ ] PARALLELDISPLAY
- [ ] RGBMATRIX
- [ ] ROTARYIO

rxhfcy, 2023-04-26:
Consider adding the esp32-c3 label?

btw ALARM sure would be nice (for deep sleep support), but I do understand features don't just magically appear out of thin air.

tannewt, 2024-03-20:
`countio`, `frequencyio` and `rotaryio` aren't easily doable because the C3 doesn't have a PCNT peripheral. It doesn't have I80 support for `paralleldisplaybus` either.

tannewt, 2024-03-22:
This enables audiobusio: https://github.com/adafruit/circuitpython/pull/9071/files#diff-48d06efbad2c96ae11fad1ca2f81801d234625a07d4a74b46a61eccdd6ad4c2dL59

arduino12, 2025-01-31:
**Hi,**

I found that only the 2MB ESP32-C3 boards have the [ALARM](https://docs.circuitpython.org/en/stable/shared-bindings/alarm/) module built-in:
[TTGO T-01C3 ESP32-C3](https://circuitpython.org/board/lilygo_ttgo_t-01c3/) and [ESP-C3-32S](https://circuitpython.org/board/ai_thinker_esp32-c3s-2m/) (2M).

I found `/ports/espressif/mpconfigport.mk` [sets](https://github.com/adafruit/circuitpython/blob/main/ports/espressif/mpconfigport.mk#L281-L285) `CIRCUITPY_ALARM = 0` for ESP32C3 [because its](https://github.com/adafruit/circuitpython/blob/main/ports/espressif/mpconfigport.mk#L4-L5) `IDF_TARGET_ARCH = riscv`:
```make
ifeq ($(IDF_TARGET),esp32c3)
  IDF_TARGET_ARCH = riscv
...
endif
...
ifeq ($(CIRCUITPY_LEGACY_4MB_FLASH_LAYOUT), 1)
ifeq ($(IDF_TARGET_ARCH), xtensa)
  CIRCUITPY_ALARM ?= 1
else
  CIRCUITPY_ALARM = 0
endif
...
endif
```
But it does that only for the 4MB boards for some reason, the 2MB boards do gets the ALARM so its not a `riscv` limitation...

1. Does anyone know why it sets `CIRCUITPY_ALARM = 0`?
2. Does anyone know how to add the ALARM module to all the ESP32C3 boards?
(I need it on [Maker Go ESP32C3 Supermini](https://circuitpython.org/board/makergo_esp32c3_supermini/))

As previously said by @rxhfcy and @davidvfx07,
It would be great to have low-power support for the ESP32-C3 -
using Arduino IDE I got down to 7uA with 2 GPIO wakeup sources (one high one low) and one GPIO state preserved!
I would like to achieve the same using circuitpython!

@dhalbert Any help will be appreciated!

dhalbert, 2025-01-31:
@arduino12 This may be due to not enough flash space right now when other things are included (2MB builds may trim further, making room for `alarm`). In CircuitPython 10 we will get rid of the OTA partition and there should be room.

arduino12, 2025-01-31:
**Hi,**

@dhalbert Thanks for your quick reply!
I changed it to `CIRCUITPY_ALARM = 1` and compiled for the [Maker Go ESP32C3 Supermini](https://circuitpython.org/board/makergo_esp32c3_supermini/) (my first time compiling circuitpython!)

And it works perfectly - got down to 7uA!

The original `adafruit-circuitpython-makergo_esp32c3_supermini-en_US-9.2.4.bin` is `1449kB`.
With the added ALARM it became `1464kB` **so I think it worth it!**

Attached here in case anyone wants to try (remove the .zip extension.. GitHub won't allow .bin so I added it).
[adafruit-circuitpython-makergo_esp32c3_supermini-en_US-9.2.4_with_alarm.bin.zip](https://github.com/user-attachments/files/18624720/adafruit-circuitpython-makergo_esp32c3_supermini-en_US-9.2.4_with_alarm.bin.zip)

**Can't wait for CP v10** 🙏!

tannewt, 2025-05-12:
This can be closed after #10229 is done.