adafruit/circuitpython #5411: time.monotonic() returns no fraction of seconds after *long* runtime.

ThomasAtBBTF, 2021-09-30:
### CircuitPython version

```python
Adafruit CircuitPython 5.4.0  LoC BeR M4 base board with samd51g19
```

### Code/REPL

```python
import time
time.monotonic()
# in code and the command line, the fraction of seconds was always .0
```

### Behavior

I did observe this behavior last night on a device that was running for many days.
Note: that also after breaking the code with ctrl-c and importing time
and entering time.monotonic() in the command line, the fraction of seconds was always .0
Only after microcontroller.reset() this behavior stopped.

I will try to reproduce this with other controllers and CP 7.0.
But it will take days until I have a result ....

dhalbert, 2021-09-30:
This is a known problem with `time.monotonic()`, because of the low precision of floats (5-6 digits). We recommend you use `time.monotonic_ns()`, if you board supports it (must have longints). Or, if it works for you, you can use `supervisor.ticks_ms()`, which wrap but keep their precision.

See https://github.com/adafruit/circuitpython/issues/342#issuecomment-337228427 for some background on the precision issue.

ThomasAtBBTF, 2021-09-30:
Ok, thank you taking a look at this so quickly and for the explanation.
It would be helpful if is this information would be available at:
https://circuitpython.readthedocs.io/en/latest/shared-bindings/time/

Also a possibility to programmatically reset to time.monotonic "0" would be helpful, as this situation can be detected quite easily, now that I know how,

closed 2021-09-30 by ThomasAtBBTF