adafruit/circuitpython #5414: Implement certain functions in supervisor as properties

microDev1, 2021-09-30:
Certain functions (e.g. `enable_autoreload()`, `disable_autoreload()` etc.) could benefit from `set` and `get` methods when implemented as a property, this would reduce the number of individual functions.

dhalbert, 2021-09-30:
Modules (as opposed to class instances) can't in general have properties, so we'd need a singleton of some kind to hold the properties. For instance, `microprocessor.cpu` is a singleton instance of `Processor`, and it holds properties like `temperature`.

MicroDev1, 2022-09-30:
Thoughts on moving the following to `supervisor.runtime`:

- [ ] set_next_stack_limit
- [ ] set_rgb_status_brightness
- [ ] disable_ble_workflow
- [x] autoreload

dhalbert, 2022-09-30:
We have very little room on the smallest builds for these things. But ble workflow is not present on SAMD21 builds, so that could be done without worry. Having a "get" on the stack limit is useful to make some calculation about increasing it.

We will want to leave the old functions in place, and remove them in 9.0.0.

Neradoc, 2022-09-30:
> We will want to leave the old functions in place, and remove them in 9.0.0.

Note that it's not what was done for `enable_auto_reload` and `disable_autoreload`.
Was there space saved from that ?

dhalbert, 2022-09-30:
> Note that it's not what was done for `enable_auto_reload` and `disable_autoreload`.

I didn't realize that. Maybe it was because it wouldn't fit otherwise.

It's true these are not used all that much, so we don't necessarily need to provide a transition.

closed 2022-10-04 by dhalbert