adafruit/circuitpython #5415: MP3 audio does not loop on 7.0.0

gamblor21, 2021-09-30:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0 on 2021-09-20; Adafruit Hallowing M4 Express with samd51j19
```

### Code/REPL

```python
import board
import digitalio
from audiomp3 import MP3Decoder
from audioio import AudioOut

speaker_enable = digitalio.DigitalInOut(board.SPEAKER_ENABLE)
speaker_enable.direction = digitalio.Direction.OUTPUT
speaker_enable.value = True

audio = AudioOut(board.SPEAKER)
mp3 = open("laugh.mp3", "rb")
decoder = MP3Decoder(mp3)

audio.play(decoder, loop=True)
```

### Behavior

Audio starts playing, plays properly once and does not loop. Only tested with the single MP3 so far. MP3 used can be found at https://github.com/gamblor21/Laughing_Skull/blob/main/laugh.mp3

jepler, 2021-10-05:
huh probably something similar to 
 * https://github.com/adafruit/circuitpython/issues/5164

jepler, 2022-02-14:
don't you need a `while True: pass` to prevent your example program from exiting right after it starts playback?

gamblor21, 2022-02-15:
I believe I was doing it in the REPL there. I still have the project sitting on a table somewhere so I will try to pull it out tomorrow and update it and see what it is doing.

gamblor21, 2022-02-15:
Just tested with 7.2.0-alpha.2 and it is still occurring both in the REPL and in my actual code.

The project I was using this on is https://github.com/gamblor21/Laughing_Skull (`code-distance.py` is the version being ran).

jepler, 2022-04-04:
Finally got around to reproducing this. On Trellis M4, the MP3 audio stops but the `playing` property of the speaker stays `True`.

jepler, 2022-05-04:
We believe this is fixed. In fact, we believe so strongly that it is fixed, that we fixed it a second time.

closed 2022-05-04 by jepler