adafruit/circuitpython #5420: Add watchdog for sam d5x/e5x

jepler, 2021-10-01:
SAM D5x/E5x family datasheet says that there's a watchdog timer peripheral available. Add support for this to the existing CircuitPython `watchdog` module.

jepler, 2021-10-01:
samd21 also has a watchdog timer, but it's less likely that there's room for an implementation on those older/smaller chips.

maholli, 2021-10-10:
I took a peek at WDT stuff while implementing deep sleep via https://github.com/adafruit/circuitpython/pull/5425. 

My take away is that it's super doable. I don't have time atm, but for anyone wanting to attempt this, here are some tips:

- use `\ports\nrf\common-hal\watchdog\` as a rough starting point
- you'll likely need to reference `\ports\atmel-samd\asf4\samd51\include\component\wdt.h` quite a bit for specific register twiddling
- Just checked the D5x/E5x errata (document DS80000748L) and there are no WDT issues reported (this bit me on the RTC for deep sleep lol)

dhalbert, 2023-10-28:
Fixed by #5483.

closed 2023-10-28 by dhalbert