adafruit/circuitpython #5426: ESP32-S2 can't boot to UF2-Bootloader from user code

FlorianBaumgartner, 2021-10-04:
### CircuitPython version

```python
Setup: Adafruit ESP32-S2 Metro, UF2 0.5.2, CircuitPython 7.0.0, Windows 10
```

### Code/REPL

```python
import microcontroller
microcontroller.on_next_reset(microcontroller.RunMode.UF2)
microcontroller.reset()
```

### Behavior

Acts just as a regular system reset. Device does not enter UF2-Bootloader.

### Description

Even by directly calling:
```
esp_reset_reason();
esp_reset_reason_set_hint((esp_reset_reason_t) 0x11F2);
esp_restart();
```
The ESP32-S2 does not enter UF2-Bootloader.
The only way is to double press the reset button, or reset the device and shortly after press the boot button.

microDev1, 2021-10-04:
Unable to reproduce this on:
```py
TinyUF2 Bootloader 0.5.2 - tinyusb (0.10.1-109-g9b3ec69b)
Adafruit CircuitPython 7.0.0 on 2021-09-20; microS2 with ESP32S2
```

hathach, 2021-10-04:
@FlorianBaumgartner you need to update the boot2 partition as well (which mean reflash the whole bootloader https://learn.adafruit.com/adafruit-metro-esp32-s2/install-uf2-bootloader)  . Updating bootloader via uf2 unfortunately does not fix this issue.

FlorianBaumgartner, 2021-10-04:
Thank you so much @hathach. That solved the problem. I didn't know that it is necessary to completely update all partitions.
Now it works fine. You can close this issue.

closed 2021-10-04 by microDev1

hathach, 2021-10-05:
@FlorianBaumgartner no problems, not all issue requires boot2 stage update like this one. Since boot2 contain bare minimum code. Though IDF recent changes break (redefine) some of the reset reason constant which we use to detect software reset.