adafruit/circuitpython #5431: rotaryio does not work properly with half-cycle-per-detente encoders

evildave666, 2021-10-06:
They are not as common as other types, but encoders with a half cycle per detent do exist and I have actually picked up a few recently.

With the current rotaryio they only advance once every 2 clicks.

It would be nice to be able to support them with an option in the constructor.

dhalbert, 2021-10-06:
We support rotary encoders that do Gray code output (the two outputs change alternately). This allows reliable detection of transitions without debouncing. I don't know about yours, but perhaps it does Gray code every other click. A link to the datasheet for yours would help us figure it out.

dhalbert, 2021-10-06:
This may be a dupe of #1677.

evildave666, 2021-10-06:
I've managed to find the original (chinese) documentation, and indeed it is a half-cycle encoder (30 detent, 15 cycles).

[https://img.alicdn.com/imgextra/i4/2452639374/O1CN015RoFrP2J7LDRRvmY3_!!2452639374.png](url)

To properly support it, it needs to output on both 00 and 11 states, not just 00.

ladyada, 2021-10-06:
this is kinda more like a motor encoder than a rotary encoder (rotary encoders are a subset of motor encoders that can stop at any point)