adafruit/circuitpython #5432: nrf: Fix flashing with pyocd

xlla, 2021-10-06:
### CircuitPython version

```python
master branch
```

### Code/REPL

```python
none
```

### Behavior

none

### Description

The latest version pyocd remove pyocd-tool and pyocd-flashtool, after create firmware, I can't flash it on to mcu board using param  FLASHER=pyocd.

### Additional information

I manage to migrate to latest pyocd's syntax , but I can't translate this sentence

`pyocd-tool -t $(MCU_VARIANT) write32 $(BOOT_SETTING_ADDR) 0x00000001`

I have done similar migration on micropython and successful flash firmware on to nrf52832 mcu board using latest pyocd. 
Micropython's flash sentence of pyocd for nrf is slighter differ from circuitpython's , if I use theirs to flash circuitpython, it could flash with some warnings
`pyocd erase -t nrf52 --chip	
0001557:INFO:eraser:Erasing chip...
0002210:INFO:eraser:Done
pyocd load -t nrf52 bluetooth/s140_nrf52_6.1.0/s140_nrf52_6.1.0_softdevice.hex
0001980:INFO:load_cmd:Loading circuitpython/ports/nrf/bluetooth/s140_nrf52_6.1.0/s140_nrf52_6.1.0_softdevice.hex
[==================================================] 100%
0015044:INFO:loader:Erased 155648 bytes (38 sectors), programmed 155648 bytes (38 pages), skipped 0 bytes (0 pages) at 11.80 kB/s
0015081:ERROR:jlink:Could not start CPU core. (ErrorCode: -1)
pyocd load -t nrf52 build-makerdiary_nrf52840_mdk-s140/firmware.hex
0001529:INFO:load_cmd:Loading circuitpython/ports/nrf/build-makerdiary_nrf52840_mdk-s140/firmware.hex
0002100:WARNING:file_programmer:Failed to add data chunk: no memory region defined for address 0x00080000
[==================================================] 100%
0031553:INFO:loader:Erased 368640 bytes (90 sectors), programmed 368640 bytes (90 pages), skipped 0 bytes (0 pages) at 12.23 kB/s
` 

and counld not get REPL to work.

I use same title as #1023 since it would not work now.

tannewt, 2021-10-07:
Make sure you are using the `main` branch, not `master`.

What version of `pyocd` are you using?

dhalbert, 2023-10-28:
Closing due to no followup.

closed 2023-10-28 by dhalbert