adafruit/circuitpython #5445: Rename HID device

benwa, 2021-10-08:
Currently, unless we recompile CircuitPython, we have to use "CircuitPython HID" for things like a game controller.
I'd like to be able to rename that, if at all possible.

dhalbert, 2021-10-08:
- Related to #4433.

fasteddy516, 2022-02-04:
I would love to see this functionality as well.  If you have multiple CP USB-HID devices connected simultaneously, selecting a specific one from a list (i.e. choosing a controller in a game, or a specialized input device for an editing suite) involves a mini-game of "which 'CircuitPython HID' device is which". 😃   

Something along the lines of `usb_hid.enable(name="My Nifty Controller", ...)` in `boot.py` would be lovely.

George-TL, 2023-01-30:
With CPv8 we have supervisor.set_usb_identification() which is step in right direction.
Unfortunately "lsusb -v" still shows "CircuitPython HID" on iInterface attribute.

usamashafiqs, 2023-08-10:
Any update on this

dhalbert, 2023-08-11:
There are no immediate plans to implement this. If someone feels like working on a PR, that would be great.

MariuszCwikla, 2024-02-28:
I would also like to have this feature.
I actually spent some time and implemented it here #8989

closed 2024-03-03 by dhalbert