adafruit/circuitpython #5451: TypeError: extra keyword arguments given

martingosnell, 2021-10-10:
### CircuitPython version

```python
adafruit-circuitpython-matrixportal_m4-en_US-7.0.0
```

### Code/REPL

```python
# Adapted from ...
# ON AIR sign for YouTube livestreaming
# Runs on Airlift Metro M4 with 64x32 RGB Matrix display & #shield
#https://cdn-learn.adafruit.com/downloads/pdf/rgb-matrix-
#automatic-youtube-on-air-sign.pdf?timestamp=1633673486

import time
import board
import displayio
import adafruit_display_text.label
from adafruit_display_shapes.rect import Rect
from adafruit_display_shapes.polygon import Polygon
from adafruit_bitmap_font import bitmap_font
#from adafruit_matrixportal.network import Network
from adafruit_matrixportal.matrix import Matrix

#Number of seconds between checking, if this is too #quick #the query quota will run out
UPDATE_DELAY = 300

# --- Display setup ---
matrix = Matrix()
display = matrix.display
#network = Network(status_neopixel=board.NEOPIXEL, debug=False)

# --- Drawing setup ---
# Create a Group
group = displayio.Group(max_size=22)
# Create a bitmap object
bitmap = displayio.Bitmap(64, 32, 2) # width, height, bit depth
# Create a color palette
color = displayio.Palette(4)
color[0] = 0x000000 # black
color[1] = 0xFF0000 # red
color[2] = 0x444444 # dim white
color[3] = 0xDD8000 # gold
# Create a TileGrid using the Bitmap and Palette
tile_grid = displayio.TileGrid(bitmap, pixel_shader=color)
# Add the TileGrid to the Group
group.append(tile_grid)

# draw the frame for startup
rect1 = Rect(0, 0, 2, 32, fill=color[2])
rect2 = Rect(62, 0, 2, 32, fill=color[2])
rect3 = Rect(2, 0, 9, 2, fill=color[0])
rect4 = Rect(53, 0, 9, 2, fill=color[0])
rect5 = Rect(2, 30, 12, 2, fill=color[0])
rect6 = Rect(50, 30, 12, 2, fill=color[0])

group.append(rect1)
group.append(rect2)
group.append(rect3)
group.append(rect4)
group.append(rect5)
group.append(rect6)

def redraw_frame(): # to adjust spacing at bottom later
    rect3.fill = color[2]
    rect4.fill = color[2]
    rect5.fill = color[2]
    rect6.fill = color[2]

# draw the wings w polygon shapes
wing_polys = []

wing_polys.append(Polygon([(3, 3), (9, 3), (9, 4), (4, 4)], outline=color[1]))
wing_polys.append(Polygon([(5, 6), (9, 6), (9, 7), (6, 7)], outline=color[1]))
wing_polys.append(Polygon([(7, 9), (9, 9), (9, 10), (8, 10)], outline=color[1]))
wing_polys.append(Polygon([(54, 3), (60, 3), (59, 4), (54, 4)], outline=color[1]))
wing_polys.append(Polygon([(54, 6), (58, 6), (57, 7), (54, 7)], outline=color[1]))
wing_polys.append(Polygon([(54, 9), (56, 9), (55, 10), (54, 10)], outline=color[1]))

for wing_poly in wing_polys:
    group.append(wing_poly)

def redraw_wings(index): # to change colors
    for wing in wing_polys:
        wing.outline = color[index]

# --- Content Setup ---
deco_font = bitmap_font.load_font("/BellotaText-Bold-21.bdf")

# Create two lines of text. Besides changing the text, you can also
# customize the color and font (using #Adafruit_CircuitPython_Bitmap_Font).

# text positions
on_x = 15
on_y = 9
off_x = 12
off_y = 9
air_x = 15
air_y = 25

text_line1 = adafruit_display_text.label.Label(
    deco_font, color=color[3], text="OFF", max_glyphs=6
)
text_line1.x = off_x
text_line1.y = off_y

text_line2 = adafruit_display_text.label.Label(
    deco_font, color=color[1], text="AIR", max_glyphs=6
)
text_line2.x = air_x
text_line2.y = air_y

# Put each line of text into the Group
group.append(text_line1)
group.append(text_line2)

def startup_text():
    text_line1.text = "ADA"
    text_line1.x = 10
    text_line1.color = color[2]
    text_line2.text = "FRUIT"
    text_line2.x = 2
    text_line2.color = color[2]
    redraw_wings(0)
    display.show(group)

startup_text() # display the startup text

def update_text(state):
    if state: # if switch is on, text is "ON" at startup
        text_line1.text = "ON"
        text_line1.x = on_x
        text_line1.color = color[1]
        text_line2.text = "AIR"
        text_line2.x = air_x
        text_line2.color = color[1]
        redraw_wings(1)
        redraw_frame()
        display.show(group)
    else: # else, text if "OFF" at startup
        text_line1.text = "OFF"
        text_line1.x = off_x
        text_line1.color = color[3]
        text_line2.text = "AIR"
        text_line2.x = air_x
        text_line2.color = color[3]
        redraw_wings(3)
        redraw_frame()
        display.show(group)

# Synchronize Board's clock to Internet
#network.get_local_time()
#mode_state = get_status()
mode_state = True    #MG instead of calling get_status (1)
update_text(mode_state)
last_check = None

while True:
    if last_check is None or time.monotonic() > last_check + UPDATE_DELAY:
        try:
            #status = get_status()
            status = True    #MG instead of calling get_status (2)

            if status:
                if mode_state == 0: # state has changed, toggle it
                    update_text(1)
                    mode_state = 1
            else:
                if mode_state == 1:
                    update_text(0)
                    mode_state = 0
            print("On Air:", status)
            #last_check = time.monotonic()
        except RuntimeError as e:
            print("Some error occured, retrying! -", e)
```

### Behavior

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 27, in <module>
TypeError: extra keyword arguments given

Code done running.

### Description

So I just want some code to simply draw a shape on an LED array display so that I can adapt it to my requirements.
I decided to use the adafruit  youtube on air example (less the network stuff) as a basis and simplify the code just to draw something (If any one has an even simpler example that works I would love to hear)..
Ive adapted the code as seen above but I do not understand the issue with line 27.
Looking into at displayio.Group the statement displayio.Group(max_size=22) looks fine
and I also tried displayio.Group(max_size=22, scale=1, x=0, y=0) which gave the same error.
Apologies if this is not a bug and I've done something ridiculous but I cannot see the problem.
Cheers

### Additional information

Tried resetting, reinstalling circuitpython, emptying trash and reloading libs, used latest install and libs downloaded a few days ago....   I anyone has some other very simply examples for darwing a shape that would be terrific, I want to get it working on the 64x32 LED matrix supplied with my matrix portal then get it working on the 2.5mm 64x64...
cheers

Neradoc, 2021-10-10:
dispayio.Group does not take a `max_size` argument anymore in CP7.
https://circuitpython.readthedocs.io/en/latest/shared-bindings/displayio/index.html#displayio.Group

You link the pdf in your code, it looks like the pdf has an outdated version of the code compared to the web version, I don't know how that works, but I'll try to get the info to the right people if they miss this.

I advise using the "download project bundle" button here instead:
https://learn.adafruit.com/rgb-matrix-automatic-youtube-on-air-sign/code-the-on-air-sign

dhalbert, 2021-10-10:
@Neradoc Thanks for diagnosing this. I've reported it to the Learn Guide folks.
@martingosnell Yes, get the code from the online guide. Sorry for the mismatch.

closed 2021-10-10 by dhalbert