adafruit/circuitpython #5470: STM32H743 Nucleo USB connection to Host not working with Circuit Python V7.0.0

S4WRXTTCS, 2021-10-16:
### CircuitPython version

```python
Adafruit Circuit Python V7.0.0 Image from the website
```

### Code/REPL

```python
N/A
```

### Behavior

N/A

### Description

The STM32H743 Nucleo board doesn't seem to connect to via either the USB Serial or Mass Storage connection to the host using the Circuit Python 7.0.0 bin image off the website or an image built from the CircuitPython GitHub repository.

I verified the board works fine using the latest version of MicroPython.

I've also used CircuitPython V7.0.0 with a couple STM32F405 boards without any issue. 

I'm going to test out some earlier builds of CircuitPython to see if those work.

zhangsiyan12134, 2024-01-20:
Hi, I just tested pre-build CP firmware ver 8.2.9 and 9.0.0 Alpha 6 on nucleo-h743zi2. This problem still exists.

I understand the reason why it is lacking the support from Adafruit, but I think you should remove the download page or the download button at least on CP website if it is no longer working. Doing so to prevent someone like me who wants to purchased a more powerful "supported" board and realized it is actually not later. As suggested above, I will try the Blinka on MicroPython for now.

tannewt, 2024-01-23:
> I understand the reason why it is lacking the support from Adafruit, but I think you should remove the download page or the download button at least on CP website if it is no longer working. Doing [...]

Please file an issue here: https://github.com/adafruit/circuitpython-org I don't agree that it should be removed, but we should make it clear that it is alpha quality. (per [the release notes](https://github.com/adafruit/circuitpython/releases/tag/8.2.9))