adafruit/circuitpython #5529: Support HID Usage Pages and Usages >255

dhalbert, 2021-10-30:
Thank you for this!  I just tried new custom HID reports with the blink(1) USB LED report descriptor (which uses FEATURE reports w/ reportIds) and TinyUSB hid_generic_inout (IN/OUT reports). But I get `ValueError: usage_page must be 1-255`. The HID report descriptors look like the below, using the 16-bit way of specifying usagePage & usage, and I think the current `usb_hid` assumes usagePage/usage are only 8-bit.  I've noticed this 16-bit usagePage/usage specification is used by several "rawhid" implementations (e.g. [teensy rawhid](https://github.com/PaulStoffregen/cores/blob/2e9aa7a34f2e7914767fde52fd1a9c3be7c30dd1/usb_rawhid/usb.c#L92), [Arduino HID library](https://github.com/NicoHood/HID/blob/master/src/SingleReport/RawHID.cpp), [TinyUSB hid_generic_inout](https://github.com/hathach/tinyusb/blob/master/src/class/hid/hid_device.h#L358) )

The TinyUSB hid_generic_inout HID Report Descriptor looks like this (one IN report, one OUT report, both 64 bytes, no reportId):

```
RAWHID_REPORT_DESCRIPTOR = bytes((
    0x06, 0x00, 0xFF,  # Usage Page (Vendor Defined 0xFF00)
    0x09, 0x01,        # Usage (0x01)
    0xA1, 0x01,        # Collection (Application)
    0x09, 0x02,        #   Usage (0x02)
    0x15, 0x00,        #   Logical Minimum (0)
    0x26, 0xFF, 0x00,  #   Logical Maximum (255)
    0x75, 0x08,        #   Report Size (8)
    0x95, 0x40,        #   Report Count (64)
    0x81, 0x02,        #   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x09, 0x03,        #   Usage (0x03)
    0x15, 0x00,        #   Logical Minimum (0)
    0x26, 0xFF, 0x00,  #   Logical Maximum (255)
    0x75, 0x08,        #   Report Size (8)
    0x95, 0x40,        #   Report Count (64)
    0x91, 0x02,        #   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
    0xC0,              # End Collection
))
```

The blink(1) HID Report Descriptor looks like this (two FEATURE reports, both with report IDs: reportId 1 is 8 bytes, reportId2 is 60 bytes): 

```
BLINK1_REPORT_DESCRIPTOR = bytes((
    0x06, 0xAB, 0xFF,              # usage page vendor defined 0xFFAB
    0x0A, 0x00, 0x20,              # usage 0x2000
    0xA1, 0x01,                    # COLLECTION (Application)
    0x15, 0x00,                    #   LOGICAL_MINIMUM (0)
    0x26, 0xff, 0x00,              #   LOGICAL_MAXIMUM (255)
    0x75, 0x08,                    #   REPORT_SIZE (8)
    0x85, 1,                       #   REPORT_ID (1)
    0x95, 8,                       #   REPORT_COUNT (8)
    0x09, 0x00,                    #   USAGE (Undefined)
    0xb2, 0x02, 0x01,              #   FEATURE (Data,Var,Abs,Buf)
    0x75, 0x08,                    #   REPORT_SIZE (8)
    0x85, 2,                       #   REPORT_ID (2)
    0x95, 60,                      #   REPORT_COUNT (60)
    0x09, 0x00,                    #   USAGE (Undefined)
    0xb2, 0x02, 0x01,              #   FEATURE (Data,Var,Abs,Buf)
    0xc0,                          # END_COLLECTION
))
```

_Originally posted by @todbot in https://github.com/adafruit/circuitpython/issues/5197#issuecomment-955086563_

ReeceRobinson, 2022-04-02:
Does anyone know what the level of effort would be to fix this issue? I really need this to be resolved for a project I want to do.

dhalbert, 2022-04-02:
I _think_ it's just a matter of changing a bunch of `uint8_t`s to `uint16_t`s, and allowing a wider range in the argument validation. The construction of the HID Report Descriptors is up to the user -- we don't construct them for you.

ReeceRobinson, 2022-04-02:
Thanks for that. I'll see if I can get a build environment working on my Mac so I can try to make the required changes.

ReeceRobinson, 2022-04-03:
I managed to get a working build of the main branch on my Mac. The changes I think are necessary are:

```diff --git a/shared-bindings/usb_hid/Device.c b/shared-bindings/usb_hid/Device.c
index 286623b0c..f85cf64ea 100644
--- a/shared-bindings/usb_hid/Device.c
+++ b/shared-bindings/usb_hid/Device.c
@@ -103,11 +103,11 @@ STATIC mp_obj_t usb_hid_device_make_new(const mp_obj_type_t *type, size_t n_args
     mp_obj_t descriptor = mp_obj_new_bytearray(descriptor_bufinfo.len, descriptor_bufinfo.buf);
 
     const mp_int_t usage_page_arg = args[ARG_usage_page].u_int;
-    mp_arg_validate_int_range(usage_page_arg, 1, 255, MP_QSTR_usage_page);
+    mp_arg_validate_int_range(usage_page_arg, 1, 0xFFFF, MP_QSTR_usage_page);
     const uint8_t usage_page = usage_page_arg;
 
     const mp_int_t usage_arg = args[ARG_usage].u_int;
-    mp_arg_validate_int_range(usage_arg, 1, 255, MP_QSTR_usage_page);
+    mp_arg_validate_int_range(usage_arg, 1, 0xFFFF, MP_QSTR_usage_page);
     const uint8_t usage = usage_arg;
 
     mp_obj_t report_ids = args[ARG_report_ids].u_obj;
diff --git a/shared-module/usb_hid/Device.h b/shared-module/usb_hid/Device.h
index 5a09d1952..f265712be 100644
--- a/shared-module/usb_hid/Device.h
+++ b/shared-module/usb_hid/Device.h
@@ -42,8 +42,8 @@ typedef struct  {
     uint8_t report_ids[CIRCUITPY_USB_HID_MAX_REPORT_IDS_PER_DESCRIPTOR];
     uint8_t in_report_lengths[CIRCUITPY_USB_HID_MAX_REPORT_IDS_PER_DESCRIPTOR];
     uint8_t out_report_lengths[CIRCUITPY_USB_HID_MAX_REPORT_IDS_PER_DESCRIPTOR];
-    uint8_t usage_page;
-    uint8_t usage;
+    uint16_t usage_page;
+    uint16_t usage;
     uint8_t num_report_ids;
 } usb_hid_device_obj_t;
```
I will continue with my testing to verify that is all that is needed.

dhalbert, 2022-04-03:
Good work! These also need to be uint16_t:
```c
     const uint8_t usage_page = usage_page_arg;
```
```c
     const uint8_t usage = usage_arg;
```
And there may be further ones in shared-module.

ReeceRobinson, 2022-04-04:
Thanks for that. Yes there were more locations to update. I think I have found them. Here is the git diff now:
```diff --git a/mpy-cross/mpy-cross.mk b/mpy-cross/mpy-cross.mk
index 642fe9b5a..c7bd91ad1 100644
--- a/mpy-cross/mpy-cross.mk
+++ b/mpy-cross/mpy-cross.mk
@@ -29,6 +29,8 @@ INC += -I$(BUILD)
 # compiler settings
 CWARN = -Wall -Werror
 CWARN += -Wpointer-arith -Wuninitialized
+# Disable errors for array-bounds warnings on "sp[-MP_OBJ_ITER_BUF_NSLOTS + 2]" access
+CWARN += -Wno-array-bounds
 CFLAGS = $(INC) $(CWARN) -std=gnu99 $(CFLAGS_MOD) $(COPT) $(CFLAGS_EXTRA)
 CFLAGS += -fdata-sections -ffunction-sections -fno-asynchronous-unwind-tables
 
diff --git a/ports/broadcom/firmware b/ports/broadcom/firmware
--- a/ports/broadcom/firmware
+++ b/ports/broadcom/firmware
@@ -1 +1 @@
-Subproject commit bf96d0eda5952595d717fedb797aeb168483e9fa
+Subproject commit bf96d0eda5952595d717fedb797aeb168483e9fa-dirty
diff --git a/shared-bindings/usb_hid/Device.c b/shared-bindings/usb_hid/Device.c
index 286623b0c..50f67e90c 100644
--- a/shared-bindings/usb_hid/Device.c
+++ b/shared-bindings/usb_hid/Device.c
@@ -103,12 +103,12 @@ STATIC mp_obj_t usb_hid_device_make_new(const mp_obj_type_t *type, size_t n_args
     mp_obj_t descriptor = mp_obj_new_bytearray(descriptor_bufinfo.len, descriptor_bufinfo.buf);
 
     const mp_int_t usage_page_arg = args[ARG_usage_page].u_int;
-    mp_arg_validate_int_range(usage_page_arg, 1, 255, MP_QSTR_usage_page);
-    const uint8_t usage_page = usage_page_arg;
+    mp_arg_validate_int_range(usage_page_arg, 1, 0xFFFF, MP_QSTR_usage_page);
+    const uint16_t usage_page = usage_page_arg;
 
     const mp_int_t usage_arg = args[ARG_usage].u_int;
-    mp_arg_validate_int_range(usage_arg, 1, 255, MP_QSTR_usage_page);
-    const uint8_t usage = usage_arg;
+    mp_arg_validate_int_range(usage_arg, 1, 0xFFFF, MP_QSTR_usage_page);
+    const uint16_t usage = usage_arg;
 
     mp_obj_t report_ids = args[ARG_report_ids].u_obj;
     mp_obj_t in_report_lengths = args[ARG_in_report_lengths].u_obj;
diff --git a/shared-bindings/usb_hid/Device.h b/shared-bindings/usb_hid/Device.h
index a5e94c4d5..dde66e270 100644
--- a/shared-bindings/usb_hid/Device.h
+++ b/shared-bindings/usb_hid/Device.h
@@ -33,11 +33,11 @@
 
 extern const mp_obj_type_t usb_hid_device_type;
 
-void common_hal_usb_hid_device_construct(usb_hid_device_obj_t *self, mp_obj_t report_descriptor, uint8_t usage_page, uint8_t usage, size_t report_ids_count,uint8_t *report_ids, uint8_t *in_report_lengths, uint8_t *out_report_lengths);
+void common_hal_usb_hid_device_construct(usb_hid_device_obj_t *self, mp_obj_t report_descriptor, uint16_t usage_page, uint16_t usage, size_t report_ids_count,uint8_t *report_ids, uint8_t *in_report_lengths, uint8_t *out_report_lengths);
 void common_hal_usb_hid_device_send_report(usb_hid_device_obj_t *self, uint8_t *report, uint8_t len, uint8_t report_id);
 mp_obj_t common_hal_usb_hid_device_get_last_received_report(usb_hid_device_obj_t *self, uint8_t report_id);
-uint8_t common_hal_usb_hid_device_get_usage_page(usb_hid_device_obj_t *self);
-uint8_t common_hal_usb_hid_device_get_usage(usb_hid_device_obj_t *self);
+uint16_t common_hal_usb_hid_device_get_usage_page(usb_hid_device_obj_t *self);
+uint16_t common_hal_usb_hid_device_get_usage(usb_hid_device_obj_t *self);
 uint8_t common_hal_usb_hid_device_validate_report_id(usb_hid_device_obj_t *self, mp_int_t report_id);
 
 #endif  // MICROPY_INCLUDED_SHARED_BINDINGS_USB_HID_DEVICE_H
diff --git a/shared-module/usb_hid/Device.c b/shared-module/usb_hid/Device.c
index de34d60d0..8a0c429f8 100644
--- a/shared-module/usb_hid/Device.c
+++ b/shared-module/usb_hid/Device.c
@@ -186,7 +186,7 @@ uint8_t common_hal_usb_hid_device_validate_report_id(usb_hid_device_obj_t *self,
     return (uint8_t)report_id_arg;
 }
 
-void common_hal_usb_hid_device_construct(usb_hid_device_obj_t *self, mp_obj_t report_descriptor, uint8_t usage_page, uint8_t usage, size_t num_report_ids, uint8_t *report_ids, uint8_t *in_report_lengths, uint8_t *out_report_lengths) {
+void common_hal_usb_hid_device_construct(usb_hid_device_obj_t *self, mp_obj_t report_descriptor, uint16_t usage_page, uint16_t usage, size_t num_report_ids, uint8_t *report_ids, uint8_t *in_report_lengths, uint8_t *out_report_lengths) {
     if (num_report_ids > CIRCUITPY_USB_HID_MAX_REPORT_IDS_PER_DESCRIPTOR) {
         mp_raise_ValueError_varg(translate("More than %d report ids not supported"),
             CIRCUITPY_USB_HID_MAX_REPORT_IDS_PER_DESCRIPTOR);
@@ -211,11 +211,11 @@ void common_hal_usb_hid_device_construct(usb_hid_device_obj_t *self, mp_obj_t re
     memcpy(self->out_report_lengths, out_report_lengths, num_report_ids);
 }
 
-uint8_t common_hal_usb_hid_device_get_usage_page(usb_hid_device_obj_t *self) {
+uint16_t common_hal_usb_hid_device_get_usage_page(usb_hid_device_obj_t *self) {
     return self->usage_page;
 }
 
-uint8_t common_hal_usb_hid_device_get_usage(usb_hid_device_obj_t *self) {
+uint16_t common_hal_usb_hid_device_get_usage(usb_hid_device_obj_t *self) {
     return self->usage;
 }
 
diff --git a/shared-module/usb_hid/Device.h b/shared-module/usb_hid/Device.h
index 5a09d1952..f265712be 100644
--- a/shared-module/usb_hid/Device.h
+++ b/shared-module/usb_hid/Device.h
@@ -42,8 +42,8 @@ typedef struct  {
     uint8_t report_ids[CIRCUITPY_USB_HID_MAX_REPORT_IDS_PER_DESCRIPTOR];
     uint8_t in_report_lengths[CIRCUITPY_USB_HID_MAX_REPORT_IDS_PER_DESCRIPTOR];
     uint8_t out_report_lengths[CIRCUITPY_USB_HID_MAX_REPORT_IDS_PER_DESCRIPTOR];
-    uint8_t usage_page;
-    uint8_t usage;
+    uint16_t usage_page;
+    uint16_t usage;
     uint8_t num_report_ids;
 } usb_hid_device_obj_t;
```
Assuming this is okay what do you want me to do with this patch?

dhalbert, 2022-04-04:
Could you make a pull request? That would be ideal - thanks.

dhalbert, 2022-04-04:
If you are not familiar with how to make a pull request, we have a guide here: https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github

ReeceRobinson, 2022-04-05:
I think I created the pull request correctly (see above https://github.com/adafruit/circuitpython/pull/6239). Let me know if I need to try again :-)

PR #6239 (2022-04-05): Enable support for extended HID usage page and usage. Related to Supp…
Enable support for extended HID usage page and usage. Related to Support HID Usage Pages and Usages >255 #5529.
Also added suppression for array-bounds warning to enable mpy-cross to build on Mac.

Fixes #5529 (EDIT by @dhalbert for automatic linkage to issue)
closed 2022-04-05 by dhalbert