adafruit/circuitpython #5538: Sending Requests Failed on second network call in CircuitPython 7.0.0

makermelissa, 2021-11-02:
This was originally found by a user in the forums at https://forums.adafruit.com/viewtopic.php?f=19&t=184484&p=895677. I was able to reproduce the issue by following the guide at https://learn.adafruit.com/magtag-google-calendar-event-display.

Here's the stack trace
```
Traceback (most recent call last):
File "code.py", line 246, in <module>
File "code.py", line 84, in get_current_time
File "adafruit_portalbase/__init__.py", line 411, in get_local_time
File "adafruit_portalbase/network.py", line 231, in get_local_time
File "adafruit_portalbase/network.py", line 200, in get_strftime
File "adafruit_requests.py", line 613, in get
File "adafruit_requests.py", line 554, in request
File "adafruit_requests.py", line 423, in _get_socket
RuntimeError: Sending request failed
```

After making is so `get_local_time` was only called once, it failed on the next network call with this stack trace:
```
Traceback (most recent call last):
  File "code.py", line 227, in <module>
  File "adafruit_oauth2.py", line 179, in refresh_access_token
  File "adafruit_requests.py", line 617, in post
  File "adafruit_requests.py", line 554, in request
  File "adafruit_requests.py", line 423, in _get_socket
RuntimeError: Sending request failed
```

The first call is successful, but any subsequent calls fail. This only happens in CircuitPython 7.0.0 and not 6.3.0. I have not tried any of the alpha, beta, or release candidate versions in between. 

Note: This issue may need to be moved to the CircuitPython repo

makermelissa, 2021-11-02:
Transferring to CircuitPython since the issue is dependent on the version of CircuitPython and the fix is probably also in the core.

tbjers, 2021-12-06:
I'm also experiencing this issue with the MagTag and Google Calendar. Both on 7.0 and 7.1.

tbjers, 2021-12-07:
@makermelissa This issue is just an issue with the example. I have traced through the code and discovered:

1. There can only be one `socket_pool` per `wifi.radio`
2. Initializing the radio twice isn't helping
3. The fix can be done entirely in the code using these libraries

My only minor gripe is that I have to load `secrets` inside my code. It would be great if there was a public accessor or a way to pass the secrets through `MagTag` into `PortalBase` and `NetworkBase`. That way they can be read once, and be one source of truth.

Here's how to fix this issue (verified on a MagTag running CircuitPython 7.1.0-beta.1):

Remove initialization of `wifi` and `requests` in `code.py`.

Move MagTag init above OAuth2 init.

Call:
```python
magtag.network.connect()

# Initialize an OAuth2 object with GCal API scope
scopes = ["https://www.googleapis.com/auth/calendar.readonly"]
google_auth = OAuth2(
    magtag.network.requests,
    secrets["google_client_id"],
    secrets["google_client_secret"],
    scopes,
    secrets["google_access_token"],
    secrets["google_refresh_token"],
)
```

Then update:

```python
def get_calendar_events #...
```

to use:

```python
magtag.network.requests.get()
```

and everything works swimmingly!

JannieT, 2022-01-03:
Thanks @tbjers ! That helped a lot.
This is my working Calendar project  that runs fine on CircuitPython v7.1.0
https://gist.github.com/JannieT/1e589b215ec05118ab78ec5d441c69e7

bserinese, 2022-01-06:
My MagTag just arrived yesterday and I spent today walking through the Calendar tutorial. Glad I found this thread because I was hitting the same errors. I've downloaded the example above, but have a question about calendar_id.

In the original code from the tutorial, the calendar_id for my personal calendar was hardcoded into the code.py. Does the full working example mentioned above now have the calendar_id in the secret.py file? 
Or, should it go here:
```
now = self._iso_date(self.clock.datetime)
minTime = self._url_encode(now)
url = ("https://www.googleapis.com/calendar/v3/calendars/{0}"
       "/events?maxResults={1}&timeMin={2}&orderBy=startTime"
       "&singleEvents=true").format(self.secrets['calendar_id'], event_count, minTime)
```

tbjers, 2022-01-07:
> My MagTag just arrived yesterday and I spent today walking through the Calendar tutorial. Glad I found this thread because I was hitting the same errors. I've downloaded the example above, but have [...]

I would definitely put the `calendar_id` in your `secrets.py` file.

t6b, 2022-02-03:
> @makermelissa This issue is just an issue with the example. I have traced through the code and discovered:
> 
> 1. There can only be one `socket_pool` per `wifi.radio`
> 2. Initializing the radio twice isn't helping
> 3. The fix can be done entirely in `code.py` using these libraries
> 
> My only minor gripe is that I have to load `secrets` inside my code. It would be great if there was a public accessor or a way to pass the secrets through `MagTag` into `PortalBase` and `NetworkBase`. That way they can be read once, and be one source of truth.
> 
> Here's how to fix this issue (verified on a MagTag running CircuitPython 7.1.0-beta.1):
> 
> Remove initialization of `wifi` and `requests` in `code.py`.
> 
> Move MagTag init above OAuth2 init.
> 
> Call:
> 
> ```python
> magtag.network.connect()
> 
> # Initialize an OAuth2 object with GCal API scope
> scopes = ["https://www.googleapis.com/auth/calendar.readonly"]
> google_auth = OAuth2(
>     magtag.network.requests,
>     secrets["google_client_id"],
>     secrets["google_client_secret"],
>     scopes,
>     secrets["google_access_token"],
>     secrets["google_refresh_token"],
> )
> ```
> 
> Then update:
> 
> ```python
> def get_calendar_events #...
> ```
> 
> to use:
> 
> ```python
> magtag.network.requests.get()
> ```
> 
> and everything works swimmingly!

Thank you kindly for posting this @tbjers. I am trying to make this work, but I still have a lot to learn. I am not exactly sure how "def get_calendar_events #...." should be adjusted to use "magtag.network.requests.get(). Any advice would be very much appreciated! Thank you.

somenice, 2022-03-08:
Hi everyone, I'm currently using
**Adafruit CircuitPython 7.1.1 on 2022-01-14; Adafruit FunHouse with ESP32S2**
and I appear to be encountering a similar issue as reported here.
Never an issue with the first request, 200 OK status with get time from Adafruit IO, but fails on second request.
I've made sure to only start one Session as suggested above and tried catching any of the following errors ConnectionError, OSError, and RuntimeError but get the following:

Traceback (most recent call last):
  File "asyncio/core.py", line 189, in run_until_complete
  File "code.py", line 311, in main
  File "asyncio/funcs.py", line 83, in gather
  File "asyncio/funcs.py", line 78, in gather
  File "asyncio/task.py", line 157, in __next__
  File "asyncio/core.py", line 189, in run_until_complete
  File "code.py", line 178, in get_time
  File "adafruit_requests.py", line 613, in get
  File "adafruit_requests.py", line 554, in request
  File "adafruit_requests.py", line 423, in _get_socket
RuntimeError: Sending request failed

Making this call every 60 seconds. 
Of course a HUGE benefit of asyncio is that even with a runtime error other functions can continue to run.

[image]

I've struggled a bit with the abstraction of Funhouse > PyPortal > Wifi/SocketPool and Adafruit_requests.
Would love to learn the most robust way to either persist a connection or repeat connection, either way.

Any help appreciated and please let me know if you need any further information.
Thank you

makermelissa, 2022-03-14:
Ok, I essentially ported this over to MatrixPortal to test how it would do on the ESP32 Coprocessor and in the process found a bunch of stuff that is creating duplicate references. So I think the root problem is caused by 2 things:

1. The code creates a lot of the networking stuff outside of the MagTag library (while at the same time automatically initializing the same stuff inside of the MagTag library)
2. The later CircuitPython versions don't handle the conflicting references very well.

So the fixes I propose are: 
1. Rewrite the example to remove a lot of the extra stuff, so it's more streamlined.
2. Come up with a simpler example that does essentially the same thing, so the multiple references issue can be fixed in the backend.

dhalbert, 2022-03-16:
@makermelissa Is the reworked library now working fine on MagTag and an M4 with AirLift (ESP32)? Should I retest both with the Google Calendar project?

makermelissa, 2022-03-16:
@dhalbert, the examples were rewritten for both MagTag and PyPortal to make better use of resources. You can retest if you would like, but I still need to come up with a simplified example that causes the problems.

makermelissa, 2022-03-17:
I've been trying to piece together a simple example, but it just keeps not failing. Unfortunately, maybe a complex example is all that does fail, like maybe it's related to memory or something. You can still access the old Google Calendar code before I fixed it up here:
https://github.com/adafruit/Adafruit_Learning_System_Guides/tree/d9e11884fbe45d1502447be6f74f665a4be47c2c/MagTag_Google_Calendar

dhalbert, 2022-07-06:
This may be fixed by https://github.com/adafruit/Adafruit_CircuitPython_Requests/pull/110 and https://github.com/adafruit/Adafruit_CircuitPython_Requests/pull/111. I will close this for now. Please reopen or make a new issue if necessary.

closed 2022-07-06 by dhalbert