adafruit/circuitpython #5543: readthedocs isn't building

jepler, 2021-11-04:
see e.g., this build: https://readthedocs.org/projects/circuitpython/builds/15185882/

```
Running Sphinx v3.5.4
Traceback (most recent call last):
  File "tools/extract_pyi.py", line 201, in convert_folder
    tree = ast.parse(fragment)
  File "/home/docs/.pyenv/versions/3.7.9/lib/python3.7/ast.py", line 35, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
  File "<unknown>", line 27
    def continuous_capture_start(self, buffer1: WriteableBuffer, buffer2: WriteableBuffer, /) -> None:
                                                                                           ^
SyntaxError: invalid syntax
make: *** [stubs] Error 1
```

This is due to syntax introduced into a comment at #5540.

dhalbert, 2021-11-04:
Would newer Sphinx handle this?

kattni, 2021-11-04:
I don't think so. I just tried to build it and I'm pretty sure I'm running the latest Sphinx locally, and it failed.

jepler, 2021-11-04:
no it's about the Python version.  This language feature, "/" in a parameter list of `def`, is new in python 3.8 or something.

dhalbert, 2021-11-04:
I was just wondering if a newer sphinx knew about that new feature, but I'm realizing it depends on Python to do the parsing. (I guess!)

We can consider moving to 3.8 for CI.

jepler, 2021-11-05:
It's the sphinx version on RTD that matters, can we select it there?

closed 2021-11-05 by dhalbert

jepler, 2021-11-05:
Builds are working now https://readthedocs.org/projects/circuitpython/builds/15188695/

PR #5545 (2021-11-05): Request RTD to use python3.9 for building
~~.. this isn't handled when the stubs extraction procedure runs on RTD, due to the python version they use.  It did not seem to be possible to select to run on a newer 3.x version.~~

Closes: #5543