adafruit/circuitpython #5548: Soft Rebooting of CP 7 every 5 seconds

kwpodolsky, 2021-11-05:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0 on 2021-09-20; Adafruit Feather RP2040 with rp2040
Board ID:adafruit_feather_rp2040
```

### Code/REPL

```python
# FILE: Code.py

import time
import board
import neopixel

#print("Hello World!")

pixels = neopixel.NeoPixel(board.NEOPIXEL, 1)

dir = 0
index = 0
colors = [[255,0,0],[0,255,0],[0,0,255],[255,255,0],[255,0,255],[0,255,255],[255,255,255]]

while True:
    colr = colors[index]
    pixels.fill((colr))
    time.sleep(0.5)
    pixels.fill((0, 0, 0))
    time.sleep(0.5)
    if dir == 0:
      if index == 6:
        dir = 1
#        index = 0
      else:
        index += 1
    else:
      if index == 0:
        dir = 0
#        index = 0
      else:
        index -= 1
```

### Behavior

Code starts to run and every 5 seconds CP does a soft reboot.
Following is on MU console.

"Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:

Code stopped by auto-reload.
soft reboot"

### Additional information

Program runs fine when issued via "import command from REPL"
I have NEVER seen this issue when using MicroPython and boot.py.
WHY does CP insist on soft rebooting or forcing user to enter REPL?
HOW can you have a program that auto runs?

kwpodolsky, 2021-11-05:
Disregard this bug issue.
Found that I needed to put...

import supervisor
supervisor.disable_autoreload() 

In my code.py file to prevent CP from soft rebooting.

This should be major NOTE somewhere in the CP tutorials.

dhalbert, 2021-11-05:
When some filesystem operation on CIRCUITPY causes a write, CircuitPython assumes that you have changed your program, and restarts `code.py`. Something on your host computer is writing to CIRCUITPY frequently. What is your host operating system, and what editor are you using? It may be a utility program that is writing to the drive or accessing it in some way that causes a metadata update, such an an indexing or backup program, or a disk-checking utility.

If you want to turn this feature off completely, you can use `supervisor.disable_autoreload()`. https://circuitpython.readthedocs.io/en/latest/shared-bindings/supervisor/index.html#supervisor.disable_autoreload. However, most people don't see the issue you're having, so it would be worth investigating why it's happening.

kwpodolsky, 2021-11-05:
NONE of the above.    If I disconnect the device and just run it off battery, it did the same thing.
It has to do with the AutoReload "feature" in CP.
Since MP doesn't have that, it was never an issue for me.

By adding the lines of code, above, problem is gone.

dhalbert, 2021-11-05:
> If I disconnect the device and just run it off battery, it did the same thing.

Auto-reload should not be triggered if the board is not connected to a host, so this is odd. What kind of battery arrangement were you using? What behavior did you see that it was auto-reloading on battery?

dhalbert, 2021-12-05:
> Auto-reload should not be triggered if the board is not connected to a host, so this is odd. What kind of battery arrangement were you using? What behavior did you see that it was auto-reloading on battery?

As I mentioned, this should not happen at all, if you are not connected to a host.

A usb "power pack" can require a certain amount of power to be drawn to stay on. Testing with a simple 5V AC adapter or charger would remove that possible issue.

Closing this for now, because I think this is a support issue, not  core issue.

akshay2000, 2023-02-04:
Leaving this comment her for posterity:

Mu can be a culprit sometimes. When I start Mu, it seems to write something to the CIRCUITPY filesystem every few seconds - causing the reload. @kwpodolsky You've probably moved on, but in case you still have your set up, can you kill Mu and see if the reload still happens?

ChuckM, 2026-10-22:
Its the neopixel library. It reboots on V10 of Circuit Python as well. On a Playground Express (SAMD21) run the basic neopixel example and it will repeatedly reboot.

dhalbert, 2026-10-22:
> 12\. Its the neopixel library. It reboots on V10 of Circuit Python as well. On a Playground Express (SAMD21) run the basic neopixel example and it will repeatedly reboot.

This does not happen in general. @ChuckM could you describe the setup?

1. What is the power source? Try it with a battery or an AC adapter if it is the host computer USB source, and vice versa.
2. Try the example above, but with lower brightness or change the color values to, say, 40, instead of 255.
3. What is the host operating system?
4. Various utility programs, especially on Windows, can cause frequent writes. Try a different computer or try the alternate power source, as above.

ChuckM, 2026-10-22:
Sure, I've pretty extensively tested this. I've tried both a USB power supply, (capable of 2.4A / port), Hub (similar), and PC (motherboard USB port). I have tried it with a Linux host and with a Windows host. With the Linux host I've mounted it noatime to avoid any touches on the file when it doesn't change. With the file open in an editor and with no files opened on the mounted system. (windows & linux). The target in this case is the Playground Express, I have two of them, both SAMD21's and both exhibit similar symptoms. It had been running at brightness 0.2 and I changed it to 0.5 but got no change. I just spent a few hours trying to figure out if `supervisor.RunReason` was a thing, and by that, if there was a way I could print out during each restart the reason for the restart according to the supervisor class. Turns out that isn't really possible. I have this code:
```python
r = supervisor.RunReason
print("Restarted because: ", end='')
if (r == supervisor.RunReason.STARTUP) : 
  print("Startup")
elif (r == supervisor.RunReason.AUTO_RELOAD) :
  print("Auto Reload")
elif (r == supervisor.RunReason.SUPERVISOR_RELOAD) :
  print("Supervisor Reload")
elif (r == supervisor.RunReason.REPL_RELOAD) :
  print("REPL Reload")
else : 
  print("Unknown Reason!!!", type(r).__name__)
```
Which always prints "unknown reason". If I don't run the neopixels it doesn't restart. I can have it print text out to the console in an endless loop. It also doesn't seem to have a problem if I just blink _one_ neopixel I didn't see anywhere to attach my JTAG probe so I'm not sure how to catch it "in the act" as it were. As a former EE my suspicion is that with all 10 neopixels on it gets to the point where the onboard power regulation sags and you get a brownout condition down stream of the USB power, or the SAMD1 misbehaves before it gets to the brownout and something like the watchdog timer hits or a hardfault which resets it. There is some correlation with temperature (this morning has been cold and rainy and it has run a long time before resetting) I suppose I could dig out the heat gun and see if I can affect it that way. 

And I was just trying to figure out if I could take a bunch of these to a conference and create a programming challenge for the attendees 😄 only to find it resetting, and searching for that symptom landed me here.

dhalbert, 2026-10-22:
Thanks for the extensive testing. So just so we can try to reproduce exactly, upload the code you are testing, unless it's already what's in the original top post. And you're using CircuitPython 10.3.1?

dhalbert, 2026-10-23:
I tried the OP example. It only blinks one NeoPixel. It does not restart as described. @ChuckM I'll try your code when you post it.

ChuckM, 2026-10-25:
Seriously appreciate your looking at this @dhalbert, so my code is here:

[code.py](https://github.com/user-attachments/files/33937728/code.py) Tried different brightness levels, but changing the speed seemed to be more impactful in terms of resets. With 50% brightness and 250mS interrupts for the rotation, it seems to be fairly reliable, as configure 10% brightness and 20mS delays it resets reasonably often. This is the boot.txt output
```
Adafruit CircuitPython 10.3.1 on 2026-09-14; Adafruit CircuitPlayground Express with samd21g18
Board ID:circuitplayground_express
UID:7EC6FF985346505120202035330C12FF
```
The interval between restarts is not constant (initially it seemed pretty steady but now it will restart after a few seconds and then it can go several minutes before restarting again, then maybe three restarts in the space of a minute Etc.) I've be trying to find an A/B test that will show it restarting or not restarting. I'm considering trying to bodge some wires on the CPU so that I could hook up my Atmel JTAG-ICE unit. I am thinking about trying the Arduino environment to see if C code has a similar issue. But I have limited cycles between now and the first couple of weeks of November.

dhalbert, 2026-10-25:
@ChuckM Your exact code is running fine on one of my CPX's. This is very odd and unusual. I tried with a heatgun and got the CPU chip up to 80 degrees C, as measured by `microcontroller.cpu.temperature`, with no problem.

1. What does `INFO_UF2.TXT` show in CPLAYBOOT?
2. When did you buy these CPX's, and where did you buy them from?
3. Could you post a picture of the back of the board?

ChuckM, 2026-10-31:
1.  Both have the following in their `INFO_UF2.TXT`

UF2 Bootloader v4.0.0 SFHWRO
Model: CPlay Express
Board-ID: SAMD21G18A-CPlay-v0
http://adafru.it/3333

2. Not sure where they came from, I got them from a person who was getting rid of their 'unused' stuff. 
3. [image]

dhalbert, 2026-11-08:
Sorry for the belated reply. It's possible whoever you got them from damaged them in some way, though damaging both would be odd. I opened a forum thread about this, in case anyone has seen this: https://forums.adafruit.com/viewtopic.php?t=225612.

mikeysklar, 2026-11-08:
@ChuckM `supervisor.RunReason` is the enum class, so that check always falls through. Could you try two things at the top of code.py?

1. Print why it started:

```python
import supervisor
print("run_reason:", supervisor.runtime.run_reason)
```

2. Then add this line and see if the restarts stop:

```python
supervisor.runtime.autoreload = False
```

If it shows `AUTO_RELOAD` and the restarts stop with autoreload off, something on the host is writing to CIRCUITPY. A brownout on SAMD21 stops in safe mode with "Power dipped" rather than rerunning code.py.

Which editor and serial terminal are you using? Some autosave or write swap files to the drive. Also, the code.py you attached has two copies of the example spliced together, which may be the editor too.

caternuson, 2026-11-09:
Just adding another not-able-to-reproduce test case.

CPX rev G running CP 10.3.1 and `code.py` from above. This has been running uninterrupted without issues for about 2 hours now:
[image]