adafruit/circuitpython #5552: Pro Micro RP2040 requires several restart before Adafruit FeatherWing OLED - 128x64 OLED start displaying

ithinuel, 2021-11-05:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0 on 2021-09-20; SparkFun Pro Micro RP2040 with rp2040
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
#
# SPDX-License-Identifier: Unlicense
"""
Author: Mark Roberts (mdroberts1243) from Adafruit code
This test will initialize the display using displayio and draw a solid white
background, a smaller black rectangle, miscellaneous stuff and some white text.

"""

import board
import displayio
import terminalio

# can try import bitmap_label below for alternative
from adafruit_display_text import label
import adafruit_displayio_sh1107

displayio.release_displays()
# oled_reset = board.D9

# Use for I2C
i2c = board.I2C()
display_bus = displayio.I2CDisplay(i2c, device_address=0x3C)

# SH1107 is vertically oriented 64x128
WIDTH = 128
HEIGHT = 64
BORDER = 2

display = adafruit_displayio_sh1107.SH1107(
    display_bus, width=WIDTH, height=HEIGHT, rotation=0
)

# Make the display context
splash = displayio.Group()
display.show(splash)

color_bitmap = displayio.Bitmap(WIDTH, HEIGHT, 1)
color_palette = displayio.Palette(1)
color_palette[0] = 0xFFFFFF  # White

bg_sprite = displayio.TileGrid(color_bitmap, pixel_shader=color_palette, x=0, y=0)
splash.append(bg_sprite)

# Draw a smaller inner rectangle in black
inner_bitmap = displayio.Bitmap(WIDTH - BORDER * 2, HEIGHT - BORDER * 2, 1)
inner_palette = displayio.Palette(1)
inner_palette[0] = 0x000000  # Black
inner_sprite = displayio.TileGrid(
    inner_bitmap, pixel_shader=inner_palette, x=BORDER, y=BORDER
)
splash.append(inner_sprite)

# Draw some white squares
sm_bitmap = displayio.Bitmap(8, 8, 1)
sm_square = displayio.TileGrid(sm_bitmap, pixel_shader=color_palette, x=58, y=17)
splash.append(sm_square)

med_bitmap = displayio.Bitmap(16, 16, 1)
med_square = displayio.TileGrid(med_bitmap, pixel_shader=color_palette, x=71, y=15)
splash.append(med_square)

lrg_bitmap = displayio.Bitmap(32, 32, 1)
lrg_square = displayio.TileGrid(lrg_bitmap, pixel_shader=color_palette, x=91, y=28)
splash.append(lrg_square)

# Draw some label text
text1 = "0123456789ABCDEF123456789AB"  # overly long to see where it clips
text_area = label.Label(terminalio.FONT, text=text1, color=0xFFFFFF, x=8, y=8)
splash.append(text_area)
text2 = "SH1107"
text_area2 = label.Label(
    terminalio.FONT, text=text2, scale=2, color=0xFFFFFF, x=9, y=44
)
splash.append(text_area2)

while True:
    pass
```

### Behavior

After a power cycle nothing shows on the display.

It takes 3 `Ctrl-C Ctrl-D` sequence over the serial port to restart the application 3 times before the display starts to work.

### Description

The code/REPL is the `displayio_sh1107_simpletest.py` example found in the bundle.
The same behaviour is observed with other `displayio_sh1107_*` examples.

### Additional information

The display is connected to the Pro Micro RP2040 using the Qwiic/Stemma connector&cable.

tannewt, 2021-11-08:
Are there any errors in the serial output when it doesn't work?

ithinuel, 2021-11-08:
No, the display just stays black.

ithinuel, 2022-06-19:
Hi, I was thinking again about this issue and the QWIIC link being dependent on the promicro-rp2040's power stage, I was wondering if it could be a "transient current" issue at the time we turn on the display (while the charge pumps are loading to generate the required 12V).
How could I test this hypothesis?

I had a look at the [FAN5331](https://www.onsemi.com/pdf/datasheet/fan5331-d.pdf) datasheet and its current limit during startup is 1A but the [promicro's RP2040](https://cdn.sparkfun.com/assets/4/a/6/d/6/SparkFun_Pro_Micro_RP2040_Schematic.pdf) voltage regulator has a maximum of 600mA.

ladyada, 2022-06-20:
you may need a delay while the display powers up. also reset may need to be connected
this featherwing is REALLY not designed for anything other than a feather, please use a 128x64 SSD1306 QT breakout instead!
https://www.adafruit.com/product/938

ithinuel, 2022-08-12:
Unfortunately that isn't really an option, the design was already fairly complete by the time this issue was discovered.
https://github.com/ithinuel/wilskeeb

ladyada, 2022-08-12:
you can probably just try re-initing after a while, or adding a stiffer pullup on RESET, but... this isnt an issue with circuitpython, its a 'creative and unsupported use of hardware' issue :)

closed 2022-08-12 by ladyada

ithinuel, 2022-08-12:
Fair enough for the circuit python. I figured that as an adafruit product, the display and rp2040 combo would still be supported in some way.

Although I'm using a Sparkfun promicro, the [KB2040](https://www.adafruit.com/product/5302) seems to have an even lower limitation on the power supply (500mA).

ladyada, 2022-08-12:
it is possible to make it work "you may need a delay while the display powers up. also reset may need to be connected" - but its not going to be as simple and ya dont wanna redesign the board