adafruit/circuitpython #5566: AES decrypt not working

bobricius, 2021-11-10:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0 on 2021-09-20; Raspberry Pi Pico with rp2040
Board ID:raspberry_pi_pico
```

### Code/REPL

```python
import time
import board
import busio
import terminalio
import displayio
import aesio
import microcontroller

from config import config
import digitalio

key = b'Sixteen byte key'
inp = b'CircuitPython!!!' # Note: 16-bytes long
outp = bytearray(len(inp))
cipher = aesio.AES(key, aesio.MODE_CBC)
cipher.encrypt_into(inp, outp)
print(key)
print(outp)
cipher.decrypt_into(outp, inp)
print(inp)
```

### Behavior

Traceback (most recent call last):
  File "<stdin>", line 20, in <module>
TypeError: object with buffer protocol required

### Description

I think decrypt not working, Maybe do not know how to use, I search examples but also on https://circuitpython.readthedocs.io/en ... index.html is on encrypt and look obsolete

Encrypt work
aesio.AES(key, aesio.MODE_CBC)

not with as in example
aesio.AES(key, aesio.mode.MODE_ECB)

Thanks very much for this project!

jepler, 2021-11-10:
Thanks!  There does seem to be something fishy going on here.

First, as to the TypeError message you got: `decrypt_into` takes its output as the second argument, but as `inp` is a bytes object it cannot be modified.  Unfortunately, the error message does not make this clear.

Second, on investigation I'm not sure whether our AES code is correct.  I was able to get a correct decryption back to the original plaintext while using CTR mode but not while using the other modes.

Here's the full program, which I used in the REPL of an ESP32-S2 board (not pico/rp2040) by using ctrl-e in the REPL to enter paste mode, pasting the full code in, and pressing ctrl-d.

```python
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import aesio
=== import os
=== from binascii import hexlify
=== 
=== 
=== for mode in (aesio.MODE_CBC, aesio.MODE_CTR, aesio.MODE_ECB):
===     print(mode)
===     key = b'Sixteen byte key'
===     inp = b'CircuitPython!!!' # Note: 16-bytes long
===     iv = os.urandom(16)
=== 
===     # Encrypt
===     outp = bytearray(len(inp))
===     cipher = aesio.AES(key, mode, iv)
===     cipher.encrypt_into(inp, outp)
===     hexlify(outp)
=== 
===     # Decrypt
===     cipher = aesio.AES(key, mode, iv)
===     inp = bytes(outp)
===     outp = bytearray(len(inp))
===     cipher.encrypt_into(inp, outp)
===     hexlify(outp)
===     print(outp)
===     print()
=== 
2
b'57620302dee74279565c03da9fb4f59f'
b'5ff6aa492c621c5b30d97281f3536082'
bytearray(b'_\xf6\xaaI,b\x1c[0\xd9r\x81\xf3S`\x82')

6
b'7f538b8c5320d3168fb97bbeffce9337'
b'43697263756974507974686f6e212121'
bytearray(b'CircuitPython!!!')

1
b'abb1a1f98f57409e455ac06e71535ffe'
b'c7516f3754877fe11346a7971e39e029'
bytearray(b'\xc7Qo7T\x87\x7f\xe1\x13F\xa7\x97\x1e9\xe0)')
```

bobricius, 2021-11-11:
With CTR it working, thanks very much, this mode is better because not need 16 bytes paging :)

jepler, 2021-11-28:
My [example code](https://github.com/adafruit/circuitpython/issues/5566#issuecomment-965796985) was wrong, since it used `encrypt_into` on the decrypt step.

PR #5638 (2021-12-01): Test aesio
This tests that the 3 modes of aesio work.  It doesn't introduce any of the incompatibilities of #5623.  Closes: #5566
closed 2021-12-01 by tannewt