adafruit/circuitpython #5581: mimxrt10xx: investigate whether uart_reset, i2c_reset need to be called

jepler, 2021-11-13:
### CircuitPython version

```python
main
```

### Code/REPL

```python
# n/a
```

### Behavior

The functions `uart_reset` and `i2c_reset` exist but are not called. Most likely, they're intended to be called from `reset_port()` but are not.

KurtE, 2022-04-14:
Just looking through the bugs listed as mimxrt10xx and was wondering if you tried adding these calls?

I do see there is a call to spi_reset().

dhalbert, 2022-04-14:
They should be added. The additions can be modeled on what happens in `port.c` in other ports.

KurtE, 2022-04-17:
@tennewt @dhalbert @ladyada  @gamblor21 @mjs513 

Sorry, I am not sure the best place to ask something like this:

What is your preferences/practices on Pull Requests and the like.
https://github.com/adafruit/circuitpython/pull/6287

Branch right now I am also testing it with a few other changes as well including for example the changes to add the uart_reset and i2c_reset.   Which is pretty straight forward:
```
diff --git a/ports/mimxrt10xx/supervisor/port.c b/ports/mimxrt10xx/supervisor/port.c
index 4552201f3..e6b8b336a 100644
--- a/ports/mimxrt10xx/supervisor/port.c
+++ b/ports/mimxrt10xx/supervisor/port.c
@@ -40,6 +40,8 @@
 #include "common-hal/pwmio/PWMOut.h"
 #include "common-hal/rtc/RTC.h"
 #include "common-hal/busio/SPI.h"
+#include "common-hal/busio/I2C.h"
+#include "common-hal/busio/UART.h"
 #include "shared-bindings/microcontroller/__init__.h"

 #include "reset.h"
@@ -270,6 +272,8 @@ safe_mode_t port_init(void) {

 void reset_port(void) {
     spi_reset();
+    i2c_reset();
+    uart_reset();

     #if CIRCUITPY_AUDIOIO
     audio_dma_reset();
```
While testing it with my UART test code on a Teensy 4, I noticed that we mistakenly added in definition for Tx4 and RX4, which was sort of obvious from my sketch output:
```
            print("Initialize Uarts")
            uarts = []
            for index in range(9):
                tx_name = "TX"+str(index)
                rx_name = "RX"+str(index)
                if hasattr(board, tx_name):
                    tx_pin = getattr(board, tx_name)
                    rx_pin = getattr(board, rx_name)
                    uart = busio.UART(tx_pin, rx_pin, baudrate=5000000)
                    uarts.append(uart)
                    print("\tAdded: ",index, " TX:",tx_pin, " RX:", rx_pin)
```
Output...
```
Initialize Uarts
	Added:  1  TX: board.D1  RX: board.D0
	Added:  2  TX: board.D8  RX: board.D7
	Added:  3  TX: board.D14  RX: board.D15
	Added:  4  TX: board.D17  RX: board.D16
	Added:  5  TX: board.D20  RX: board.D21
	Added:  6  TX: board.D24  RX: board.D25
	Added:  7  TX: board.D29  RX: board.D28
	Added:  8  TX: board.TX8  RX: board.RX8
```
The above output is nice as It shows me which pins to connect for daisy chain:
D1-D7, D8-D15... Depending on how many I wish to chain to each other...

Fix is simple:
````
diff --git a/ports/mimxrt10xx/boards/teensy40/pins.c b/ports/mimxrt10xx/boards/teensy40/pins.c
index 263603dcf..9223f8b3e 100644
--- a/ports/mimxrt10xx/boards/teensy40/pins.c
+++ b/ports/mimxrt10xx/boards/teensy40/pins.c
@@ -132,7 +132,5 @@ STATIC const mp_rom_map_elem_t board_module_globals_table[] = {
     {MP_OBJ_NEW_QSTR(MP_QSTR_TX6), MP_ROM_PTR(&pin_GPIO_AD_B0_12)},
     {MP_OBJ_NEW_QSTR(MP_QSTR_RX7), MP_ROM_PTR(&pin_GPIO_EMC_32)},
     {MP_OBJ_NEW_QSTR(MP_QSTR_TX7), MP_ROM_PTR(&pin_GPIO_EMC_31)},
-    {MP_OBJ_NEW_QSTR(MP_QSTR_RX8), MP_ROM_PTR(&pin_GPIO_B1_13)},
-    {MP_OBJ_NEW_QSTR(MP_QSTR_TX8), MP_ROM_PTR(&pin_GPIO_B1_12)},
 };
 MP_DEFINE_CONST_DICT(board_module_globals, board_module_globals_table);
```
Should this be it's own issue/PR...

To also make all of the UARTS to work in particular logical Serial2 to work with Teensy 4, 4.1 MicroMod, I needed to disable the 
Serial Port Hook as the current beta/sources blindly mucks with LPUART4.  In this case I did issue a PR:
https://github.com/adafruit/circuitpython/pull/6261
Which is still needing review...   But again would be better to combine some/all of these into 1?

Thanks
Kurt

dhalbert, 2022-04-18:
If the parts of a PR are rellated and should be reviewed together, then they certainly can be in one PR. But if they are unrelated, probably it's better to split up. We don't have a hard and fast rule about dividing up or combining PR's.

I do not understand what you mean about `Serial2`. In CircuitPython, there is no such designation. If you mean that the DEBUG UART permanently uses up a UART, that should not be the case. The UART should be used only if the UART TX and RX pins are defined or some DEBUG UART flag is set.

KurtE, 2022-04-18:
> If the parts of a PR are rellated and should be reviewed together, then they certainly can be in one PR. But if they are unrelated, probably it's better to split up. We don't have a hard and fast rule about dividing up or combining PR's.
> 

Thanks, makes sense.  I have contributed stuff to projects on both ends of this, so just trying to make sure.

> I do not understand what you mean about `Serial2`. In CircuitPython, there is no such designation. 

You are right that Circuit Python does not Mention Serial2, This is the Arduino designation for a Serial port.  For example 
Teensy 4:
[image]
Note the T4 card shows RX2 and TX2, which we have included in the pins.c file to imply UART2, however should be noted, 
that the Serial port number does not actually correspond to the internal Hardware LPUART.  On the T4, T4.1...
Serial1 is LPUART6, Serial2 is LPUART4, ...

>If you mean that the DEBUG UART permanently uses up a UART, that should not be the case. The UART should be used only if the UART TX and RX pins are defined or some DEBUG UART flag is set.

I think it is DEBUG UART?  It is the Serial Port Hook code change that went into the beta code, which includes current code base.
Right now it is setup that everything output to the REPL port is output on LPUART4 and I think input from LPUART4 also goes the other way as well.

Discussed this in issue: https://github.com/adafruit/circuitpython/issues/6241

I do have PR: https://github.com/adafruit/circuitpython/pull/6261
That puts an ifdef within the file that only impacts it on mimxrt... to disable this code unless value set.
 I noticed some of the other ports have it disabled one way or another, including  simply not having the code. 

I explained a few options about this in the PR.

PR #8928 (2024-02-16): Numerous iMX fixes and SDK update
* Fixes #4954. Leaving GPIO in output mode messed with ADC readings.
* Fixes #8018 and fixes #6537. UART rx was more complicated than it needed to be.
* Fixes #5581. uart reset is handled by finalizer. Added i2c_reset() call and reservation check.
closed 2024-02-16 by jepler