adafruit/circuitpython #5586: Add ru-Ru translation option to weblate

Pixel-Clay, 2021-11-15:
As a guy who is interested in electronics and programming, I'd like to participate in CircuitPython translation. I strongly believe that CircuitPython will be the next milestone in the Russian maker sphere. 

With the appearance of rp2040 and raspberry pi pico on the market, AliExpress in particular (this is the primary place where we source our components from) and its' low price tag, I strongly believe that CircuitPython will be the revolution in the Russian maker community. As for now, people use Arduino clones, and there is one company(Amperka.ru) that uses the espruino environment and makes JavaScript-compatible boards. Imagine the possibilities that the ease of Python will bring to the community. More people will get to know the incredible world of electronics and microcontrollers just because you don't need to learn C++, which is way harder than Python!

As for languages, Russian is my native language, as I was risen and live there, so there shouldn't be much trouble with it. English isn't a problem for me at al because my English level is C1 Advanced(European level) and I am currently learning to master the language with the C2 Proficiency level.

tannewt, 2021-11-15:
A Russian translation would be amazing! I've added it to our Weblate, the system we use for translations. You should be able to add translations for it here: https://hosted.weblate.org/translate/circuitpython/main/ru/?q=state:%3Ctranslated

dhalbert, 2021-11-17:
@Pixel-Clay your translation work is now available in the latest builds at https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin. 7.1.0-beta.1 will also include Russian builds.

closed 2021-11-17 by dhalbert