adafruit/circuitpython #5590: Fix Sparkfun MicroMod SAMD51 external SPI flash pin mapping

dhalbert, 2021-11-16:
#5565 notes that there is a design bug on v1.2 of this board: The MOSI and MISO pins to the external flash were switched. Also see sparkfun/MicroMod_Processor_Board-SAMD51#2, which notes they hope to have fixed ones in stock soon. They are going to offer an exchange. When we know the new pin mapping, let's make a new PR that corrects the pin assignments.

dhalbert, 2021-12-09:
It looks like the revised board is imminent, but a new schematic hasn't yet been posted.

Tagging @stonehippo for interest.

stonehippo, 2021-12-09:
@dhalbert thanks. I've checked in with Sparkfun, seems like I should be able to exchange for the new module sometime soon. Between that and a revised schematic, I'll be able to put together and test a PR with the correct pins for the SPI flash.

stonehippo, 2022-02-03:
I just got a ship notification for the replacement MicroMod. I'll post up a PR once I've confirmed that everything works.

dhalbert, 2022-02-03:
> I just got a ship notification for the replacement MicroMod. I'll post up a PR once I've confirmed that everything works.

Thanks! I got a back-in-stock notification, but the documentation hadn't yet been updated.

stonehippo, 2022-02-03:
I noticed that, too. I'm hoping the get the latest schema doc up before it gets here. If not, I'll poke Sparkfun for it.

stonehippo, 2022-02-09:
New 1.3 module is here, but the schematic is still for 1.2. I've requested the updated version.

stonehippo, 2022-02-12:
@dhalbert bad news, I think. It looks like the 1.3 version of the MicroMod SAMD51 is _not_ fixed for external flash. They haven't published the new schematic, so I could have this wrong, but in toning out the module, it looks like the CIPO pad was corrected, but the CS and COPI are still swapped in a way that won't work with SERCOM 2. So it's a 50% fix, which is a 100% fail to get the flash working.

I'll be sure I haven't made a mistake once the new schematic is up, so I will try again. But my guess is that both 1.2 and 1.3 revisions of this board are broken for external flash, so CP will have to use the internal flash on the SAMD51, making this more like a non-Express board. I'm also going to follow up on the Sparkfun forum.

Here's how it's all laying out.

[image]

I'm kinda bummed about this.

stonehippo, 2022-04-16:
@tannewt @dhalbert I'm back to this. It turns out the flash on the revised chip DOES work. This was confirmed with the [Sparkfun Arduino flash library](https://github.com/sparkfun/MicroMod_Processor_Board-SAMD51/issues/2#issuecomment-1096945200), so yay!

So I have tried to get this board working, but I've run into a roadblock, based in part on how much I haven't learned about CP internals yet. One of the quirks of this board is that write-protect (WP) and HOLD pins on the Winbond flash module are connected to GPIO (a lot of designs just tie those pins to VCC, which enables the module). This isn't really a problem, per se, since those pins can just be pulled high to make the module writable. In the Sparkfun lib example, I can do just that and everything works great.

But I'm runnning into an issue with CP. I _thought_ that I could simply pull the pins in question up in the `board_init()` in `board.c`, with this:

```
void board_init(void) {
   never_reset_pin_number(PIN_PB22); // external flash WP
   never_reset_pin_number(PIN_PB23); // external flash HOLD

   // set the WP pin high (could also made an input and pulled up
   gpio_set_pin_function(PIN_PB22, GPIO_PIN_FUNCTION_OFF);
   gpio_set_pin_direction(PIN_PB22, GPIO_DIRECTION_OUT);
   gpio_set_pin_level(PIN_PB22, true);

   // set the HOLD pin high (could also made an input and pulled up
   gpio_set_pin_function(PIN_PB23, GPIO_PIN_FUNCTION_OFF);
   gpio_set_pin_direction(PIN_PB23, GPIO_DIRECTION_OUT);
   gpio_set_pin_level(PIN_PB23, true);
}
```

This doesn't work, in that CP can't find the flash module and so doesn't mount CIRCUITPY. The pins do get set to high, so the code is correct as far as it goes. I tried introducing some delay with `mp_hal_delay_ms()`, too, thinking that the pins just needed to switch on, but no luck.

Just to be sure I haven't got it wrong, I hacked together a "haxpress" with the board and an identical Winbond module that I have, and co-opted the main SPI port for it with a custom build of the SAMD51 MicroMod uf2. This let me test that my config was fine. I was able to get that module to work without issue, as long as I tied the WP/HOLD pins to VCC. As soon as I tried to use another pair of pins configured like the above (I chose PB06/PB07, because they're easy to get to), I'm back to safe mode. The pins are high, but the module doesn't work.

All of this leads me to believe that the external flash is trying to load before `board_init()` is called, and so it doesn't matter what I do there. Do I have that right? And if so, what's the best path to getting those pins pulled high _before_ CP tries to get to the flash module?

stonehippo, 2022-04-17:
Ok, I took a look at `main.c` and I can see that the CIRCUITPY filesystem gets init'ed before `board_init()` is called. So I going to try adding something like this before the filesystem gets set up:

```
    #if CIRCUITPY_EXTERNAL_FLASH_SETUP
    // Set up anything that might need to get done before we try to use SPI flash
    // This is needed for some boards where flash relies on GPIO setup to work
    external_flash_setup();
    #endif
 ```

And then add `void external_flash_setup(void)` as something that can exist either in `supervisor/board.h` and the `board.c` file. That seems like the right place to put it, for boards that might have similar setup, but it also touches the whole board setup, so maybe it's a bit drastic. Thoughts?

stonehippo, 2022-04-17:
OK, that did it! I've _finally_ got CIRCUITPY mounting on this board, and in a way that I think is universal if needed without disrupting every other board config. I'm going to clean this up and submit a PR for review.

dhalbert, 2022-04-18:
> `#if CIRCUITPY_EXTERNAL_FLASH_SETUP`

Instead of creating a new compiler flag, you can have a default routine that does nothing. See examples in `circuitpython/supervisor/shared`, and look for routines with an `MP_WEAK` declaration. These "weak" routines are used only if there is a not a correspond non-weak definition.

stonehippo, 2022-04-18:
Thanks, @dhalbert. If that's preferred over the flag, I'll give it a try and update the PR.

dhalbert, 2022-04-18:
Yes, I'd recommend the weak routine. Thanks!

stonehippo, 2022-04-18:
OK, I've made that change, tested with builds of a few boards (the MicroMod and others) and updated the PR.

closed 2022-04-19 by dhalbert