adafruit/circuitpython #5594: Allow getting clean REPL without running code.py

aivarannamaa, 2021-11-16:
Thonny IDE allows testing local scripts without saving them to the device first -- when the user selects "Run" command, the code will be submitted to the raw (or paste mode) REPL.

Before this it tries to clear the environment (in order to not depend on the state from previous commands). With MicroPython device it simpl orders soft-reboot in the raw mode, which disables running main.py. With CircuitPython this can't be used, so it simply soft-reboots and tries to interrupt code.py as quickly as possible.

CP 7.0 introduced `supervisor.set_next_code_file`, but it looks like this is not helpful here, as `set_next_code_file(None)` simply restores the default behavior.

My request: please consider adding a way for rebooting the VM without executing code.py, for example:

* by introducing `supervisor.reload_to_repl()` 
* or by making `supervisor.set_next_code_file("")` skip running scripts after reload 
* or (preferably) by following MicroPython's policy of not running the main script if Ctrl-D is sent at the raw REPL.

tannewt, 2021-11-17:
I'm fine with changes to raw repl since Thonny is the only user of it that I know of.

This isn't a high priority for us to do though. I bet you can sort it out. We have a single `main.c` so you'll only need to fix it once. The loop that switches between code.py and the repl is here: https://github.com/adafruit/circuitpython/blob/main/main.c#L851-L865

aivarannamaa, 2021-11-18:
Great! I'll take a look.

PR #5602 (2021-11-23): Don't run main script when soft-rebooting in raw mode
Fixes #5594
closed 2021-11-23 by tannewt