adafruit/circuitpython #5600: RP2040 - SDCard w busio.SPI not working

timonsku, 2021-11-18:
### CircuitPython version

```python
CircuitPython 7.1.0-beta.0
```

### Code/REPL

```python
import os
# import audiobusio
import audiopwmio
import audiocore
import audiomixer
# import bitbangio
import busio
import sdcardio
import storage
import board
import array
import time
import math
import sys
# from audiomp3 import MP3Decoder
import analogio
import digitalio

print("Files on SD Card:")

sd = sdcardio.SDCard(busio.SPI(board.GP6,board.GP7,board.GP4), board.GP9)
vfs = storage.VfsFat(sd)
storage.mount(vfs, '/sd')
print(os.listdir('/sd'))
```

### Behavior

```
code.py output:
Files on SD Card:
Traceback (most recent call last):
  File "code.py", line 22, in <module>
OSError: no SD card

Code done running.
```

### Description

On CPY 7.0 this works fine but on 7.1.0 it does not find the SDcard and prints this error message.
Using Raspberry Pi Pico.

jepler, 2021-11-29:
It's working here, feather rp2040, adafruit tft featherwing
```python
Adafruit CircuitPython 7.1.0-beta.0 on 2021-11-29; Adafruit Feather RP2040 with rp2040
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Files on SD Card:
['LOST.DIR', '.android_secure', 'DCIM', 'Android']

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

My test program was
```
import board
import busio 
import sdcardio
import storage
import os

print("Files on SD Card:")

sd = sdcardio.SDCard(busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO), board.D5) 
vfs = storage.VfsFat(sd)
storage.mount(vfs, '/sd')
print(os.listdir('/sd'))
```

timonsku, 2021-11-29:
hm maybe somehow related to the specific SPI bus used?

jepler, 2021-12-01:
If you can, please test with a different SD card too.

timonsku, 2021-12-02:
I don't have the hardware on hand any more but I did use a different SD shortly and that had the same issue.
Given that it worked fine with identical hardware with 7.0 I don't think it was an issue with the SDCard. Downgrading made it work again.
I can reproduce the hardware again is necessary or at least the same setup. Its just a Pico with a SDcard connector hooked up to the pins in the code.

dhalbert, 2021-12-08:
I found some cards that do work with both 7.0.0 and 7.2.0-something, and some that don't. I did a bisect, and the commit that introduces the problem is, f22299d2b374a8150ec63d9338175aa51b593dd6, part of #5510.

dhalbert, 2021-12-08:
Replicated on SAMD51, so not just RP2040.

Using this test program (on Grand Central, use dedicated SD card pins instead):
```py
import board
import busio
import sdcardio
import storage

spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
cs = board.A0

sdcard = sdcardio.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
print("mounted")
```

On a Grand Central M4, I have some cards that work, and three that fail this way (`DEBUG_PRINT` is on in `sdcardio`):
```
dafruit CircuitPython 7.0.0-516-gf22299d2b-dirty on 2021-12-08; Adafruit Grand Central M4 Express with samd51p20
>>> 
>>> import sd
cmd   0 [00] arg=          0 [00000000] len=0 data wait
cmd   0 [00] arg=          0 [00000000] len=0 data wait
cmd   0 [00] arg=          0 [00000000] len=0 data wait
cmd   0 [00] arg=          0 [00000000] len=0 data wait
cmd   0 [00] arg=          0 [00000000] len=0 data wait
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "sd.py", line 9, in <module>
OSError: no SD card
```
I tested the same three cards on a Metro M4 with this SD card breakout: https://www.adafruit.com/product/4682, using 6-inch jumpers. Two of them fail, but in a different way than above:
```
>>> import sd
cmd   0 [00] arg=          0 [00000000] len=0 data wait
cmd   8 [08] arg=        426 [000001aa] len=4 wait
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "sd.py", line 9, in <module>
OSError: couldn't determine SD card version
```
One that failed on the GC4, works on the MM4 with breakout:
```
Adafruit CircuitPython 7.0.0-516-gf22299d2b-dirty on 2021-12-08; Adafruit Metro M4 Express with samd51j19
>>> 
>>> import sd
cmd   0 [00] arg=          0 [00000000] len=0 data wait
cmd   8 [08] arg=        426 [000001aa] len=4 wait
cmd  58 [3a] arg=          0 [00000000] len=4 wait
cmd  55 [37] arg=          0 [00000000] len=0 data wait
cmd  41 [29] arg= 1073741824 [40000000] len=0 data wait
cmd  58 [3a] arg=          0 [00000000] len=4 wait
cmd  55 [37] arg=          0 [00000000] len=0 data wait
cmd  41 [29] arg= 1073741824 [40000000] len=0 data wait
cmd  58 [3a] arg=          0 [00000000] len=4 wait
cmd  55 [37] arg=          0 [00000000] len=0 data wait
cmd  41 [29] arg= 1073741824 [40000000] len=0 data wait
cmd  58 [3a] arg=          0 [00000000] len=4 wait
cmd  55 [37] arg=          0 [00000000] len=0 data wait
cmd  41 [29] arg= 1073741824 [40000000] len=0 data wait
cmd  58 [3a] arg=          0 [00000000] len=4 wait
cmd  55 [37] arg=          0 [00000000] len=0 data wait
cmd  41 [29] arg= 1073741824 [40000000] len=0 data wait
cmd  58 [3a] arg=          0 [00000000] len=4 wait
cmd  55 [37] arg=          0 [00000000] len=0 data wait
cmd  41 [29] arg= 1073741824 [40000000] len=0 data wait
cmd  58 [3a] arg=          0 [00000000] len=4 wait
cmd   9 [09] arg=          0 [00000000] len=16 data wait
cmd  16 [10] arg=        512 [00000200] len=0 data wait
cmd  17 [11] arg=          0 [00000000] len=512 data wait
cmd  17 [11] arg=       8192 [00002000] len=512 data wait
mounted
```

jepler, 2021-12-08:
I do see that the CMD25 work could have changed how card init functions, by making a failure to "reach the idle state" stop a command from being issued (and an error instead).  I think that in normal operation it should be an error if the card fails to reach an idle state when expected, but this is apparently an unfair expectation when switching the card to SPI mode.  This patch aims to fix it, but I don't have any affected hardware to test...

https://gist.github.com/jepler/8e5c829cfad6b2eadbf681dd00530ebd

closed 2021-12-09 by jepler

strau0106, 2023-03-03:
Are there any updates on this issue, because I think I am experiencing the same issues. 

my hw: 

KB2040
Adafruit SPI/SDIO microsdcard breakout
Sandisk SDHC 16GB 

I have recompiled cpy to include the debug flag in sdcardio: producing the follwoing output: 
```Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
cmd   0 [00] arg=          0 [00000000] len=0 data
cmd   0 [00] arg=          0 [00000000] len=0 data
cmd   0 [00] arg=          0 [00000000] len=0 data
cmd   0 [00] arg=          0 [00000000] len=0 data
cmd   0 [00] arg=          0 [00000000] len=0 data
Traceback (most recent call last):
  File "code.py", line 8, in <module>
OSError: no SD card```

With the following code: ```py
import board
import busio
import sdcardio
import storage

spi = board.SPI()
cs = board.D6
sdcard = sdcardio.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
with open("/sd/test.txt", "w") as f:
    f.write("Hello World!\n")```

I am pretty sure it's not a hardware issue, I have checked my wiring thrice and the sd card works on the computer. (not on macos but just on windows, but that could also be a dongle problem.)

thx!

PR #5688 (2021-12-09): sdcardio: Don't require reaching idle state during early init
.. which is what the addition of cmd25 made happen. We still signal an error when failing to reach the idle state at any other time, which _is_ different than adafruit_sdcard but I think that it is correct.

This fixes #5600 according to Dan's testing on a GCM4 on the internal SD card slot. There are still some ambiguous results on the MM4 with external SD card slot on 6" jumper wires.
commit 11c573ef (2021-12-09): sdcardio: Don't require reaching idle state during early init

.. which is what the addition of cmd25 made happen. We still signal
an error when failing to reach the idle state at any other time,
which _is_ different than adafruit_sdcard but I think that it is
correct.

This fixed #5600 according to Dan's testing on a GCM4 on the internal
SD card slot. There are still some ambiguous results on the MM4 with
external SD card slot on 6" jumper wires.