adafruit/circuitpython #5609: ulab bool array sum produce wrong result

mrdalgaard, 2021-11-22:
### CircuitPython version

```python
Adafruit CircuitPython 7.1.0-beta.0-75-g932131b4f on 2021-11-18; Adafruit Feather Bluefruit Sense with nRF52840
```

### Code/REPL

```python
>>> ulab.numpy.sum(ulab.numpy.array([False, False, True], dtype=ulab.numpy.bool))
1
>>> ulab.numpy.sum(ulab.numpy.array([True, False, True], dtype=ulab.numpy.bool))
2
>>> ulab.numpy.sum(ulab.numpy.array([True, False, False], dtype=ulab.numpy.bool))
0
>>> ulab.numpy.sum(ulab.numpy.array([True, True, False], dtype=ulab.numpy.bool))
1
```

### Behavior

The 0 was not expected as there is one True in the array.

### Description

Produces correct result if dtype=bool is not used. (default int32 array works)
Produces correct result if sum() is used (non ulab) on same array
Error reproduced on M4 as well.
Desktop numpy produces correct result (not sure of version)

jepler, 2021-11-30:
Thanks for your report. I've reported it upstream at https://github.com/v923z/micropython-ulab/issues/457. Once upstream fixes the bug, we can incorporate the new version into a future CircuitPython release.

As you've identified, using the non-ulab `sum()` function in this case can get you an accurate result.

v923z, 2021-11-30:
@mrdalgaard @jepler Thanks for taking the issue upstream! Based on Jeff's comment, I have updated the code, added a small test script, and generated a new tag, https://github.com/v923z/micropython-ulab/releases/tag/3.3.7

v923z, 2021-11-30:
@tannewt Scott, is it possible to set up github in such a way that whenever you add the `ulab` tag, I get a notification? I would have fixed the glitch much earlier, but I took note of it only, when Jeff specifically opened the issue over there.

closed 2021-12-01 by tannewt

tannewt, 2021-12-01:
@v923z I don't know of one but I can try to remember to assign them to you or mention you in a comment.

v923z, 2021-12-01:
> @v923z I don't know of one but I can try to remember to assign them to you or mention you in a comment.

@tannewt Assigment is OK. However, there is this https://github.com/marketplace/actions/issue-label-notifier It would be awesome, if you could set this up. Such a thing might be useful for others on your team.

PR #5642 (2021-12-01): Update ulab to 3.3.7
Closes: #5609
commit 1b87d2bc (2021-12-01): Update ulab to 3.3.7

Closes: #5609