adafruit/circuitpython #5612: CP 7.1.0 RP2040 Starts automatically after Ctrl-C

b-blake, 2021-11-23:
### CircuitPython version

```python
Adafruit CircuitPython 7.1.0-beta.0-8-g28b5733bd on 2021-11-12; Adafruit Feather RP2040 with rp2040
Board ID:adafruit_feather_rp2040
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT
# adafruit_circuitpython_adafruitio usage with an esp32spi_socket
import board
import busio
import displayio
from   digitalio import DigitalInOut, Direction, Pull
import gc
import microcontroller
import neopixel
from   random import randint
import storage
import terminalio
import time
import supervisor

from   adafruit_bme280 import basic as adafruit_bme280
import adafruit_bus_device
from   adafruit_datetime import date
import adafruit_ds18x20
import adafruit_ds3231
import adafruit_esp32spi.adafruit_esp32spi_socket as socket
from   adafruit_esp32spi import adafruit_esp32spi
from   adafruit_esp32spi import PWMOut
from   adafruit_io.adafruit_io import IO_HTTP, AdafruitIO_RequestError
from   adafruit_ntp import NTP
from   adafruit_onewire.bus import OneWireBus
import adafruit_requests as requests
import adafruit_rgbled
import adafruit_sdcard

try:
    from secrets import secrets
except ImportError:
    print("WiFi secrets are kept in secrets.py, please add them there!")
    raise

print("\fProgram Start")
print(secrets["location"], secrets["zip"])
# SD Card or Flash Memory
# If the switch pin is connected to ground CircuitPython can write to the root drive
switch           = DigitalInOut(board.D9)
switch.direction = Direction.INPUT
switch.pull      = Pull.UP
print("Write to SD Card:", switch.value)
print("Write to Flash  :", (not switch.value))
# Variable Definitions
Show_Board_Pin_Assignments = False
AUTOREL      = True        # True=enable_autoreload(), False=disable_autoreload()
SD_CARD      = True        # Should we look for an SD Card?
SET_RTC      = True        # Should we set the SYS clock at boot and each hour?
HOUR         = 60*60       # Minutes*Seconds = 3600 = 1 Hour
LOCAL        = (-5*HOUR)   # -4=Summer EDT -14400, -5=Winter EST -18000
DAY_OFFSET   = 1           # Day of Week Offset
Show_Weather = True        # Test Internet connection
Show_Colors  = False       # Show Neo Pixel LED color tests
Show_BME280  = True        # 
Refresh_Time = None        # prime the RTC reset code to set at boot/restart
OLED         = True        # Do we want to use the OLED display?
OLED_ADDRESS = 0x3C        # Address of OLED Display
[... 546 lines cut ...]
#    print("Auto-Reload Enabled.")
    if(OLED or BOXES): print(gc.collect(), gc.mem_free())
    Waiting(15, 1)
    if(AUTOREL): supervisor.disable_autoreload()
#    print("Auto-Reload Disabled.")
    NeoPix[0] = ((7,7,7))
    AirLift_GRB.color = AirLift_BLACK
    NeoPix_EXTRA.fill(BLACK)
    print("")
```

### Behavior

code.py starts running by itself while REPL is on the terminal screen with >>> being displayed.  This takes about 25 seconds to happen.  Keyboard untouched.

### Description

If I write edited code.py back to my RP2040 while supervisor.disable_autoreload() is in effect,
and then I go to the REPL screen, hit Ctrl-C while supervisor.enable_autoreload() is in effect and review data.
If I hit Ctrl-D within ~25 seconds, code.py starts normally.
If I wait more than ~25 seconds code.py starts without my request.
I presume the code s   Iwrite I did dwhile  supervisor.disable_autoreload() was in effect effects the start of code.py.
My code runs in disable mode for 2 to 5 seconds then runs in enable mode for 15 seconds..

Edditing issues with input ...

b-blake, 2021-11-23:
Please excuse typos in Description.  Edit location and cursor location got out of sync for some reason.  Last two lines should read:

I presume the code I write while supervisor.disable_autoreload() was in effect, causes the start of code.py.
My code runs in disable mode for 2 to 5 seconds then runs in enable mode for 15 seconds..

jepler, 2021-11-23:
You can use the edit button (inside the 3 dots menu) to edit your comment at any time to fix typos.

b-blake, 2021-11-23:
> You can use the edit button (inside the 3 dots menu) to edit your comment at any time to fix typos.

Thank you!

b-blake, 2021-11-24:
This simple code shows the issue on my machine.  The wait time to auto start varies.

Resave the code.py during **disable**
Hit Ctrl-C during **enable**

import time
import supervisor

while True:
    supervisor.disable_autoreload()
    print("Disable for 5 Seconds")
    time.sleep(5)
    supervisor.enable_autoreload()
    print("Enable for 10 Seconds")
    time.sleep(10)

Bruce

dhalbert, 2022-05-14:
Needs retest after recent rework of autoreload logic.

b-blake, 2022-05-14:
Dan,

Unable to reproduce problem.  It looks like you fixed it.

Bruce

dhalbert, 2022-05-14:
Thanks for testing!

closed 2022-05-14 by dhalbert