adafruit/circuitpython #5614: Adafruit Feather ESP32S2 I2C - OSError: Unsupported Operation

dcalvo, 2021-11-24:
### CircuitPython version

```python
Adafruit CircuitPython 7.1.0-beta.0-91-gaf2ac0aa6 on 2021-11-23; Adafruit Feather ESP32S2 with ESP32S2
```

### Code/REPL

```python
import time
import board
import digitalio
import adafruit_icm20x

ip = digitalio.DigitalInOut(board.I2C_POWER_INVERTED)
ip.switch_to_output()
ip.value = True

i2c = board.I2C()
icm = adafruit_icm20x.ICM20649(i2c)

while True:
    print("Acceleration: X:%.2f, Y: %.2f, Z: %.2f m/s^2" % (icm.acceleration))
    print("Gyro X:%.2f, Y: %.2f, Z: %.2f rads/s" % (icm.gyro))
    print("")
    time.sleep(0.5)
```

### Behavior

```python
code.py output:
Traceback (most recent call last):
  File "code.py", line 11, in <module>
  File "/lib/adafruit_icm20x.py", line 569, in __init__
  File "/lib/adafruit_icm20x.py", line 229, in __init__
  File "/lib/adafruit_icm20x.py", line 235, in initialize
  File "/lib/adafruit_icm20x.py", line 311, in accelerometer_range
  File "/lib/adafruit_register/i2c_bits.py", line 73, in __set__
OSError: [Errno 19] Unsupported operation

Code done running.
```
When `I2C_POWER_INVERTED` is `True`, I2C communication throws the OSError. If `I2C_POWERED_INVERTED` is `False`, the ICM20649 module works as intended.

### Description

After some digging in the discord and issues here, it seems like this issue was discussed and (partially?) resolved in #5595. I'm not sure if it's intended behavior for the pin to have to be set false in order to use I2C, but I figure I would open an issue in case it helps debug the problem further or if anyone else stumbles on this issue in future.

CaribouJohn, 2021-11-24:
I am getting an issue after my script runs for some period of time - fairly arbitrary time period 

from adafruit_bme280 import basic as adafruit_bme280
i2c = board.I2C()  # uses board.SCL and board.SDA
bme280 = adafruit_bme280.Adafruit_BME280_I2C(i2c)

Traceback (most recent call last):
  File "code.py", line 123, in <module>
  File "adafruit_bme280/basic.py", line 200, in temperature
  File "adafruit_bme280/basic.py", line 115, in _read_temperature
  File "adafruit_bme280/basic.py", line 147, in _get_status
  File "adafruit_bme280/basic.py", line 301, in _read_byte
  File "adafruit_bme280/basic.py", line 375, in _read_register
OSError: [Errno 19] Unsupported operation

Is this related (same general error?)

ladyada, 2021-11-25:
make sure you are setting pin 7 to be output and LOW to enable the i2c power!

https://learn.adafruit.com/adafruit-esp32-s2-feather/pinouts#bme280-temperature-humidity-and-pressure-sensor-3106370-11

CaribouJohn, 2021-11-30:
> make sure you are setting pin 7 to be output and LOW to enable the i2c power!
> 
> https://learn.adafruit.com/adafruit-esp32-s2-feather/pinouts#bme280-temperature-humidity-and-pressure-sensor-3106370-11

Many thanks :) I suspected it was me

closed 2021-11-30 by ladyada