adafruit/circuitpython #5619: Large Broadcom Ports Folder

matthewjerome, 2021-11-26:
### CircuitPython version

```python
When running `git submodule update --init`, the size is >8gb just for `circuitpython/ports/broadcom/firmware`

Is there a way to target a different repo or branch when building?
```

### Code/REPL

```python
git submodule update --init
```

### Behavior

long module sync time, local dev image size required

matthewjerome, 2021-11-26:
Build location has 20gb capacity :
[image]

rpavlik, 2021-11-27:
Probably the best approach is to just avoid initializing those large submodules that you don't need. I believe there may also be a submodule deinit command.

matthewjerome, 2021-11-28:
Just ran `git submodule deinit "ports/broadcom/peripherals"` and `git submodule deinit "ports/broadcom/firmware"` which show the messages `Cleared directory 'ports/broadcom/firmware'` and `Cleared directory 'ports/broadcom/peripherals'`. After running `git submodule update --init`, the request for `broadcom` still happens. Tried with a fresh copy and then also removed references to `broadcom` in `.gitmodules` but then the error appears `fatal: No url found for submodule path 'ports/broadcom/firmware' in .gitmodules`.

rpavlik, 2021-11-29:
Don't pass --init to the submodules update command, that initializes all currently used submodules, including the ones you just de-initted

jepler, 2021-11-29:
I agree it would be nice to find a way so that the majority of developers, who do no build the broadcom port, can avoid this extremely time (during the initial clone / submodule init) and disk space intensive submodule (forever). If this means changing our documentation to not recommend `git submodule update --init`, so be it, but the 'correct' alternative will need to be short and easy to remember!

I investigated whether specifying that the submodule was `shallow` helped, but it appears not to. This included if I created a branch or a tag that pointed exactly at the desired ref, in a fork of rpi-firmware (with the submodule url changed to point to my fork).

These files are not "source files" in any traditional sense, and keeping their history is not useful. I wonder if Adafruit should create a fresh repo with the content of rpi-firmware but not the history, and point to this in gitmodules.

rpavlik, 2021-11-29:
oh, it's the rpi stuff that's causing the issue. I was wondering what the broadcom ports folder was... Yeah, could do a fresh repo, or could use FetchContent.cmake to just download the files required from rpi-firmware.  https://cmake.org/cmake/help/v3.13/module/FetchContent.html

dhalbert, 2021-11-29:
I did `git submodule update --init --depth 1` and that was much faster, though not instantaneous. We could recommend that, maybe?

dhalbert, 2021-11-29:
> I did `git submodule update --init --depth 1` and that was much faster, though not instantaneous. We could recommend that, maybe?

@tannewt points out `--depth` 1` may not include the commit you want.

tannewt, 2021-11-30:
Sorry about this folks. I'm a bit spoiled with my gigabit internet.

I've made a PR #5627 which will prevent the large download by setting `shallow = true`. This will fail, however, because the latest commit isn't the one we use. So, I added `make fetch-submodules` for the top level that does the specific fetch trick that the CI does. It should be used for updating instead of `git submodule update` directly.

RetiredWizard, 2021-11-30:
Thanks Scott! It wasn't only the speed, I was pulling my hair out trying to free up space on my little build PIs, got up to 15G but still wasn't enough.

Anyway, I'm really clueless when it comes to make so I"m sure I'm doing something stupid. This is what I get when I try the make fetch-submodules:

>  make fetch-submodules BOARD=adafruit_feather_esp32s2
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
find: ‘../../extmod/ulab/code’: No such file or directory
make: *** No rule to make target 'fetch-submodules'.  Stop.

closed 2021-11-30 by dhalbert

matthewjerome, 2021-11-30:
Just tried to build again - the build process is different but still has a huge size associated with it. Seems to fail around `ports/broadcom/firmware`. Take a look at the terminal output :

```
$ git submodule sync --quiet --recursive
$ git submodule update --init

Submodule 'data/nvm.toml' (https://github.com/adafruit/nvm.toml.git) registered for path 'data/nvm.toml'
Submodule 'extmod/ulab' (https://github.com/v923z/micropython-ulab) registered for path 'extmod/ulab'
Submodule 'frozen/Adafruit_CircuitPython_APDS9960' (https://github.com/adafruit/Adafruit_CircuitPython_APDS9960) registered for path 'frozen/Adafruit_CircuitPython_APDS9960'
Submodule 'frozen/Adafruit_CircuitPython_BLE' (https://github.com/adafruit/Adafruit_CircuitPython_BLE) registered for path 'frozen/Adafruit_CircuitPython_BLE'
Submodule 'frozen/Adafruit_CircuitPython_BLE_Apple_Notification_Center' (https://github.com/adafruit/Adafruit_CircuitPython_BLE_Apple_Notification_Center) registered for path 'frozen/Adafruit_CircuitPython_BLE_Apple_Notification_Center'
Submodule 'frozen/Adafruit_CircuitPython_BusDevice' (https://github.com/adafruit/Adafruit_CircuitPython_BusDevice.git) registered for path 'frozen/Adafruit_CircuitPython_BusDevice'
Submodule 'frozen/Adafruit_CircuitPython_CircuitPlayground' (https://github.com/adafruit/Adafruit_CircuitPython_CircuitPlayground.git) registered for path 'frozen/Adafruit_CircuitPython_CircuitPlayground'
Submodule 'frozen/Adafruit_CircuitPython_Crickit' (https://github.com/adafruit/Adafruit_CircuitPython_Crickit) registered for path 'frozen/Adafruit_CircuitPython_Crickit'
Submodule 'frozen/Adafruit_CircuitPython_DRV2605' (https://github.com/adafruit/Adafruit_CircuitPython_DRV2605) registered for path 'frozen/Adafruit_CircuitPython_DRV2605'
Submodule 'frozen/Adafruit_CircuitPython_DS3231' (https://github.com/adafruit/Adafruit_CircuitPython_DS3231) registered for path 'frozen/Adafruit_CircuitPython_DS3231'
Submodule 'frozen/Adafruit_CircuitPython_Display_Shapes' (https://github.com/adafruit/Adafruit_CircuitPython_Display_Shapes) registered for path 'frozen/Adafruit_CircuitPython_Display_Shapes'
Submodule 'frozen/Adafruit_CircuitPython_Display_Text' (https://github.com/adafruit/Adafruit_CircuitPython_Display_Text) registered for path 'frozen/Adafruit_CircuitPython_Display_Text'
Submodule 'frozen/Adafruit_CircuitPython_DotStar' (https://github.com/adafruit/Adafruit_CircuitPython_DotStar.git) registered for path 'frozen/Adafruit_CircuitPython_DotStar'
Submodule 'frozen/Adafruit_CircuitPython_ESP32SPI' (https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI) registered for path 'frozen/Adafruit_CircuitPython_ESP32SPI'
Submodule 'frozen/Adafruit_CircuitPython_FocalTouch' (https://github.com/adafruit/Adafruit_CircuitPython_FocalTouch) registered for path 'frozen/Adafruit_CircuitPython_FocalTouch'
Submodule 'frozen/Adafruit_CircuitPython_HID' (https://github.com/adafruit/Adafruit_CircuitPython_HID.git) registered for path 'frozen/Adafruit_CircuitPython_HID'
Submodule 'frozen/Adafruit_CircuitPython_IRRemote' (https://github.com/adafruit/Adafruit_CircuitPython_IRRemote.git) registered for path 'frozen/Adafruit_CircuitPython_IRRemote'
Submodule 'frozen/Adafruit_CircuitPython_LC709203F' (https://github.com/adafruit/Adafruit_CircuitPython_LC709203F) registered for path 'frozen/Adafruit_CircuitPython_LC709203F'
Submodule 'frozen/Adafruit_CircuitPython_LIS3DH' (https://github.com/adafruit/Adafruit_CircuitPython_LIS3DH.git) registered for path 'frozen/Adafruit_CircuitPython_LIS3DH'
Submodule 'frozen/Adafruit_CircuitPython_LSM6DS' (https://github.com/adafruit/Adafruit_CircuitPython_LSM6DS) registered for path 'frozen/Adafruit_CircuitPython_LSM6DS'
Submodule 'frozen/Adafruit_CircuitPython_MIDI' (https://github.com/adafruit/Adafruit_CircuitPython_MIDI) registered for path 'frozen/Adafruit_CircuitPython_MIDI'
Submodule 'ports/atmel-samd/Adafruit_CircuitPython_Motor' (https://github.com/adafruit/Adafruit_CircuitPython_Motor.git) registered for path 'frozen/Adafruit_CircuitPython_Motor'
Submodule 'atmel-samd/frozen/Adafruit_CircuitPython_NeoPixel' (https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel) registered for path 'frozen/Adafruit_CircuitPython_NeoPixel'
Submodule 'frozen/Adafruit_CircuitPython_ProgressBar' (https://github.com/adafruit/Adafruit_CircuitPython_ProgressBar) registered for path 'frozen/Adafruit_CircuitPython_ProgressBar'
Submodule 'frozen/Adafruit_CircuitPython_RFM69' (https://github.com/adafruit/Adafruit_CircuitPython_RFM69.git) registered for path 'frozen/Adafruit_CircuitPython_RFM69'
Submodule 'frozen/Adafruit_CircuitPython_RFM9x' (https://github.com/adafruit/Adafruit_CircuitPython_RFM9x.git) registered for path 'frozen/Adafruit_CircuitPython_RFM9x'
Submodule 'frozen/Adafruit_CircuitPython_Register' (https://github.com/adafruit/Adafruit_CircuitPython_Register.git) registered for path 'frozen/Adafruit_CircuitPython_Register'
Submodule 'frozen/Adafruit_CircuitPython_Requests' (https://github.com/adafruit/Adafruit_CircuitPython_Requests) registered for path 'frozen/Adafruit_CircuitPython_Requests'
Submodule 'frozen/Adafruit_CircuitPython_SD' (https://github.com/adafruit/Adafruit_CircuitPython_SD.git) registered for path 'frozen/Adafruit_CircuitPython_SD'
Submodule 'frozen/Adafruit_CircuitPython_ST7789' (https://github.com/adafruit/Adafruit_CircuitPython_ST7789) registered for path 'frozen/Adafruit_CircuitPython_ST7789'
[... cut ...]
Submodule path 'ports/atmel-samd/freetouch': checked out 'b6859a349efb79e2008d6f6a34e406e93a8e19c0'
Submodule path 'ports/atmel-samd/peripherals': checked out 'd3b20192cf94fdea68a412596e082108ba5ebbf0'
remote: Enumerating objects: 1783364, done.
remote: Counting objects: 100% (1783361/1783361), done.
remote: Compressing objects: 100% (196993/196993), done.
fatal: write error: No space left on device8.22 GiB | 10.20 MiB/s
Fetched in submodule path 'ports/broadcom/firmware', but it did not contain 6a5207946edcd45813d1dd1572ca8bd8101b68b6. Direct fetching of that commit failed.
```

dhalbert, 2021-11-30:
Do not do this anymore:
```sh
$ git submodule sync --quiet --recursive
$ git submodule update --init
```
Instead, do this:
```sh
$ cd circuitpython     # go to top level
$ make fetch-submodules
```
That does something trickier, which fetches only as much of a submodule as needed. This was added in #5627, which closed this issue.

The https://learn.adafruit.com/building-circuitpython guide has been updated to reflect this.

jerryneedell, 2021-11-30:
After doing a clean clone and the `make fetch-submodules` I get this with `git diff`
Did  miss something?

```
jerryneedell@Mac-mini circuitpython % git diff
diff --git a/ports/broadcom/firmware b/ports/broadcom/firmware
--- a/ports/broadcom/firmware
+++ b/ports/broadcom/firmware
@@ -1 +1 @@
-Subproject commit 6a5207946edcd45813d1dd1572ca8bd8101b68b6
+Subproject commit 6a5207946edcd45813d1dd1572ca8bd8101b68b6-dirty
jerryneedell@Mac-mini circuitpython % 
```

jerryneedell, 2021-11-30:
Hmmm -- I reran this on my linux box and did not see the diff ....
never-mind...

jerryneedell, 2021-11-30:
very odd -- this is reproducible on the Mac, but non under Linux (Ubuntu 20.04)

Does anyone know how to fix it on the Mac?

tannewt, 2021-11-30:
@jerryneedell what does `git status` show in the firmware directory and at the top level?

dhalbert, 2021-11-30:
Also what are the versions of `git` you are using?

jerryneedell, 2021-11-30:
```
jerryneedell@Mac-mini firmware % git status
HEAD detached at 6a52079
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   modules/5.10.81+/kernel/net/netfilter/xt_DSCP.ko
	modified:   modules/5.10.81+/kernel/net/netfilter/xt_HL.ko
	modified:   modules/5.10.81+/kernel/net/netfilter/xt_RATEEST.ko
	modified:   modules/5.10.81+/kernel/net/netfilter/xt_TCPMSS.ko
	modified:   modules/5.10.81-v7+/kernel/net/netfilter/xt_DSCP.ko
	modified:   modules/5.10.81-v7+/kernel/net/netfilter/xt_HL.ko
	modified:   modules/5.10.81-v7+/kernel/net/netfilter/xt_RATEEST.ko
	modified:   modules/5.10.81-v7+/kernel/net/netfilter/xt_TCPMSS.ko
	modified:   modules/5.10.81-v7l+/kernel/net/netfilter/xt_DSCP.ko
	modified:   modules/5.10.81-v7l+/kernel/net/netfilter/xt_HL.ko
	modified:   modules/5.10.81-v7l+/kernel/net/netfilter/xt_RATEEST.ko
	modified:   modules/5.10.81-v7l+/kernel/net/netfilter/xt_TCPMSS.ko
	modified:   modules/5.10.81-v8+/kernel/net/netfilter/xt_DSCP.ko
	modified:   modules/5.10.81-v8+/kernel/net/netfilter/xt_HL.ko
	modified:   modules/5.10.81-v8+/kernel/net/netfilter/xt_RATEEST.ko
	modified:   modules/5.10.81-v8+/kernel/net/netfilter/xt_TCPMSS.ko

no changes added to commit (use "git add" and/or "git commit -a")
jerryneedell@Mac-mini firmware % 
```

jerryneedell, 2021-11-30:
jerryneedell@Mac-mini firmware % git --version
git version 2.31.1
jerryneedell@Mac-mini firmware %

jerryneedell, 2021-11-30:
that was on Mac

on Linux ```jerryneedell@jerryneedell-ubuntu-macmini:~/projects/circuitpython/ports/broadcom$ cd firmware/
jerryneedell@jerryneedell-ubuntu-macmini:~/projects/circuitpython/ports/broadcom/firmware$ git status
HEAD detached at 6a52079
nothing to commit, working tree clean
jerryneedell@jerryneedell-ubuntu-macmini:~/projects/circuitpython/ports/broadcom/firmware$ git --version
git version 2.25.1
```

tannewt, 2021-11-30:
I wonder if we need to do some sort of clean after checking out the correct commit. Files from the missing commit may not be deleted on checkout of a different commit.

matthewjerome, 2021-12-03:
using an older version of git

```
$ which git
/usr/local/bin/git

$ git --version
git version 2.23.0
```

PR #5627 (2021-11-30): Add `make fetch-submodules` to update submodules
This also sets rpi-firmware as shallow in case submodules are updated manually.

Fixes #5619
commit 0a27f7c3 (2021-11-30): Add `make fetch-submodules` to update submodules

This also sets rpi-firmware as shallow in case submodules are
updated manually.

Fixes #5619