adafruit/circuitpython #5625: Sony Spresense CircuitPython SPK install not working

apws, 2021-11-29:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0 on Sony Spresense
```

### Code/REPL

```python
N/A - issues with initial firmware setup
```

### Behavior

N/A

### Description

hi, I described my issues first here, as Sony has bugs in his getting started:
https://forum.developer.sony.com/topic/523/circuitpython-install-issues

but after flashing, I am unable to work with the thing - Spresense inserted in extension board exposes CIRCUITPY on extension board USB only(?), its very unreliable and I see no serial console nowhere...

(I know that this port is probably not so much hype anymore, but I wanted to try it here also, as I am testing more circuitpython hw now)

dhalbert, 2021-11-29:
@kamtom48

dhalbert, 2021-11-29:
@kamtom480

apws, 2021-11-29:
weird, I tried it now connected to different computer, may be using different CP210x driver too and at is *better*, CP is available on the extension board usb, but it still disconnects the CIRCUITPY after a while - this is Win11 machine, no antivirus (I have kaspersky on previous one, but NO issues with CP till now on any board - RP2040, ESP32s2, SAMD21...) and I used different usb cable too... now  I am trying bootloader 202 + CP as in sony doc

apws, 2021-11-29:
no - usb device not recognized still

Device USB\VID_0000&PID_0002\6&3ae63acd&0&2 was configured.

Driver Name: usb.inf
Class Guid: {36fc9e60-c465-11cf-8056-444553540000}
Driver Date: 06/21/2006
Driver Version: 10.0.22000.1
Driver Provider: Microsoft
Driver Section: BADDEVICE.Dev.NT
Driver Rank: 0xFF0000
Matching Device Id: USB\DEVICE_DESCRIPTOR_FAILURE
Outranked Drivers: usb.inf:USB\DEVICE_DESCRIPTOR_FAILURE:00FF2000
Device Updated: false
Parent Device: USB\ROOT_HUB30\5&1a805c57&0&0

apws, 2021-11-30:
so, final results of testing:
I tried also recoverytool to clean main board and then tested both bootloader 2.3.0 +CP7 then (clean again) and 2.0.2 +CP7
After such clean flash, it works for a while, MuEditor connects to CIRCUITPY and console, but when I try test script (attached), it probably stops working and windows reports after minute or so - usb device not recognized...

apws, 2021-11-30:
so, final results of testing:
I tried also recoverytool to clean main board and then tested both bootloader 2.3.0 +CP7 then (clean again) and 2.0.2 +CP7
After such clean flash, it works for a while, MuEditor connects to CIRCUITPY and console, but when I try test script, it probably stops working and windows reports after minute or so - usb device not recognized...

[pin-mappings-speedtest.txt](https://github.com/adafruit/circuitpython/files/7622160/pin-mappings-speedtest.txt)

kamtom480, 2021-11-30:
Thanks for finding this. It seems the problem is with the `time.time()` function. This was causing your problems on Spresense. I made a PR to fix it.

closed 2021-11-30 by dhalbert