adafruit/circuitpython #5628: Zero 2W hangs sporadically when using USB mass storage

tannewt, 2021-11-30:
The Zero 2W hangs sometimes when accessing the CIRCUITPY drive. The backtrace ends in the irq handlers-esq thing. It's not clear how it's getting there.

I did some pin based tracing a while back and it seemed to occur after a USB interrupt but not during it.

RetiredWizard, 2021-12-26:
I've been seeing this with my builds from 7.1.0-beta.3-174. Just adding my observations in case they provide any clues.

I've been tracking a memory leak in my python application which involves running the app for several hours. My code is constantly reading from flash and I've run several times for 2 or 3 hours when connected via UART without hanging.

When I've tried to run the same tests connected via the serial USB connector the PI Zero 2W has suddenly frozen at seeming random points.

RetiredWizard, 2022-01-12:
This seems to be happening faster on the latest build

          Adafruit CircuitPython 7.2.0-alpha.1-74-g9c6e34df0-dirty on 2022-01-11; Raspberry Pi Zero 2W with bcm2837

I've tried 6 or 7 boots now and when connecting via the serial USB connector I haven't gone more than 10 minutes without freezing and most times the serial interface hung in less than 5 minutes.

I tried a short run connected via the UART pins and that seemed to be working fine but I didn't let that test run more than five or ten minutes.

I just reflashed the 7.1.0-beta.3-174 version and it's been running 20 minutes so far.

honzasram, 2022-05-08:
I've met same issue on latest release (7.2.5). Is there any other idea what to do except use RP2040 instead of Zero 2W?

dhalbert, 2022-05-08:
The `broadcom` builds, including Zero 2W, are at a very, very early development stage; there's a lot more debugging to do.