adafruit/circuitpython #5630: Support USB host

tannewt, 2021-11-30:
Supporting USB host of keyboards would allow us to have a self-contained workflow on the Pi 400.

hathach, 2021-11-30:
I will try to work on the dwc2 host driver for tinyusb. It is only 2.0 comparing to xhci though it will be much simpler/easier to add support. It will also enable host support on other ports such esp32sx and stm32 as well.

tannewt, 2021-11-30:
@hathach The keyboard on the Pi 400 is connected through the xhci host (not dwc2) so dwc2 host support won't work for it. It'd be good to have on the other ports though.

hathach, 2021-12-01:
ah you are right, just run lsusb on pi400. keyboard is `holtek` device and attached to xhci roothub. I will try to pull out the xhci specs to see if it is similar to ehci.

tannewt, 2021-12-01:
This video was a pretty good overview: https://www.youtube.com/watch?v=SSrFv4a-zgU

ATMakersBill, 2022-02-06:
USB Host support (on any board) would be hugely helpful to our community.  I've seen this solution https://github.com/sekigon-gonnoc/Pico-PIO-USB and would love to see something like that make it into CP (even if only on the RP2040) at some point.  Either way, thanks for working on this.