adafruit/circuitpython #5633: Add Raspberry Pi wifi support

tannewt, 2021-11-30:
Raspberry Pis use a Cypress wifi (formerly broadcom) chip for WiFi and BLE. Adding support for it would give WiFi access to the port.

MicroPython has support for the chips for the PyBoard D boards that use a WiFi chip from the same family. MP code: https://github.com/micropython/micropython/tree/master/drivers/cyw43

dhalbert, 2023-10-28:
Implemented.

andrethomas, 2024-01-27:
Hi, How can the wifi be used on RPi zero w?

tannewt, 2024-01-29:
Sorry, this hasn't been implemented. @dhalbert confused this with the Pico W, which has the same chip. We don't have a plan for the Zero W.

asmagill, 2024-02-17:
Don't have a plan as in not going to do it, or as in it's a low enough priority it hasn't even reached the planning stages (I know the broadcom port is pretty experimental at the moment)?

dhalbert, 2024-02-17:
We're not actively working on enhancing the broadcom port right now. Not "never", but not now either.

tannewt, 2024-02-20:
The wifi chip is similar to the one on the Pico W but the code we use for it is licensed for RP2040 use only. So, we'd need to get a broader license or replace the code. (Not important for adafruit-funded dev.)

earlthesquirrel, 2025-03-06:
Is this implemented for the CM4?  I don't see it listed there, but would think if it works on the Raspberry Pi 4, it would work on this. (I'm not clear from this issue exactly what boards are covered)

tannewt, 2025-03-06:
No. It doesn't work on Pi 4 either.

earlthesquirrel, 2025-03-06:
Is there any Wifi solution for CircuitPython on the Raspberry Pi?  Or wired network?

tannewt, 2025-03-06:
Nope. Bare metal raspberry pi support is still very alpha. Blinka on Linux is a better solution.

*Maybe* our move to Zephyr will help us add support for wifi on bare metal broadcom but that'll depend on Zephyr's support for it. We won't add it ourselves.