adafruit/circuitpython #5639: vectorio shapes ignore pixel_shader pi-zero-2-w

rsbohn, 2021-11-30:
### CircuitPython version

```python
Adafruit CircuitPython 7.1.0-beta.1-4-g8d6957b79 on 2021-11-30; Raspberry Pi Zero 2W with rp3a0
```

### Code/REPL

```python
import board
import displayio
import vectorio
display = board.DISPLAY
splash = displayio.Group()
# color background
pal0 = displayio.Palette(1)
pal0[0] = 0x339933
bg = displayio.Bitmap(display.width, display.height, 1)
grid = displayio.TileGrid(bg, pixel_shader=pal0)
splash.append(grid)

pal1 = displayio.Palette(1)
pal1[0] = 0x993333
circle = vectorio.Circle(radius=200,x=10,y=10,pixel_shader=pal1)
splash.append(circle)

while True:
    pass
```

### Behavior

You get a green screen with a large black circle in the top left.

Expected result: The large circle is red.

rsbohn, 2021-11-30:
Oops, forgot `display.show(splash)`. Circle is still black, but now you can see the green background.

tannewt, 2021-12-01:
Thanks for the bug. This may be related to the new addition of 32-bit color (really 24 bit) in displayio.

PR #5728 (2021-12-15): Support 32 bit color with vectorio
Fixes #5639
commit 01f8cffd (2021-12-15): Support 32 bit color with vectorio

Fixes #5639
closed 2021-12-15 by dhalbert