adafruit/circuitpython #5643: FrequencyIO on CircuitPython 7.0 only returns 0. Same setup works with 6.3.0.

FevenKitsune, 2021-12-01:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0 on 2021-09-20; Adafruit Feather M4 Express with samd51j19
```

### Code/REPL

```python
# CircuitPython 7.0
import frequencyio
import board
import time

frequency = frequencyio.FrequencyIn(board.D11)

# Loop while printing the detected frequency
while True:
    print(frequency.value)

    # Optional clear() will reset the value
    # to zero. Without this, if the incoming
    # signal stops, the last reading will remain
    # as the value.
    frequency.clear()
    time.sleep(0.1)
```

### Behavior

Signal is generated with an external signal generator (-40.0mV-3.26v Pk-Pk, 1kHz Square Wave).

Serial output with CircuitPython 7.0.0:
```
0
0
0
0
0
...
```

Serial output with CircuitPython 6.3.0:
```
1000
1164
1000
1000
1058
...

Behavior is observed at higher frequencies as well.

### Additional information

Did some debugging with @dhalbert on the Adafruit Discord server.

- Swapped coax cable.
- Validated normal GPIO operation with pin.
![Oscilloscope 1kHz Bug Report](https://user-images.githubusercontent.com/9381835/144164355-9bcee5be-adb3-4d13-b3f7-c4cfc490d5cf.png)

dhalbert, 2021-12-05:
This appears to be some kind of initialization problem. It works the first time:
```py
Adafruit CircuitPython 7.0.0 on 2021-09-20; Adafruit Feather M4 Express with samd51j19
>>> 
>>> import board,frequencyio
>>> frequency = frequencyio.FrequencyIn(board.D11)
>>> frequency.value
529
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello World!

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0 on 2021-09-20; Adafruit Feather M4 Express with samd51j19
>>> import board,frequencyio
>>> frequency = frequencyio.FrequencyIn(board.D11)
>>> frequency.value
0
>>> 
```

dhalbert, 2021-12-08:
Fixed by #5665.

PR #5665 (2021-12-07): add a frequencyin_reset() for VM restart
Fixes #5643.

`FrequencyIn` was assuming it was started cleanly. But in the case of a VM restart (e.g. soft boot), some resetting needed to be done.
closed 2021-12-08 by dhalbert