adafruit/circuitpython #5644: Forum User reports adafruit_sdcard and esp32spi fail to share an spi bus

jepler, 2021-12-01:
### CircuitPython version

```python
"I am on CircuitPython 7, and tried reverting back to 6.3, same results on both."
```

### Code/REPL

```python
# If you have an AirLift Featherwing or ItsyBitsy Airlift:
    esp32_cs = DigitalInOut(board.D13)
    esp32_ready = DigitalInOut(board.D11)
    esp32_reset = DigitalInOut(board.D12)

    spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
    # spi = board.SPI()
    esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)

    sdcard = sdcardio.SDCard(spi, board.SD_CS)
    vfs = storage.VfsFat(sdcard)
    storage.mount(vfs, "/sd")
```

### Behavior

['snare.wav', 'hat.wav', 'kick.wav']
Traceback (most recent call last):
File "code.py", line 59, in <module>
File "adafruit_esp32spi/adafruit_esp32spi.py", line 335, in status
File "adafruit_esp32spi/adafruit_esp32spi.py", line 325, in _send_command_get_response
File "adafruit_esp32spi/adafruit_esp32spi.py", line 293, in _wait_response_cmd
File "adafruit_esp32spi/adafruit_esp32spi.py", line 277, in _check_data
RuntimeError: Expected A0 but got 80

### Description

Internet lore says that it's possible to share an SPI bus with an SD card and another device, as long as the SD card is initialized first.

User reports that the same error occurs when the SD card is initialized first. However, a full example in this order was not furnished as far as I saw.

If it's not a hardware impossibility, then it seems most likely the problem resides in the sd card code, perhaps failing to do "extra clocks" at the end of a transaction.

The recently added "CMD25" stuff does not seem relevant here, as the SD card is not being written in the example (and the problem is reported before the addition of CMD25).

### Additional information

See https://forums.adafruit.com/viewtopic.php?f=60&t=185718

nuwaveit, 2021-12-01:
I am the one who opened this on the forum.  I'm using the PyGamer with an Airlift ESP32 wing, which works fine with the sample code.  Inserting an SD card (no code change) causes a "Timeout waiting for SPI char".  Init the SD card causes the above error where it gets the wrong response.  Simply eject the SD card and reload, the wifi works again.

jepler, 2021-12-01:
I don't have quite the right hardware set-up, but I did find an airlift featherwing.

I put it together with a Feather Adalogger and a Feather M4 Express.

I've got CircuitPython 7.1.0-beta.1 loaded. My airlift version is old, 1.2.2, while you've got an up to date version.

Unfortunately, I'm not reproducing the problem yet. With or without the SD card inserted, the AirLift example completes OK.  Even if I delete the lines to initialize the SD card and remove/re-insert it so it is not yet initialized into SPI mode.

Here's my full test program:
```python
# SPDX-FileCopyrightText: 2019 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

import os
import time

from adafruit_esp32spi import adafruit_esp32spi
from digitalio import DigitalInOut
import adafruit_esp32spi.adafruit_esp32spi_socket as socket
import adafruit_requests as requests
import board
import busio
import sdcardio
import storage

# Get wifi details and more from a secrets.py file
try:
    from secrets import secrets
except ImportError:
    print("WiFi secrets are kept in secrets.py, please add them there!")
    raise

print("ESP32 SPI webclient test")

JSON_URL = "http://api.coindesk.com/v1/bpi/currentprice/USD.json"

# If you have an AirLift Featherwing or ItsyBitsy Airlift:
esp32_cs = DigitalInOut(board.D13)
esp32_ready = DigitalInOut(board.D11)
esp32_reset = DigitalInOut(board.D12)

# Adalogger Featherwing
sd_cs = board.D10

spi = board.SPI()
# SD card must be initialized before AirLift
try:
    print("Attempting to mount sd card")
    sdcard = sdcardio.SDCard(spi, sd_cs) 
    vfs = storage.VfsFat(sdcard)
    storage.mount(vfs, '/sd')  
    print(os.listdir("/sd"))
except Exception as e:
    print("no sd card:", e)
    print("continuing")

esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)
requests.set_socket(socket, esp)
    
if esp.status == adafruit_esp32spi.WL_IDLE_STATUS:
    print("ESP32 found and in idle mode")
print("Firmware vers.", esp.firmware_version)
print("MAC addr:", [hex(i) for i in esp.MAC_address])
    
print("Connecting to AP...")
while not esp.is_connected:
    try:
        esp.connect_AP(secrets["ssid"], secrets["password"])
    except RuntimeError as e:
        print("could not connect to AP, retrying: ", e)
        continue
print("Connected to", str(esp.ssid, "utf-8"), "\tRSSI:", esp.rssi)
print("My IP address is", esp.pretty_ip(esp.ip_address))
    
for i in range(1):
    print(i) 
    r = requests.get(JSON_URL)
    data = r.json()
    print (data)
    r.close() 
        
print("Done!")
```

Typical output:
```
code.py output:
ESP32 SPI webclient test
Attempting to mount sd card
['overlays', 'kernel7.img', 'kernel.img', 'kernel7l.img', 'issue.txt', 'bcm2710-rpi-3-b-plus.dtb', 'kernel8.img', 'bcm2708-rpi-b-plus.dtb', 'bcm2708-rpi-b-rev1.dtb', 'bcm2708-rpi-b.dtb', 'bcm2708-rpi-cm.dtb', 'bcm2708-rpi-zero-w.dtb', 'bcm2708-rpi-zero.dtb', 'bcm2709-rpi-2-b.dtb', 'bcm2710-rpi-2-b.dtb', 'configtxt.bak', 'config.txt', 'bcm2710-rpi-3-b.dtb', 'bcm2710-rpi-cm3.dtb', 'bcm2711-rpi-4-b.dtb', 'cmdline.txt', 'bcm2711-rpi-400.dtb', 'bcm2711-rpi-cm4.dtb', 'COPYING.linux', 'start.elf', 'start_cd.elf', 'start_db.elf', 'start_x.elf', 'fixup.dat', 'fixup_cd.dat', 'fixup_db.dat', 'fixup_x.dat', 'bootcode.bin', 'start4.elf', 'start4cd.elf', 'start4db.elf', 'start4x.elf', 'fixup4.dat', 'fixup4cd.dat', 'fixup4db.dat', 'fixup4x.dat', 'LICENCE.broadcom']
ESP32 found and in idle mode
Firmware vers. bytearray(b'1.2.2\x00')
MAC addr: ['0x8', '0x8e', '0x88', '0x1c', '0x52', '0x10']
Connecting to AP...
Connected to OpenWrt-secret 	RSSI: -59
My IP address is 10.0.3.112
0
{'time': {'updated': 'Dec 1, 2021 21:30:00 UTC', 'updatedISO': '2021-12-01T21:30:00+00:00', 'updateduk': 'Dec 1, 2021 at 21:30 GMT'}, 'disclaimer': 'This data was produced from the CoinDesk Bitcoin Price Index (USD). Non-USD currency data converted using hourly conversion rate from openexchangerates.org', 'bpi': {'USD': {'code': 'USD', 'description': 'United States Dollar', 'rate_float': 56577.6, 'rate': '56,577.6017'}}}
Done!

Code done running.
```
[image]

nuwaveit, 2021-12-01:
Would it help if I provided my code file?  It looks practically same as yours.  I can also try to record a video of it working and not working.

Is there any diagnostics I can try on my end?  I tried adding debug=True to the esp32 ctor, but really didn't add much value...

jepler, 2021-12-01:
If you can, please test with a different SD card too.

nuwaveit, 2021-12-02:
@jepler Different SD Card didnt make a difference.  I copied your code from above, still gave the same output:

`code.py output:
ESP32 SPI webclient test
Attempting to mount sd card
['logs', 'overlays', 'prj', '9.0.3.gz', 'arcade.cfg', 'bcm2708-rpi-0-w.dtb', 'bcm2708-rpi-b.dtb', 'bcm2708-rpi-b-plus.dtb', 'bcm2708-rpi-cm.dtb', 'bcm2709-rpi-2-b.dtb', 'bcm2710-rpi-3-b.dtb', 'bcm2710-rpi-cm3.dtb', 'bootcode.bin', 'cmdline.txt', 'cmdline3.txt', 'config.txt', 'COPYING.linux', 'fixup.dat', 'kernel4922.img', 'LICENCE.broadcom', 'README', 'RELEASE', 'start.elf']
Traceback (most recent call last):
  File "code.py", line 52, in <module>
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 335, in status
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 325, in _send_command_get_response
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 292, in _wait_response_cmd
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 271, in _wait_spi_char
RuntimeError: Timed out waiting for SPI char

Code done running.`

jerryneedell, 2021-12-02:
I am trying to reproduce this, I have a pygamer with an airlift featherwing,
So far, ping @jepler  code with the SDCard on SD_CS  the Airlift works fine with or with out the card inserted BUT it is so far, unable to detect any SDCards..... still investigating

jerryneedell, 2021-12-02:
After  trying several different SDCards, I found on that works and it works OK with the airlift
```
Adafruit CircuitPython 7.1.0-beta.1-19-gb83e09858-dirty on 2021-12-02; Adafruit PyGamer with samd51j19
>>> 
>>> import sdtest
ESP32 SPI webclient test
Attempting to mount sd card
['lib', 'hello.py', '.Trash-1000', '.Spotlight-V100', '.fseventsd', '.Trashes', 'swfolog.txt', 'swfoplayback.txt']
ESP32 found and in idle mode
Firmware vers. bytearray(b'1.7.1\x00')
MAC addr: ['0xa8', '0x44', '0x57', '0x12', '0xcf', '0xa4']
Connecting to AP...
Connected to Needell Airport    RSSI: -75
My IP address is 10.0.0.139
0
{'time': {'updated': 'Dec 2, 2021 14:51:00 UTC', 'updatedISO': '2021-12-02T14:51:00+00:00', 'updateduk': 'Dec 2, 2021 at 14:51 GMT'}, 'disclaimer': 'This data was produced from the CoinDesk Bitcoin Price Index (USD). Non-USD currency data converted using hourly conversion rate from openexchangerates.org', 'bpi': {'USD': {'code': 'USD', 'description': 'United States Dollar', 'rate_float': 57105.3, 'rate': '57,105.2617'}}}
Done!
>>> 
```
Note, the "dirty" build is not relevant to this test - i have some changes to unrelated code taht I am testing,

jerryneedell, 2021-12-02:
I am unable to read many of my SDCards on this PyGamer... So far only the one works
It is an 8Gbyte card -- not sure of the mauna. just says Micro Sd Ultimate 3.0 70MB/Sec

but to the original issue, the airlift works with it in or out.....

jerryneedell, 2021-12-02:
hmmm -- if I switch to using the Adafruti_CircuitPythom_SD (adafruit_sdcard) library, i can read al of my SDCards.

jepler, 2021-12-02:
adafruit_sdcard defaults to 1_320_000 as the SPI clock while sdcardio uses 8_000_000.  You can try changing the frequency.

nuwaveit, 2021-12-02:
if I do this:
sdcard = sdcardio.SDCard(spi, sd_cs, baudrate=1320000)
I get no sd card, and still the ESP32 errors...

gonna see if i can try the adafruit_sdcard lib...

nuwaveit, 2021-12-02:
same result with using example from @jerryneedell and adding adafruit_sdcard.py to the board:

```
code.py output:
ESP32 SPI webclient test
Attempting to mount sd card
no sd card: no SD card
continuing
Traceback (most recent call last):
  File "code.py", line 51, in <module>
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 335, in status
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 325, in _send_command_get_response
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 292, in _wait_response_cmd
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 271, in _wait_spi_char
RuntimeError: Timed out waiting for SPI char
```

dhalbert, 2021-12-02:
@nuwaveit Do you have any brand-name cards (e.g. Sandisk) that work? The `sdcardio` and `adafruit_sdcard` libraries use the slower SPI interface to cards, instead of the SDIO interface. We have found problems with a variety of cards.

nuwaveit, 2021-12-02:
@dhalbert yeah, im using a newer Sandisk, and an older Kingston.  I have a 3rd with no name on it but has the "microSD" logo...

jerryneedell, 2021-12-02:
FYI -- I tried using baudrate=1320000 with sdcardio and it still fails to detect the card for me.
Same card is fine with adafruit_sdcard.

nuwaveit, 2021-12-02:
For the fact that I can consistently get this error sounds to me like there's a bug somewhere, maybe caused by a timing issue, or something is not getting reset...just my hunch...

File "adafruit_esp32spi/adafruit_esp32spi.py", line 335, in status
...
RuntimeError: Expected A0 but got 80

jepler, 2021-12-03:
I testedwith the above airlift setup all these cards, still using `sdcardio`. (see my full program listing above)

[image]

 * kingston 32gb micro sdhc class 10
 * no-name 128MB card (https://www.adafruit.com/product/5250)
 * g.skill micro sd hc 32GB class 10
 * sandisk edge 16GB micro sd HC class 10 (possibly https://www.adafruit.com/product/2820)
 * 2GB micro sd "SD-CO2G"

All passed, mounting SD and then doing the wifi fetch.

I had some cards appear to fail at first when I hot plugged them and then soft-restarted.  All the cards worked when I fully power-cycled my testing rig (unplugged USB & plugged back in) -- the failure mode was "no sd card", after which the wifi portion of the test ran successfully.

I did similar tests using a TFT featherwing and an RP2040 Feather.  All cards mounted fine there.  This setup did not have an airlift featherwing so I didn't test the wifi portion there.

jerryneedell, 2021-12-03:
¯\_(ツ)_/¯

nuwaveit, 2021-12-03:
I didn't try it where I fully power off and unplug the USB to see if I can make the WiFi work with a mounted SD card.  I will try once I'm back home. 

But aside from that, what's the next step?  It seems from @jerryneedell that his PyGamer rejected several SD cards as well, and I've tried 4 or 5 as well, all cause the WiFi to fail. Is there a way to raise this to the dev team that worked on PyGamer?  I do not have any other device like the adalogger to compare...

jerryneedell, 2021-12-04:
I have not been able to reproduce your finding on mine. It still fails to detect the cards even after power cycle when using sdcardio.

Also on my pygamer, I have not been able to reproduce the wifi issue reported by @nuwaveit 

If there is a difference in how the SDCards mount after power cycle vs soft reboot, should consider this a "bug".

jepler, 2021-12-04:
I only saw a problem when hot-plugging the SD card.  If the SD card is inserted, it is fine no matter how many times I re-run the code.

jerryneedell, 2021-12-04:
Ah -- that sounds better.  It's hard to see why the pygamer should be different that any other SDCard reader. I have not seen this issue that I recall with other boards with SDCards. I'll try a few more configurations.

jerryneedell, 2021-12-04:
I have reproduced the SDCard issue with a Feather M4 express and an adalogger featherwing
using sdcardio
```

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-629-g102af55e0 on 2021-11-05; Adafruit Feather M4 Express with samd51j19
>>> import sdd10
Attempting to mount sd card
no sd card: no SD card
continuing
>>>
soft reboot
```
using adafruit_sdcard
```
Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-629-g102af55e0 on 2021-11-05; Adafruit Feather M4 Express with samd51j19
>>> import sdd10_sdcard
Attempting to mount sd card
['sdtest.xxx', 'test.txt', 'sdfile.txt', 'temp_log.txt', 'DATALOG.TXT', 'lib', '.Trash-1000', 'cave', 'font5x8.bin', 'stmpe610_nrf52.py', 'paint_nrf52.py', 'bitmapfont.mpy']
>>> 

```
code
```
# SPDX-FileCopyrightText: 2019 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

import os
import time

from digitalio import DigitalInOut
import board
import busio
import sdcardio
import storage

# Adalogger Featherwing
sd_cs = board.D10

spi = board.SPI()
# SD card must be initialized before AirLift
try:
    print("Attempting to mount sd card")
    sdcard = sdcardio.SDCard(spi, sd_cs) 
    vfs = storage.VfsFat(sdcard)
    storage.mount(vfs, '/sd')  
    print(os.listdir("/sd"))
except Exception as e:
    print("no sd card:", e)
    print("continuing")
```
and
```
# SPDX-FileCopyrightText: 2019 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

import os
import time

from digitalio import DigitalInOut
import board
import busio
import adafruit_sdcard
import storage

# Adalogger Featherwing
sd_cs = DigitalInOut(board.D10)

spi = board.SPI()
# SD card must be initialized before AirLift
try:
    print("Attempting to mount sd card")
    sdcard = adafruit_sdcard.SDCard(spi, sd_cs) 
    vfs = storage.VfsFat(sdcard)
    storage.mount(vfs, '/sd')  
    print(os.listdir("/sd"))
except Exception as e:
    print("no sd card:", e)
    print("continuing")

```

jerryneedell, 2021-12-04:
ah -- upgrading to latest CP now works with sdcardio on the feather m4 express.

jerryneedell, 2021-12-04:
OK -- I give up -- I rebuild CP for the Pygamer and now it is working with sdcardio on cards that had been failing.....

@nuwaveit Can you try with the lates build of CP from https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/pygamer/en_US/

use the 7.2.0 build

nuwaveit, 2021-12-04:
Will do later tonight and report back.  @jerryneedell

nuwaveit, 2021-12-05:
OK, using the 7.2.0 alpha build didn't make any difference.  I have 1 old SD that consistently won't read, but that's OK.  All my other SDs read.  The Wifi does not work no matter which card is in there, and no matter which SD library is used.  But I did get slightly different errors at one point when testing with my SD cards out of my raspberry pis:

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
ESP32 SPI webclient test
Attempting to mount sd card
['overlays', 'System Volume Information', 'kernel7.img', 'kernel.img', 'issue.txt', 'bcm2710-rpi-3-b-plus.dtb', 'kernel7l.img', 'kernel8.img', 'bcm2708-rpi-b-plus.dtb', 'bcm2708-rpi-b-rev1.dtb', 'bcm2708-rpi-b.dtb', 'bcm2708-rpi-cm.dtb', 'bcm2708-rpi-zero-w.dtb', 'bcm2708-rpi-zero.dtb', 'bcm2709-rpi-2-b.dtb', 'bcm2710-rpi-2-b.dtb', 'COPYING.linux', 'cmdline.txt', 'config.txt', 'bcm2710-rpi-3-b.dtb', 'bcm2710-rpi-cm3.dtb', 'bcm2710-rpi-zero-2.dtb', 'bcm2711-rpi-4-b.dtb', 'bcm2711-rpi-400.dtb', 'bcm2711-rpi-cm4.dtb', 'start.elf', 'start_cd.elf', 'start_db.elf', 'start_x.elf', 'fixup.dat', 'fixup_cd.dat', 'fixup_db.dat', 'fixup_x.dat', 'bootcode.bin', 'start4.elf', 'start4cd.elf', 'start4db.elf', 'start4x.elf', 'fixup4.dat', 'fixup4cd.dat', 'fixup4db.dat', 'fixup4x.dat', 'LICENCE.broadcom']
Traceback (most recent call last):
  File "code.py", line 3, in <module>
  File "test_wifi_github.py", line 51, in <module>
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 335, in status
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 325, in _send_command_get_response
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 293, in _wait_response_cmd
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 277, in _check_data
**RuntimeError: Expected A0 but got 9F**
```

The one below is interesting, because I usually got that error when the card was inserted, but did not actually initialize the SD card in code.  But as you can see, its reading the card, but still throws the SPI timeout error.

```
code.py output:
ESP32 SPI webclient test
Attempting to mount sd card
['overlays', 'bcm2708-rpi-b-plus.dtb', 'COPYING.linux', 'LICENCE.broadcom', 'issue.txt', 'bcm2708-rpi-b-rev1.dtb', 'bcm2708-rpi-b.dtb', 'bcm2708-rpi-cm.dtb', 'bcm2708-rpi-zero-w.dtb', 'bcm2708-rpi-zero.dtb', 'bcm2709-rpi-2-b.dtb', 'bcm2710-rpi-2-b.dtb', 'bcm2710-rpi-3-b-plus.dtb', 'bcm2710-rpi-3-b.dtb', 'bcm2710-rpi-cm3.dtb', 'bcm2711-rpi-4-b.dtb', 'bcm2711-rpi-400.dtb', 'bcm2711-rpi-cm4.dtb', 'bootcode.bin', 'cmdline.txt', 'config.txt', 'fixup.dat', 'fixup4.dat', 'fixup4cd.dat', 'fixup4db.dat', 'fixup4x.dat', 'fixup_cd.dat', 'fixup_db.dat', 'fixup_x.dat', 'kernel.img', 'kernel7.img', 'kernel7l.img', 'kernel8.img', 'start.elf', 'start4.elf', 'start4cd.elf', 'start4db.elf', 'start4x.elf', 'start_cd.elf', 'start_db.elf', 'start_x.elf', 'System Volume Information']
Traceback (most recent call last):
  File "code.py", line 3, in <module>
  File "test_wifi_github.py", line 51, in <module>
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 335, in status
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 325, in _send_command_get_response
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 292, in _wait_response_cmd
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 271, in _wait_spi_char
**RuntimeError: Timed out waiting for SPI char**
```

jerryneedell, 2021-12-09:
is this related to #5688 ?

dhalbert, 2021-12-10:
We fixed #5600 with #5688, and that bug fix is now in "Absolute Newest". It's possible that fix will also fix the bug that some of you have reported in this issue. Could you try the "Absolute Newest" build if you have a chance to see if you still see the problem? Thanks.

nuwaveit, 2022-02-04:
Sorry its been a long while, I just tried this version, still not working for me....

[adafruit-circuitpython-pygamer-en_US-20220204-a39f0cf.uf2](https://adafruit-circuit-python.s3.amazonaws.com/bin/pygamer/en_US/adafruit-circuitpython-pygamer-en_US-20220204-a39f0cf.uf2)

DJDevon3, 2024-02-10:
While not specifically related to esp32spi I was having difficulty getting sdcardio to share a bus. Came across this issue via google search.  @jepler code did help.  Was able to share SCK, MOSI, MISO with sdcardio and a SPI TFT. Each peripheral has its own CS pin. So can at least confirm the issue isn't the fault of sdcardio.

```py
import sdcardio
import storage
from adafruit_display_text import label
from circuitpython_st7796s import ST7796S

displayio.release_displays()
# 4.0" ST7796S Display
DISPLAY_WIDTH = 480
DISPLAY_HEIGHT = 320

tft_cs = board.D9
tft_dc = board.D10
tft_rst = board.D17
sd_cs = board.D5

spi = board.SPI()
# Initialize SPI SDCard prior to other SPI peripherals!
try:
    print("Attempting to mount sd card")
    sdcard = sdcardio.SDCard(spi, sd_cs) 
    vfs = storage.VfsFat(sdcard)
    storage.mount(vfs, '/sd')  
    print(os.listdir("/sd"))
except Exception as e:
    print("no sd card:", e)
    print("continuing")

display_bus = displayio.FourWire(spi, command=tft_dc, chip_select=tft_cs, reset=tft_rst)
display = ST7796S(display_bus, width=DISPLAY_WIDTH, height=DISPLAY_HEIGHT, rotation=180)

# System Stats
u_name = os.uname()
print("Board: ", u_name[4])
print("Type: ", u_name[0])
print("Version: ", u_name[3])
```
code.py output:
```py
Attempting to mount sd card
['System Volume Information', 'Astral_Fruit_8bit.bmp', 'vbat_spritesheet.bmp', 'purbokeh_8.bmp']
Board:  FeatherS3 with ESP32S3
Type:  ESP32S3
Version:  8.2.9 on 2023-12-06

Code done running.
```
Ultimately in my case it was due to poor breadboard connections. I realize that doesn't apply if it's on a PCB but the successful SPI bus sharing with sdcardio might be helpful here.