adafruit/circuitpython #5645: Error building CircuitPython using terminal command make BOARD/Syntax errors in mksdiodata.py and mkcandata.py that prevent building

bibletech55, 2021-12-01:
I've been following several Adafruit tutorials, such as this one, https://learn.adafruit.com/building-circuitpython?view=all, as I've been working on adding TinyZero support from this post: https://github.com/adafruit/circuitpython/issues/5393. I was working on the Build CircuitPython section, but I've encountered a few strange problems: 

First, make BOARD kept failing at line 5 in mksdiodata.py - in ports/atmel-samd/tools. I fixed it by removing the f at the beginning of each string (why that worked, I have no idea) as it would fail at each f-string line, and I also removed the f at the beginning of each string in mkcandata.py in the same folder, as it would experience the same error.

But I have a new problem. CircuitPython refuses to build because of a "no rule" error. To investigate, I used -k at the end of the command to force past the error and discovered many, many more "no rule" errors for various files. After researching the issue, I looked into my circuitpython folder (the one that holds my fork of the GitHub repo), and discovered that, as far as I can tell so far, the "no rule" errors are files that are missing from their folders!! Basically, the folders that are supposed to contain files needed to build are empty for some reason!
I validated this by checking GitHub's repo, and sure enough, GitHub's repo does have the needed files - do I copy each of the files' folders from GitHub's repo and add them to their respective locations in my fork (so that I don't have to go through downloading the whole fork again, as I have a pretty complicated setup that involves a computer and a Raspberry Pi for building CircuitPython, as well as a USB drive to transfer the files)? Also, any ideas why the needed folders disappeared in the first place?

I'll be honest, I'm not completely sure what I'm doing - this is my first project using Raspian/Linux terminal commands this extensively. Just a heads up that I'm new at this :)

tannewt, 2021-12-01:
> First, make BOARD kept failing at line 5 in mksdiodata.py - in ports/atmel-samd/tools. I fixed it by removing the f at the beginning of each string (why that worked, I have no idea) as it would fail [...]

The f prefix is called an f-string. It is in newer versions of Python. What version are you using? You likely want python 3.8 or newer.

> But I have a new problem. CircuitPython refuses to build because of a "no rule" error.

Did you do `make fetch-submodules`? Some files get pulled in as submodules.

The Discord #circuitpython-dev is the best place to get help setting up a CircuitPython build. https://adafru.it/discord

dhalbert, 2022-02-18:
Closing this for now. Please let us know if you are still having trouble, or ask in https://adafru.it/discord in the #circuitpython-dev channel, which is a good place to ask for build issues.