adafruit/circuitpython #5646: board "NodeMCU ESP-12K" (ESP32s2) doesn't boot after flash, no CIRCUITPY, no console

apws, 2021-12-01:
### CircuitPython version

```python
CircuitPython 7.0.0
```

### Code/REPL

```python
N/A
```

### Behavior

N/A

### Description

After flashing to this BIN from here it does nothing. With original stock firmware board has some serial console OK (see below).
https://circuitpython.org/board/ai_thinker_esp_12k_nodemcu/

commands:
esptool --port COM10 --baud 460800 erase_flash
esptool --port COM10 --baud 460800 write_flash 0 adafruit-circuitpython-ai_thinker_esp_12k_nodemcu-en_US-7.0.0.bin

I tried to peek into flashed backup and thanks to 2nd board, I also reverted original stock firmware to confirm it is OK, but CircuitPython cant boot.

Its again quite old board (excuse me) with ESP-12K-08, so presumably 8MB PSRAM (OT: spresense runs, tnx)

### Additional information

original firmware boots with this

ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x1ae4
load:0x4004c000,len:0x1888
load:0x40050000,len:0x20f0
entry 0x4004c348

################################################
arch:ESP32S2, 0
compile_time:Ai-Thinker|B&T
ble_mac:0000DCb1FD3F
wifi_mac:84F7037bE780
sdk_version:v4.2.1-64-gd9ec7df39
firmware_version:2.1.0
compile_time:Jun 30 2021 17:39:22

ready

################################################

tannewt, 2021-12-02:
It looks like this board has a USB to serial converter on the USB plug. You'll need to add a USB breakout cable to the native USB pins to get the CIRCUITPY drive and REPL to appear.

apws, 2021-12-02:
wow, ya, thanks for info, I felt something weird here, best to avoid such hw - I trusted the downloads web (lack of) details... tnx, so its solved

closed 2021-12-02 by apws

apws, 2021-12-02:
perhaps, adding link to this similar one on ESP-12K page in downloads helps - info about native usb on GPIO19/20 (but its not available to edit docs from web)
https://circuitpython.org/board/espressif_saola_1_wrover/

tannewt, 2021-12-03:
@apws Please double check my note here: https://github.com/adafruit/circuitpython-org/pull/802