adafruit/circuitpython #5650: Broadcom build for RPI 4B not booting

jerryneedell, 2021-12-03:
I have built CP for both the RPI Zero 2W and the 4B.
When I connect a USB to Serial converter to GND, TX,RX on the Zero 2W it boots to the REPL as expected.

When I try the same on the 4B, I see no output on the serial port.
If I connect an HDMI monitor, I see no text to the screen, but it does go to a large multicolored square on the display.

I'm curious if anyone else has tried the 4B? 

I have tried 7.1.0-beta.1 from circuitpython.org as well just to verify it is not my build -- same results.

jerryneedell, 2021-12-05:
Some additional information:
I hooked up my JLink and configureg openocd using the raspberrypi_pi4.cfg  file
It connects, but The code just does not appear to be running
```
Are you sure you want to change the file? (y or n) y
Reading symbols from kernel8.elf...
(gdb) cont
Continuing.
^Cbcm2711.a72.1 halted in AArch64 state due to debug-request, current mode: EL3H
cpsr: 0x000003cd pc: 0x200
MMU: disabled, D-Cache: disabled, I-Cache: disabled
bcm2711.a72.2 halted in AArch64 state due to debug-request, current mode: EL3H
cpsr: 0x000003cd pc: 0x200
MMU: disabled, D-Cache: disabled, I-Cache: disabled
bcm2711.a72.3 halted in AArch64 state due to debug-request, current mode: EL3H
cpsr: 0x000003cd pc: 0x200
MMU: disabled, D-Cache: disabled, I-Cache: disabled

Program received signal SIGINT, Interrupt.
0x0000000000000200 in ?? ()
(gdb) disass
No function contains program counter for selected frame.
(gdb) info stack
#0  0x0000000000000200 in ?? ()
#1  0x0000000000000000 in ?? ()
Backtrace stopped: not enough registers or memory available to unwind further
(gdb) 
```
It stops at 0x200 and seems to sit there.
when I enter cont then control C -- it is still at 0x200

here is a dump of info reg
```x0             0x40                64
x1             0x22                34
x2             0x840370a620c40185  9512570696753217925
x3             0xaa0081cf28010080  12249933713177116800
x4             0x9030800004c129cd  10389945077911857613
x5             0x8411c2e48d412f77  9516601774488760183
x6             0x42c2918822084a80  4810567365857659520
x7             0x68205001a12e8098  7503084947128615064
x8             0x8100100902488044  9295447261771759684
x9             0x10fcede3554088c0  1224114759381584064
x10            0x5d202de8200c808   419400869985961992
x11            0x4a62870e28d04b5   334999687588021429
x12            0xe089631b0018344f  16179572103922070607
x13            0x560c206404211063  6200366400920424547
x14            0x40b9c61d00460407  4663976716979799047
x15            0x60008200c8cc7     1688884757695687
x16            0x148050020a2ac000  1477268647468253184
x17            0xe432085026ea5854  16443214329826400340
x18            0x3404100d4c3140c   234258652448625676
x19            0x240c8182c70904    10146849545980164
x20            0x4673014444ae0802  5076402597698340866
x21            0x2c23881db04880b8  3180535422946672824
x22            0x4858a0d04001448a  5213093384968815754
x23            0x826411200000400   587228346983646208
x24            0x6cea416181210b09  7848157187680570121
x25            0xc448050300142f04  14143560140201406212
x26            0xa955ffb60002c410  12201939922579538960
x27            0x120643e13c3e9f39  1298800177200406329
x28            0xc8181000c991a19c  14418291802594451868
x29            0x281024188000201   180427940549362177
x30            0x50                80
sp             0x11003db1580a184   0x11003db1580a184
pc             0x200               0x200
cpsr           0x3cd               [ SP=1 EL=2 nRW=0 F I A D ]
fpsr           0x0                 0
fpcr           0x0                 0
ELR_EL1        0xb420102c3dc29400  0xb420102c3dc29400
ESR_EL1        0x0                 0
SPSR_EL1       0x10                16
ELR_EL2        0xb84a487ad321034b  0xb84a487ad321034b
ESR_EL2        0x0                 0
SPSR_EL2       0x10                16
ELR_EL3        0x200               0x200
ESR_EL3        0x96000040          2516582464
SPSR_EL3       0x3cd               973
(gdb) 
```

If I use the JLink ond openocd with the raspberrypi_pi3.cfg on mu zero2w, then I am able to see the code running and use gdb as expected. 

here is an info stack from the zero2w
```
(gdb) info stack
#0  0x00000000000bf220 in serial_bytes_available ()
    at ../../supervisor/shared/serial.c:152
#1  0x00000000000b71b8 in mp_hal_stdin_rx_chr ()
    at ../../supervisor/shared/micropython.c:52
#2  0x00000000000e1de4 in count_cont_bytes (end=0xfc "", 
    start=0x301990 "UQU5UUU\325\321UQq\024U\025U\005UUUUUEUU\025\025\027UUUUQTUQ\025UTUUuUtQUqUW\025\005EU\025W\025\025UUUUWTUUT\325UUuUEuTQPUQQ\025U\035U]UU\025UWU\027U\027UUU\321UUUUU\325\025\325UuTU]\024UUUUEUU\rUUWEWU]\325UUUUUUUEqSUUUUQDUWU\005EUUmUYWU\025WEUUUQ\325U\325QUUQ\321\321uQUTUUW\025UU\025UMU\025UUwU\027UuuUUUUQU"...) at ../../shared/readline/readline.c:63
#3  readline_process_char (c=<optimized out>)
    at ../../shared/readline/readline.c:358
#4  0x00000000000ebf44 in mp_reader_new_stdin (buf_max=256, 
    reader_stdin=0x266280, reader=0x7feb8) at ../../shared/runtime/pyexec.c:311
#5  do_reader_stdin (c=<optimized out>) at ../../shared/runtime/pyexec.c:328
#6  pyexec_raw_repl () at ../../shared/runtime/pyexec.c:558
#7  0x0000000000000000 in ?? ()
(gdb) q
A debugging session is active.
```

any suggestions on how to proceed with the pi4b?

jerryneedell, 2021-12-06:
Tried disabling CIRCUITPYTHON_USB_MSC in mpconfigport.mk but it did not change the behavior :-(

tannewt, 2021-12-06:
Thanks for trying! I think it'll require some printf debugging to see if it makes it into circuitpython at all (it should.)

PR #5676 (2021-12-08): Full UART support for RPI
This adds PL011 UART support which is more plentiful.

It also:
* Fixes PI4 build by including .dtb files on the SD card.
* Enables the activity LED as the status LED on PI4 and CM4 I/O.
* Adds that LED as board.LED.

Fixes #5650 and progress on #5629
commit 92a4261a (2021-12-08): Full UART support for RPI

This adds PL011 UART support which is more plentiful.

It also:
* Fixes PI4 build by including .dtb files on the SD card.
* Enables the activity LED as the status LED on PI4 and CM4 I/O.
* Adds that LED as board.LED.

Fixes #5650 and progress on #5629
closed 2021-12-08 by dhalbert