adafruit/circuitpython #5651: TINY2040 Onboard LED pins not available with Tiny2040 version of Circuitpython 7.0.0

BlueFinBima, 2021-12-03:
### CircuitPython version

```python
adafruit-circuitpython-pimoroni_tiny2040-en_GB-7.0.0
Adafruit CircuitPython 7.0.0 on 2021-09-20; Pimoroni Tiny 2040 with rp2040
Board ID:pimoroni_tiny2040
```

### Code/REPL

```python
import board, busio, adafruit_rgbled
led= adafruit_rgbled.RGBLED(board.GP18, board.GP19, board.GP20, invert_pwm=True)
```

### Behavior

Adafruit CircuitPython 7.0.0 on 2021-09-20; Pimoroni Tiny 2040 with rp2040
>>> import board, busio, adafruit_rgbled
>>> led= adafruit_rgbled.RGBLED(board.GP18, board.GP19, board.GP20, invert_pwm=True)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
**AttributeError: 'module' object has no attribute 'GP18'**
>>> 

### Description

The only information I have for the Tiny2040 from Pimoroni is that the on-board RGB LED is on GPIO18-20.  Happy to accept that I'm supposed to be referencing the LEDs via some other method, however it appears that GP18 through to GP20 have been disabled in CircuityPython for the Tiny2040.
When I use the Pi Pico firmware (same level), I can use the LED as I would expect.
From a user perspective, either the Tiny2040 firmware needs to stop restricting GP18,19,20 or Tiny2040 documentation needs to be improved to show the correct way to use the LED with CircuitPython (I found a howto for MicroPython, but I want to use CircuitPython).

### Additional information

Works fine with the adafruit-circuitpython-raspberry_**pi_pico**-en_GB-7.0.0 firmware.  

I saw PR #4796 from earlier this year and @Neradoc tested this or something similar.  Not sure what to make of that PR.

dhalbert, 2021-12-03:
The LED pins are given purpose-specific names, instead of GPIO18, 19, and 20. Here is the complete pin naming:

https://github.com/adafruit/circuitpython/blob/543e4d8c4d95871301d4526486e74692d8a36d1e/ports/raspberrypi/boards/pimoroni_tiny2040/pins.c#L6-L35

dhalbert, 2021-12-03:
You can also see the pins by doing `dir(board)`. The chip-level pin names are available with `microcontroller.pin`. (This is for a different board.)
```py
>>> import board,microcontroller.pin
>>> dir(board)
['__class__', '__name__', 'A0', 'A1', 'A2', 'A3', 'BUTTON', 'D0', 'D1', 'D10', 'D2', 'D3', 'D4', 'D5', 'D6', 'D7', 'D8', 'D9', 'I2C', 'MISO', 'MOSI', 'NEOPIXEL', 'NEOPIXEL_POWER', 'RX', 'SCK', 'SCL', 'SCL1', 'SDA', 'SDA1', 'SPI', 'TX', 'UART', 'board_id']
>>> dir(microcontroller.pin)
['__class__', 'GPIO0', 'GPIO1', 'GPIO10', 'GPIO11', 'GPIO12', 'GPIO13', 'GPIO14', 'GPIO15', 'GPIO16', 'GPIO17', 'GPIO18', 'GPIO19', 'GPIO2', 'GPIO20', 'GPIO21', 'GPIO22', 'GPIO23', 'GPIO24', 'GPIO25', 'GPIO26', 'GPIO27', 'GPIO28', 'GPIO29', 'GPIO3', 'GPIO4', 'GPIO5', 'GPIO6', 'GPIO7', 'GPIO8', 'GPIO9']
>>> 
```

BlueFinBima, 2021-12-03:
Hi Dan,  

many thanks for the quick response.  

I'm not sure how many of the CircuitPython users would make the jump from a MicroPython constant such as `MP_QSTR_LED_R` to needing to use board.LED_R without this information being documented (and I wasn't able to find it documented).
I don't know whether there is precedent for the LED_x naming convention in CircuitPython - a quick search did not confirm this - however I accept that is is logical if the board also uses the same conventions.  

Pimoroni only refer to the GP18,19,20 for the LEDs which is certain to cause confusion.  If you have contacts in Pimoroni, I would appreciate you emphasising the need for their documentation to match the CircuitPython Firmware for their device.  

#5160 seems to demonstrate a desire to get in sync, but to me it seems a little extreme to use the silkscreen as the definitive source of what is exposed / named in CircuitPython but the intent is welcomed.     

I will confess to being new to python, so I apologise if my comments highlight my experience with CircuitPython.  I really do appreciate the work you and the team does on CircuitPython, and I welcome any further comments/corrections on this mismatch.

Best wishes

BlueFinBima, 2021-12-03:
For anyone reading this with the same problem I experienced and running the Tiny2040 firmware level above, the intended syntax for using the internal RGB LED is as follows  

```python
import board, busio, adafruit_rgbled
led= adafruit_rgbled.RGBLED(board.LED_R, board.LED_G, board.LED_B, invert_pwm=True)
```

BlueFinBima, 2021-12-03:
Dan, thanks for your comments about `dir(board)` and `dir(microcontroller.pin)`.  I'd used the former before, but my inexperience did not immediately cause me to think about doing this.

dhalbert, 2021-12-03:
You're welcome! The Tiny2040 board definition (and other Pimoroni boards) were supplied by someone working with Pimoroni -- we did not make the pin choices, but they do go with our general philosophy. We rely on the manufacturers to document their pin names in their equivalent to a board's Learn Guide.

Closing this for now.

closed 2021-12-03 by dhalbert