adafruit/circuitpython #5658: M1 macOS Monterey USB enumeration issue

savt22, 2021-12-04:
### CircuitPython version

```python
Adafruit 7.0.0; issue present both on Pimoroni Keybow 2040 and Raspberry Pi Pico (with Pimoroni RGB Keypad).
```

### Code/REPL

```python
import time
from keybow2040 import Keybow2040
#from keybow_hardware.pim56x import PIM56X as Hardware # for Keybow 2040
from keybow_hardware.pim551 import PIM551 as Hardware # for Pico RGB Keypad Base

import usb_midi
import adafruit_midi
from adafruit_midi.note_off import NoteOff
from adafruit_midi.note_on import NoteOn

keybow = Keybow2040(Hardware())
keys = keybow.keys

midi = adafruit_midi.MIDI(midi_out=usb_midi.ports[1], out_channel=0)
rgb = (0, 0, 0)

#sets keyboard layout, creates slight flicker when holding down button. White and black notes on piano different colours 
while True:
    keybow.update()

    for key in keys:
        if key.pressed:
            key.set_led(*rgb)
        else:
            keybow.set_led(0, 100, 100, 100)
            keybow.set_led(1, 50, 100, 10)
            keybow.set_led(2, 100, 100, 100)
            keybow.set_led(3, 50, 100, 10)
            keybow.set_led(4, 100, 100, 100)
            keybow.set_led(5, 100, 100, 100)
            keybow.set_led(6, 50, 100, 10)
            keybow.set_led(7, 100, 100, 100)
            keybow.set_led(8, 50, 100, 10)
            keybow.set_led(9, 100, 100, 100)
            keybow.set_led(10, 50, 100, 10)
            keybow.set_led(11, 100, 100, 100)
            keybow.set_led(12, 200, 100, 100)
            keybow.set_led(13, 100, 200, 100)
            keybow.set_led(14, 100, 100, 200)
            keybow.set_led(15, 100, 50, 100)

#stops hanging notes 

    if keys[15].pressed:
        M = keys[15]
        @keybow.on_release(M)
        def release_handler(M):

            midi.send([NoteOff(60, 0),
                       NoteOff(61, 0),
                       NoteOff(62, 0),
                       NoteOff(63, 0),
                       NoteOff(64, 0),
                       NoteOff(65, 0),
                       NoteOff(66, 0),
                       NoteOff(67, 0),
                       NoteOff(68, 0),
                       NoteOff(69, 0),
                       NoteOff(70, 0),
                       NoteOff(71, 0),
                       NoteOff(72, 0),
[... 1528 lines cut ...]
It works as expected on Windows, and if I flash the RP2040 and code it using sudo nano code.py on MacOS, it works too.

However, as soon as I do a hard reset or restart my Mac, the Pico or Keybow is no longer recognised.

### Description

On the Console, 6 kernel errors appear when plugging the programmed Pico or Keybow:

130345.648474 IOUSBHostDevice@02241000: IOUSBHostDevice::start_block_invoke: device descriptor fragment is invalid

130345.651657 IOUSBHostFamily::validateEndpointMaxPacketSize: USB 2.0 5.[5-8].3: endpoint 0x00 invalid wMaxPacketSize 0x0000

130345.651866 AppleUSB20HubPort@02241000: AppleUSBHostPort::enumerateDeviceComplete_block_invoke: enumeration failed

130345.651890 AppleUSB20HubPort@02241000: AppleUSBHostPort::terminateDevice: destroying 0x0000/0000/0000 (IOUSBHostDevice): enumeration failure

130345.651963 AppleUSB20HubPort@02241000: AppleUSBHostPort::disconnect: persistent enumeration failures

130345.651990 IOUSBHostDevice@02241000: IOUSBHostDevice::start_block_invoke: device will not be registered for matching

### Additional information

Does anyone have any ideas on the issue? I have tried CircuitPython 6 and the 7.1 beta. I've tried creating a boot.py file that delays running code.py, disables USB_HID and enables usb_midi, but this had no effect. I have also tried plugging it in to an old Intel iMac running El Capitan, and it doesn't mount on that either. 

Everything initially works as usual on the Mac, the RPi2 drive mounts on first connection and holding BOOTSEL. After flashing with the CircuitPython uf2 it reboots as CIRCUITPY, and mounts without issue. But after adding in my code and the required libraries and files to lib (Adafruit_midi, Keybow_hardware, Adafruit_dotstar.mpy, Keybow.py) and putting my code in code.py, it no longer mounts and presents the above errors in console.

dhalbert, 2021-12-05:
> I've tried creating a boot.py file that delays running code.py, disables USB_HID and enables usb_midi, but this had no effect. I have also tried plugging it in to an old Intel iMac running El Capitan, and it doesn't mount on that either.

Delay in boot.py won't really change USB startup, because the device doesn't present as USB until boot.py has finished running. Have you tried just inserting a several-second sleep at the beginning of `code.py`?

Try a really minimal example that just sends a single MIDI note every second, or on a button push, or something like that. Use the minimum number of libraries. I looked at the Keybow library and I don't see anything suspicious, but it's good to narrow the problem down.

Also try some different USB cables. If you are using a USB hub, try not using it. Try to eliminate anything shared between the older  iMac and the new Mac.

savt22, 2021-12-07:
Hi Dan, 

Thank you for your help. I had a play around yesterday evening and I have narrowed down the problem. 

I tried with and without a dock, with different cables, and then I started playing round with the code. The Keybow midi example from their GitHub worked fine, and mounted without issue. So that ruled out it being a general library issue as my code uses the same libraries and builds on that example. 

So I started adding more bits from my code until it stopped working.

If I remove the bottom three modifier key combinations, below, it mounts on Mac without issue. With the multiple keypress modifiers, in addition to not mounting, it also appears to temporarily crash the Mac's audio codec. 

#majadd9
    if keys[15].pressed and not keys[14].pressed \
    and keys[13].pressed and not keys[12].pressed:
#minadd9
    if not keys[15].pressed and keys[14].pressed \
    and not keys[13].pressed and keys[12].pressed:
#halfdim 
    if keys[15].pressed and keys[14].pressed \
    and keys[13].pressed and keys[12].pressed:

This is my first project, so it's likely there is a better way of doing what I've done. But these key press combos do work without issue on Windows.

dhalbert, 2021-12-09:
It looks like you defined some keyrelease handlers only if keys 12-15 are pressed in a certain combination. That would happen on startup only. Is that what you want to do?

If you don't and import and don't use the Keybow library at all, then is there no crash? I am trying to figure out if it's Keybow-related or MIDI-related.

savt22, 2021-12-10:
Here is a video showing the functionality of keys 12-15: https://www.youtube.com/watch?v=sJZyv5DZECY

To help narrow it down, I rewrote it so that I didn't import usb_midi or adafruit_midi. I kept the modifier key combos the same, but programmed it to just change the colour of the keys rather than play midi notes/chords: 

import time
time.sleep(5)
from keybow2040 import Keybow2040
from keybow_hardware.pim56x import PIM56X as Hardware

keybow = Keybow2040(Hardware())
keys = keybow.keys

while True:
    keybow.update()

    if keys[15].pressed and not keys[14].pressed \
    and not keys[13].pressed and not keys[12].pressed:
        C = keys[0]
        @keybow.on_press(C)
        def press_handler(C):
            keybow.set_all(255, 225, 255)

    if not keys[15].pressed and keys[14].pressed \
    and not keys[13].pressed and not keys[12].pressed:
        C = keys[0]
        @keybow.on_press(C)
        def press_handler(C):
            keybow.set_all(0, 50, 255)

    if not keys[15].pressed and not keys[14].pressed \
    and keys[13].pressed and not keys[12].pressed:
        C = keys[0]
        @keybow.on_press(C)
        def press_handler(C):
            keybow.set_all(50, 0, 100)

    if not keys[15].pressed and not keys[14].pressed \
    and not keys[13].pressed and keys[12].pressed:
        C = keys[0]
        @keybow.on_press(C)
        def press_handler(C):
            keybow.set_all(255, 0, 50)

#modifier key combos that were causing the Mac issues when using it to play midi chords
    if keys[15].pressed and not keys[14].pressed \
    and keys[13].pressed and not keys[12].pressed:
        C = keys[0]
        @keybow.on_press(C)
        def press_handler(C):
            keybow.set_all(100, 0, 255)

    if not keys[15].pressed and keys[14].pressed \
    and not keys[13].pressed and keys[12].pressed:
        C = keys[0]
        @keybow.on_press(C)
        def press_handler(C):
            keybow.set_all(255, 0, 0)

    if keys[15].pressed and keys[14].pressed \
    and keys[13].pressed and keys[12].pressed:
        C = keys[0]
        @keybow.on_press(C)
        def press_handler(C):
            keybow.set_all(0, 0, 255)

This code runs completely fine on Mac. So, if I don't import usb_midi or adafruit_midi, but still import Keybow and include the modifier combination key presses, it enumerates and functions as intended. 

I think it must be related specifically to midi, then. As the issue is only present when modifier key combinations are used to define midi chords. 

I hope this helps, let me know if there is anything else you would like me to test.