adafruit/circuitpython #5662: ESP32S3: `import wifi` crashes circuitpython

jepler, 2021-12-04:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0-alpha.0-5-g73ee59881-dirty on 2021-12-04; ESP32-S3-DevKitC-1 with ESP32S3
```

### Code/REPL

```python
import wifi
```

### Behavior

On the REPL:
```
Adafruit CircuitPython 7.2.0-alpha.0-5-g73ee59881-dirty on 2021-12-04; ESP32-S3-DevKitC-1 with ESP32S3
>>> import wifi

[tio 09:24:44] Disconnected
```

On the serial debug console (build with DEBUG=1):
```
I (9913) pp: pp rom version: e7ae62f
I (9923) net80211: net80211 rom version: e7ae62f
I (9933) wifi:wifi driver task: 3fce4ecc, prio:23, stack:3584, core=0
I (9933) system_api: Base MAC address is not set
I (9933) system_api: read default base MAC address from EFUSE
I (9933) wifi:wifi firmware version: f84e709
I (9943) wifi:wifi certification version: v7.0
I (9943) wifi:config NVS flash: disabled
I (9943) wifi:config nano formating: disabled
I (9953) wifi:Init data frame dynamic rx buffer num: 8
I (9953) wifi:Init management frame dynamic rx buffer num: 8
I (9963) wifi:Init management short buffer num: 32
I (9963) wifi:Init dynamic tx buffer num: 16
I (9973) wifi:Init tx cache buffer num: 16
I (9973) wifi:Init static tx FG buffer num: 2
I (9973) wifi:Init static rx buffer size: 1600
I (9983) wifi:Init static rx buffer num: 4
I (9983) wifi:Init dynamic rx buffer num: 8
I (9993) wifi_init: rx ba win: 6
I (9993) wifi_init: tcpip mbox: 32
I (9993) wifi_init: udp mbox: 6
I (10003) wifi_init: tcp mbox: 6
I (10003) wifi_init: tcp tx win: 2880
I (10013) wifi_init: tcp rx win: 2880
I (10013) wifi_init: tcp mss: 1440
I (10013) wifi_init: WiFi IRAM OP enabled
I (10023) wifi_init: WiFi RX IRAM OP enabled
I (10033) phy_init: phy_version 302,ede2b68,Oct 20 2021,19:34:11
E (10033) phy_init: esp_phy_load_cal_data_from_nvs: NVS has not been initialized. Call nvs_flash_init before starting WiFi/BT.
W (10043) phy_init: failed to load RF calibration data (0x1101), falling back to full calibration
I (10093) wifi:set rx active PTI: 0, rx ack PTI: 0, and default PTI: 0
I (10093) wifi:mode : sta (7c:df:a1:e0:73:d8)
I (10093) wifi:enable tsf
W (10093) wifi: sta start
```

### Description

As also reported by discord user "wgp", wifi doesn't seem to work at all with the esp32-s3

dhalbert, 2021-12-05:
Reproduced on ESP32-S3 Box as well.

RetiredWizard, 2022-01-04:
I'm seeing the same thing on an  ESP32-S3-DevKitC-1:

Adafruit CircuitPython 7.2.0-alpha.1-16-gf2f38c7df-dirty on 2022-01-03; ESP32-S3-DevKitC-1 with ESP32S3

RetiredWizard, 2022-01-07:
The S3 Dev Kit I'm using has two micro USB connectors, one shows up as /dev/ttyUSB0 and the other as /dev/ttyACM0. CircuitPython grabs the ACM0 port for the REPL and the DEBUG=1 output goes to the USB0 port. 

After placing some mp_printf statements that go to the USB0 port and some ESP_LOGW statements that go to the ACM0 port, it looks to me like the micro controller is not hanging, it's just reseting the /dev/ttyACM0 port. The device disappears from linux but the USB0 port is still there and continues to print debug information out until it has nothing more to do.

I've placed print statements to both ports before and after the esp_wifi_start statement and here are the results:

/dev/ttyACM0

         Adafruit CircuitPython 7.2.0-alpha.1-39-g513a954d9-dirty on 2022-01-07; ESP32-S3-DevKitC3
         >>> import wifi
         Before esp_wifi_start

/dev/ttyUSB0

         I (19045) wifi_init: udp mbox: 6
         I (19045) wifi_init: tcp mbox: 6
         I (19055) wifi_init: tcp tx win: 2880
         I (19055) wifi_init: tcp rx win: 2880
         I (19065) wifi_init: tcp mss: 1440
         I (19065) wifi_init: WiFi IRAM OP enabled
         I (19065) wifi_init: WiFi RX IRAM OP enabled
         W (19075) myRadio: Before esp_wifi_start
         I (19075) phy_init: phy_version 302,3ddfab3,Dec  3 2021,14:39:12
         E (19085) phy_init: esp_phy_load_cal_data_from_nvs: NVS has not been initialized. Call nvs_flash_init before starting WiFi/BT.
         W (19095) phy_init: failed to load RF calibration data (0x1101), falling back to full calibration
         I (19145) wifi:mode : sta (7c:df:a1:e0:71:34)
         I (19145) wifi:enable tsf
         W (19145) wifi: sta start
         W (19145) myRadio: After esp_wifi_start

From the DEBUG=1 messages it looks to me like the esp_wifi_start completed and program control continued. The REPL interface was somehow reset though so the user is disconnected from the microcontroller.

RetiredWizard, 2022-01-07:
Just for kicks, I modified an rgbblink program so that along with blinking the neopixel it manipulated a gpio pin in a way that the DEBUG output was updated and then added an "import wifi" statement.

When I imported the updated rgbblink from the REPL, the REPL froze up, but the neopixel kept blinking and the debug output continued to display updates for the GPIO pin updates.

tannewt, 2022-01-07:
@hathach Sounds like this could be a USB issue.

hathach, 2022-01-10:
tinyuf2 seems to work on S3 as well on S2. Maybe there is some conflict, I will try to test this out tomorrow and see if I could spot anything.

closed 2022-01-14 by tannewt

casainho, 2023-07-06:
Please reopen this issue, as I have it on ESP32-S3 DevKitC N8R2, with CP 8.2.0.

I did apply this PR #5858 on CP 8.2.x and it solved my issue!!

PR #5858 (2022-01-14): Fix USB when enabling wifi
Fixes #5662