adafruit/circuitpython #5682: Unknown crash with invalid exception object

rbrian, 2021-12-08:
### CircuitPython version

```python
CircuitPython 7.1.0 beta1, on Feather RP2040
```

### Code/REPL

```python
See attached zip. Haven't been able to find a more minimal example.
```

### Behavior

When powering up the board with no serial monitor attached (was using TeraTerm), it will more often than not proceed normally for a few seconds, then throw an exception. Inside the exception handler, referring to the exception fields would throw another exception (in 7.0.0) - in 7.1.0b1, str(err) returns an empty string. 

### Description

With serial monitor attached, it didn't fail in 30 or more tries. Without, it's well over 50% (although not 100%). If it starts OK, then it'll be OK after a reset - it takes failing when power is first applied. 

Secondary issues, not sure if they're related: 

- Have had multiple instances where the file system got corrupted while cycling the USB connection. Re-flashing the UF2 didn't help, drive was still bad. Had to go to safe mode and use storage.erase_filesystem() from the repl. 
- Occasional (3-4 times total) got a message something like "No pullup on SDA/SCL, please check your connections" on the line initializing the accelerometer. Took a power cycle to fix (would still happen after reset button).  Accelerometer functions properly the rest of the time. 

### Additional information

Hardware: Feather 2040 with PropMaker feather wing, string of 20 neopixel "buttons" plugged into the wing's neopixel port. Switch attached to the wing's "switch" pads to provide read-only file system for the file logger to work.

[Code.zip](https://github.com/adafruit/circuitpython/files/7672699/Code.zip)

rsbohn, 2021-12-08:
This sounds like it could be a power problem. Can you replicate it with the neopixel string disconnected?

dhalbert, 2021-12-22:
We fixed an exception problem in 7.1.0-beta.3. Did you upgrade? Maye that fixed it.

rbrian, 2021-12-24:
I've been unable to reproduce. What made it go away was unplugging and replugging the neopixel chain. Must have been a strange electrical problem from having the plug slightly loose. I'll reopen if I see it again.

closed 2021-12-24 by rbrian