adafruit/circuitpython #5685: PIO ASM Pin Mapping Adafruit ItsyBitsy RP2040

willishf, 2021-12-08:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0 on 2021-09-20; Adafruit ItsyBitsy RP2040 with rp2040
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2021 Jeff Epler, written for Adafruit Industries
#
# SPDX-License-Identifier: MIT
#
# Adapted from the example https://github.com/raspberrypi/pico-exa ... /hello_pio

import time
import board
import rp2pio
import adafruit_pioasm

fh_clock_cycle = adafruit_pioasm.assemble(
"""
set pins, 0
set pins, 1
"""
)

sh_clock_cycle = adafruit_pioasm.assemble(
"""
set pins, 0
set pins, 1
"""
)

print("Starting")

sh = rp2pio.StateMachine(
sh_clock_cycle, frequency=800000, first_set_pin=board.D4
)

fh = rp2pio.StateMachine(
fh_clock_cycle, frequency=100000, first_set_pin=board.D0
)

while True:
time.sleep(1)
```

### Behavior

Working with PIO to do some clock timing. First example to output with PIO ASM to D4 worked as expected. Trying to add another clock output to D3, D2 etc didn't work. Spent lots of time trying to figure out what was not working in the software.

Started probing around other pins and found the clock cycle that should have been on DO was on D1. Started testing different DX pins and then probing pins would find the clock signal.

Download the appropriate packages for the board so don't think this is a case of the wrong board library.

The following code changing D0 to other pins should allow a review for correctness of pin mapping. D4 for whatever reason is correctly output to D4.

dhalbert, 2021-12-09:
`D0` is `GPIO1`, and is marked as `RX` on the board (and `GPIO1` on the bottom). `D1` is `GPIO0` and is marked as `TX` on the board. I'm not sure, were you expecting `GPIO0` to be marked as [D0](url)? The [Dn](url) pins are consistent across ItsyBitsy boards, but don't correspond to the `GPIOn` pins. `D4` happens to correspond to `GPIO4`, but that is kind of a coincidence.

willishf, 2021-12-09:
> expecting GPIO0 to be marked as D0 <http://url>? The Dn <http://url> pins

Thanks for the clarification I missed the connection between pin number on
the board and other naming conventions.

closed 2021-12-09 by tannewt