adafruit/circuitpython #5689: Support for nrf52833 internal pull ups and pull downs.

scath999, 2021-12-09:
The nrf52833 series of boards uses internal pull ups and pull downs to configure its pins for use with peripherals. 

Currently, there does not appear to be a way to set these internal pull ups on selected pins to make use of them using `busio.I2C`.

Pending the outcome of this feature request, does anyone have any workarounds or ideas?

dhalbert, 2021-12-09:
We discussed this to some extent in #4488, which was to use the RP2040 internal pullups, but make it a compile-time. Those pullups are considerably weaker (50-80kohms), and so they weren't as useful for I2C. The nRF pullups are 11-16kohms.

You could swipe the code idea from that PR, though think you want to make it a run-time option in the long run.

scath999, 2021-12-09:
@dhalbert Thanks for directing me to that issue.

Is it ok to leave this open while I do some research and test that PR out?

dhalbert, 2021-12-09:
> Is it ok to leave this open while I do some research and test that PR out?

Sure, it's an interesting feature request.

scath999, 2021-12-09:
I've taken a look at the issue(s) outlined, and the PR submitted. Couple of questions when you have a moment:

1) I'm sure it's my own lack of familiarity with git and the CPY repo itself, but I can't find where those changes are in the code. They don't appear on my side, nor do they appear in [mpconfigboard.h](https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/boards/pimoroni_keybow2040/mpconfigboard.h), which is where issue [#4873](https://github.com/adafruit/circuitpython/pull/4378) appears to make changes.

2) I can find the section of I2C.c that checks for the pullups, and I'm thinking those checks are good to keep in place. They'll alert me (or someone else) if I've forgotten to set the pull `up` on a pin being used with i2c. Am I thinking about this right?

3) The proposed solution appears to make use of bitbangio - a module not included in nrf/common-hal/. Guidance on how to include this is welcome, as I don't think it will be as simple as copying that module over from some other port.

Cheers!

dhalbert, 2021-12-09:
Re 1.: There were some proposed changes in one of the commits in #4488: https://github.com/adafruit/circuitpython/pull/4488/commits/29bf9e71a0a9e75c2f771077c9b36af3af041297. That code was never added.

Re 2.: If internal pullups are enabled, we don't need to check for pullups. See the code in 1.

Re 3.: The `bitbangio` stuff is separate, it's about a flaw in the RP2040 I2C peripheral, and isn't relevant to nRF. By the way, `bitbangio` is in `shared-module`, because it's not port-dependent.

dhalbert, 2021-12-09:
That's fine. I understand, and the example I pointed to you was for "all time" setting. So what would make sense is to add an optional `pullup` argument to `busio.I2C()` and `bitbangio.I2C()`, that defaults to False. It would get passed to all the `common_hal` I2C creation routines.

Is there any case where you want a pull-down? For I2C, no, but are there other peripherals for which you want a pull-down?

scath999, 2021-12-10:
I agree with your thoughts here (and would defer to your better knowledge and experience anyway).

I'm trying to imagine a place where someone might want to change a pullup to a pulldown, but I'm not experienced enough here. I defer to the better experience and knowledge of the community - would love to hear their thoughts on this.

I'm going to clone a fresh copy of CPY (mine is quite dirty as I've played with the code over the past month). I'll play around with it and see if I can come up with something that makes sense. If I do, I'll put a PR together.

scath999, 2021-12-10:
Ok. I've reviewed what I think is the relevant code. 

If I'm understanding the process correctly, i2c implementations rely on definitions for the `DEFAULT_I2C_BUS_` values for the Data and Clock channels to be declared in the boards `mpconfigboard.h` file.

That doesn't happen in the pca10100 mpconfig, which makes sense when there are no predefined i2c channels on the board.

As @dhalbert mentioned, an ideal next step is to pass an optional parameter for the i2c on pca10100 that allows us to define the SDA and SCL channels we want to use, set the pull on those channels to `up`, and return the interface. 

The `busio_i2c_obj_t` struct contains two `digitalio_digitalinout_obj`s structs, each referencing the indicated pin (SDA and SCL). Those structs contain a `mcu_pin_obj_t` pointer, which (presumably) references the physical pin indicated. 
`digitalio_digitalinout_obj_t`s can be given a pull, apparently via `common_hal_digitalio_digitalinout_set_pull(
    digitalio_digitalinout_obj_t *self, digitalio_pull_t pull)`.

Is it reasonable to:
1) Declare two pins (one for SDA and SCL) using something like `SDA/SCL = digitalio.DigitalInOut(microcontroller.pin.<pin>)`
2) Set the `pull` on those pins to `up` using something like `SDA.pull = digitalio.PULL.UP`
3) Change the board_i2c method to accept digitalio objects as optional parameters, and check for a specific MCU (like the pca10100) pending the result.
4) Extract and pass the `mcu_pin_obj_t` pointers from each pin into the call to `common_hal_board_create_i2c()`
5) Change `common_hal_busio_i2c_construct()`s parameters to accept those pointers as arguments (if presented) or use the `DEFAULT_I2C_BUS_` arguments if not.
6) Allow execution to proceed from that point onward as normal.

I don't think this is what @dhalbert really envisioned, but perhaps its a starting point?

Cheers.

dhalbert, 2021-12-10:
I was thinking of something simpler:
```py
i2c = busio.I2C(board.SCL, board.SDA, pullup=True)
```
So the Python interface in `shared_bindings/busio/I2C.c` would add an optional keyword argument. Pass that on to `common_hal_busio_i2c_construct()`, which would take care of setting the pullups when it set up the pins, using the specific mechanisms provided by the HAL for the chip.

> The busio_i2c_obj_t struct contains two digitalio_digitalinout_obj structs
You are suggesting adding that, is that correct? It doesn't have those objects now. I would not do that. We can just set the pullups when we set up the I2C (TWIM) peripheral.

We would add this to all ports, not just nRF. We want the interface to be uniform across all ports. On some ports (RP2040), it would not work very well because the builtin pullups quite a hight resistance for I2C.

For `board.I2C()`, yes, there would be some specific compile-time flag to set the value of `pullup` for the default pins.

Also we would add it to `bitbangio.I2C`. That _does_ use `DigitalInOut` objects, in a port-independent way.

scath999, 2021-12-10:
Ah, I see. 

Along with solving the issue, I'm also trying to think about how people will use the interface - which I understand you're concerned about too, from your response.

I don't know about all cases, but that wouldn't help in my case since there are no pre-defined SCL and SDA channels (which partly inspired this feature request). 

This could be gotten around by allowing `board.SCL` and `board.SDA` arguments to be substituted with `microcontroller.pin.<pin>` but that takes us back to where we started, now with a layer of protection that is being escaped: 

What if someone is using external pullups and their wiring _does_ need to be checked? 

Does implementing the solution at such a high level create a possible danger for people who aren't using mcus with internal pullups/pulldowns, and may try to use that call in ignorance (as I did)? I don't know enough about this stuff to foresee a danger, but you do.

No, I wasn't suggesting adding the two digitalio obj structs to busio: You're right - the `busio_i2c_obj_t` contains a `bitbangio_i2c_obj_t`, which _then_ exposes the the `digitalio` objects; I missed that in my write up. 

Thoughts?

dhalbert, 2021-12-10:
`board.SCL` and `board.SDA` are just pin names, based on the silkscreen markings on the board. They could be `D0` and `D1`, `GPIO8` and `GPIO9`, etc. The pin objects in `microcontroller.Pin` are the same kinds of pin objects as in `board`.

> Does implementing the solution at such a high level create a possible danger for people who aren't using mcus with internal pullups/pulldowns, and may try to use that call in ignorance (as I did)?

All microcontrollers I know have at least pull-ups. If they didn't, we would throw an error when someone specified `pull=True`. The default will always be `pull=False`, for upward compatibility.

> No, I wasn't suggesting adding the two digitalio obj structs to busio: You're right - the `busio_i2c_obj_t` contains a `bitbangio_i2c_obj_t`, which _then_ exposes the the `digitalio` objects; I missed that in my write up. 

Only the `raspberrypi` includes a `bitbangio_i2c_obj_t` in its `busio.I2C` object. That is only because the I2C hardware on the RP2040 cannot do zero-length I2C writes. So we use bitbangio to do that if it's requested. Look at the `nrf` or `atmel-samd` ports for more typical implementations.

scath999, 2021-12-10:
> `board.SCL` and `board.SDA` are just pin names, based on the silkscreen markings on the board. They could be `D0` and `D1`, `GPIO8` and `GPIO9`, etc. The pin objects in `microcontroller.Pin` are the same kinds of pin objects as in `board`.
> 

Yes, I understand. That said, there is an issue with this for the 52833 (or at least my DK): D0 does not necessarily correspond to P0_00. (For example, on my board P0_20 is addressed by `board.D11`.) I haven't mapped out all of the correspondences yet, but can do that. 

> All microcontrollers I know have at least pull-ups. If they didn't, we would throw an error when someone specified `pull=True`. The default will always be `pull=False`, for upward compatibility.
> 

Fair enough. My question was more about trying to remove assumptions about how developers will use the interface, but I defer to your better experience here.

> Only the `raspberrypi` includes a `bitbangio_i2c_obj_t` in its `busio.I2C` object. That is only because the I2C hardware on the RP2040 cannot do zero-length I2C writes. So we use bitbangio to do that if it's requested. Look at the `nrf` or `atmel-samd` ports for more typical implementations.

This is confusing. I'm looking at the definition for `busio_i2c_obj_t` right now, and it appears in `circuitpython/shared-module/busio/I2C.h`

```
#ifndef MICROPY_INCLUDED_ATMEL_SAMD_SHARED_MODULE_BUSIO_I2C_H
#define MICROPY_INCLUDED_ATMEL_SAMD_SHARED_MODULE_BUSIO_I2C_H

#include "shared-module/bitbangio/I2C.h"

#include "py/obj.h"

typedef struct {
    mp_obj_base_t base;
    bitbangio_i2c_obj_t bitbang;
} busio_i2c_obj_t;

#endif // MICROPY_INCLUDED_ATMEL_SAMD_SHARED_MODULE_BUSIO_I2C_H
```

Please forgive my ignorance if I'm just reading that wrong, but it seems like this includes the `amtel-samd` ports.

Can you clarify?

To summarize, the solution seems completely doable (and reasonably quickly), but my biggest question is where to implement it and which methods are safest to use. It's clear that you're concerned about the same thing, so I think we're on the same page here.

Thanks again :)

dhalbert, 2021-12-10:
> I haven't mapped out all of the correspondences yet, but can do that.

Yes, you have to do that by reading the schematic to make up the `pins.c` file. Look at the nRF52840 DK board for something similar.

> This is confusing. I'm looking at the definition for `busio_i2c_obj_t` right now, and it appears in `circuitpython/shared-module/busio/I2C.h`

Ah, OK, that file is only for the case where there is no implementation of `busio.I2C`. In that case we fall back to `bitbangio.I2C`. The include guards are wrong, and should say `SHARED_MODULE`.

Look instead at say `ports/nrf/common-hal/busio/I2C.c`.

scath999, 2021-12-10:
> > I haven't mapped out all of the correspondences yet, but can do that.
> 
> Yes, you have to do that by reading the schematic to make up the `pins.c` file. Look at the nRF52840 DK board for something similar.

Can do. Will get back to you on this.

> Ah, OK, that file is only for the case where there is no implementation of `busio.I2C`. In that case we fall back to `bitbangio.I2C`. The include guards are wrong, and should say `SHARED_MODULE`. [...]

That directory doesn't exist in my `clone` of the repo: 

[image]

dhalbert, 2021-12-10:
Does the red mean the directory is empty?

https://github.com/adafruit/circuitpython/blob/main/ports/nrf/common-hal/busio/I2C.c should exist in your repo. Not sure what's wrong with your clone.

scath999, 2021-12-10:
Ignore that post - I missed the `/common-hal` in your suggestion lol

scath999, 2021-12-10:
> > I haven't mapped out all of the correspondences yet, but can do that.
> 
> Yes, you have to do that by reading the schematic to make up the `pins.c` file. Look at the nRF52840 DK board for something similar.

Can do - will get back to you on this.

> Ah, OK, that file is only for the case where there is no implementation of `busio.I2C`. In that case we fall back to `bitbangio.I2C`. The include guards are wrong, and should say `SHARED_MODULE`.
> 
> Look instead at say `ports/nrf/common-hal/busio/I2C.c`.

Ah, I see it now. It seems like this struct only stores a uint8 with the pin number. Is it safe/wise to change that to a `digitalio_digitalinout_obj_t`, or is there a better/safer way to add the ability to set the pull `up` (maybe somewhere further up or down the hal)?

tannewt, 2021-12-10:
> Ah, I see it now. It seems like this struct only stores a uint8 with the pin number. Is it safe/wise to change that to a `digitalio_digitalinout_obj_t`, or is there a better/safer way to add the ability to set the pull `up` (maybe somewhere further up or down the hal)?

I'd just copy how digitalio does it. No need to have a full object. common-hal can do whatever it wants to the mcu hal.

scath999, 2021-12-11:
Ok. 

I've written two test methods to see how they work. Posting here for some initial feedback.

Within `ports/nrf/shared-bindings/board/__init__.c`:
```
#if BOARD_I2C
#include "shared-bindings/busio/I2C.h"
#include "shared-bindings/digitalio/DigitalInOut.h"
#endif

(...) 

#if BOARD_I2C
(...)
mp_obj_t board_i2c_with_pins(digitalio_digitalinout_obj_t scl_pin, digitalio_digitalinout_obj_t sda_pin) {
    mp_obj_t singleton = common_hal_board_get_i2c();
    if (singleton != NULL && !common_hal_busio_i2c_deinited(singleton)) {
        return singleton;
    }

    assert_pin_free(sda_pin->pin);
    assert_pin_free(scl_pin->pin);
    return common_hal_board_create_i2c_with_pins(digitalio_digitalinout_obj_t scl_pin, digitalio_digitalinout_obj_t sda_pin);
}```

Within `ports/nrf/shared-module/board/__init__.c`:
```
mp_obj_t common_hal_board_create_i2c_with_pins(digitalio_digitalinout_obj_t scl_pin, digitalio_digitalinout_obj_t sda_pin) {
    // All callers have either already verified this or come so early that it can't be otherwise.
    assert(i2c_singleton == NULL || common_hal_busio_i2c_deinited(i2c_singleton));
    busio_i2c_obj_t *self = &i2c_obj;
    self->base.type = &busio_i2c_type;

    digitalio_pull_t pull = PULL_UP;

    common_hal_digitalio_digitalinout_set_pull(scl_pin, pull)
    common_hal_digitalio_digitalinout_set_pull(sda_pin, pull)

    common_hal_busio_i2c_construct(self, scl_pin, sda_pin, 100000, 255);
    i2c_singleton = (mp_obj_t)self;
    return i2c_singleton;
}```

This builds just fine, but I have no idea if make would throw and error if I missed an `include` or referenced something improperly.

I have not deployed to my board - hoping people can point out the big errors I'm probably making and save me some headaches ;)

Cheers.

dhalbert, 2021-12-11:
Don't add this to `board`. I am proposing to add an extra optional argument to `shared-bindings/busio/I2C.c`. That argument would then passed to the common-hal routines in `busio/I2C.c` in each port. You would not call `common_hal_digitalio_digitalinout_set_pull()`. Instead you would use the port-specific way of setting a pullup for each port. That is, it would be the code inside `common_hal_digitalio_digitalinout_set_pull()`.

scath999, 2021-12-11:
You're saying add the argument to `common_hal_busio_i2c_construct`?

tannewt, 2021-12-13:
@scath999 I'd suggest opening a draft PR with your potential changes. It's much easier for us to comment on code changes in a PR.

I think about you want `MP_QSTR_internal_pull` instead of `MP_QSTR_bool` because that's the name you'll use from Python. Then, you want to pass its value into common_hal. This is shared-bindings code that is port independent. common-hal code is port specific.

scath999, 2021-12-13:
@tannewt 

Thanks for getting back. I'm confused about passing the `pull` into `common_hal` at this point in the code: I don't know what method I'd call to do that, and there doesn't seem to be room in `common_hal_busio_i2c_construct`. 

@dhalbert mentioned setting the pull in a port-specific manner (see [this ](https://github.com/adafruit/circuitpython/issues/5689#issuecomment-990998010) and [this ](https://github.com/adafruit/circuitpython/issues/5689#issuecomment-991452336), but perhaps I misunderstood.

I appreciate your patience.

dhalbert, 2021-12-13:
> there doesn't seem to be room in `common_hal_busio_i2c_construct`.

Yup, `common_hal_busio_i2c_construct()` will need another argument added.

scath999, 2021-12-13:
Ah, I see. The method I'm currently working in is shared - I need to put this into the common-hal so it doesn't get run on a board that isn't my board?

dhalbert, 2021-12-14:
Each port has its own version of `common_hal_busio_i2c_construct()`. They all have the same signature, so you need to add the new argument to all of them. The version in `shared-module` is only a proxy for `bitbangio`.

As @tannewt mentioned, seeing your actual code would help. You could point us to a repo that you keep up to date.

scath999, 2021-12-14:
I've created a draft pull request for this issue [here](https://github.com/adafruit/circuitpython/pull/5718).

dhalbert, 2021-12-14:
> The nrf52833 series of boards uses internal pull ups and pull downs to configure its pins for use with peripherals.

When you say this, what do you mean? Do you mean the nRF52833 DK (PCA10100)? That board has no on-board I2C peripherals that I see. It resembles the other nRF DK boards, I think.

In general, external I2C breakouts have pull-ups provided. Adafruit's breakouts all have pullups. We don't put I2C pullups on our microcontroller dev boards unless there are on-board I2C peripherals.

So before we go further with the PR, could I understand the initial requirements?

dhalbert, 2021-12-15:
I'm including comments from the PR, from this comment: https://github.com/adafruit/circuitpython/pull/5718#issuecomment-994120624

>>Don't do this for all 52833 builds. I thought the reason for this was just that your particular board had no pullups for on-board I2C peripherals. Is that the case?

>I see your point. It was my understanding that all 52833 boards have this functionality, which appears to be a wrong assumption.

>>Ideally the [sensor] board would have on-board pullups.

>I totally get this, and I admit I'm concerned that the internal pullups won't be strong enough, but I'll need to show that adding them on is necessary before I ask the team to rework the board.

In general, most sensor data sheets spec a 10kohm pullup. Some sensors spec 4.7k, and we have found that necessary in some cases. The 11k-16k internal pullups on the nRF chips are a bit marginal. If this is a special-purpose board, you might be able to get away this, but if there's more than one sensor, you might want stronger pullups. In any case, it should be tested.

Some other chip families have even higher-value pullups, which are unsuitable for I2C

On our own boards, we use external pullups for all on-board sensors, and this is what I've seen on other microcontroller boards as well.

>> The microcontroller boards don't in general have pullups except when there are on-board sensors.

>A bit confused by this - do you mean sensors that are integrated into the chipset, or 3rd-party sensors that have been wired into the system? Are you asking because this is important to this case, or important for CPY and this feature? This is an honest question; I'm not doubting you, I just want to understand the implications of what I'm trying to add for my project, and CPY in general.

By "on-board sensors", I mean sensors in separate packages that are on the board in addition to the microcontroller.

So my question is whether your design team has decided that the internal nRF52833 pullups are "good enough" for the external sensors you are including on the board, and have they tested that, etc. Do you have examples of other commercial boards that rely on the internal pullups only for external sensors?

scath999, 2021-12-15:
> When you say this, what do you mean? Do you mean the nRF52833 DK (PCA10100)? That board has no on-board I2C peripherals that I see. It resembles the other nRF DK boards, I think.

In general, external I2C breakouts have pull-ups provided. Adafruit's breakouts all have pullups. We don't put I2C pullups on our microcontroller dev boards unless there are on-board I2C peripherals.

So before we go further with the PR, could I understand the initial requirements? This would be a worthy feature, but I just want to make sure that it is an unmeet need for your use cases.

Thanks for this. I understand better now, and I'll clarify:

The board we're building has multiple sensors that have been attached to different pins, and an internally made sensor as well. Two of those peripherals (there are more than two) are meant to use the I2C protocol, but are physically wired to different pins on the MCU.

I think that decision was made explicitly - I'm assuming the hardware team thought the level of resistance provided by the internal pullups would be sufficient. This may not end up being the case, but I'll need to prove it before asking them to solder on external pullups.

The sensors we're using don't appear to have pullups attached - I am confirming this with hardware, but this is my suspicion given my experience so far: if the pullups were already integrated into the POC, wouldn't CPY have detected them?

The goal is to be able to read these sensors in a sweep: turn each on in sequence, read and record data, then sleep and wait for a predetermined period of time before repeating the cycle. 

I'm inferring that this makes the 833 a natural choice: any channel can be used for I2C, so there are fewer design limitations. We can attach those two sensors to any pins and declare them as I2C channels - that's what I gather the idea is, anyway.

To answer your question about other boards, this is a new design and this team has not put any other boards into the commercial market. Our senior hardware developers have a lot of experience, but then again they approved this design.

Does this help?

Thanks again :)

dhalbert, 2021-12-15:
> The sensors we're using don't appear to have pullups attached - I am confirming this with hardware, but this is my suspicion given my experience so far: if the pullups were already integrated into the POC, wouldn't CPY have detected them?

I haven't seen any external sensors incorporate pullups on chip: they are always supplied externally. So this makes sense.

There is some discussion of using the internal pullups on nRF chips in these Nordic DevZone threads (and others, but these are the most interesting). None of these are using the nRF52833 but the similiar nRF chips all use similar peripherals, with the internal pullups, which are similar in the chips. In the Nordic SDK (which we don't use), the internal pullups are enabled by default. The '833 is like the most of the other modern nRF chips: it has the full crossbar you mention, so that any pins can be used for the I2C peripherals on-chip.

The '833 is mostly a cut-down '840, which we are quite familiar with. We are also familiar with the '832, which we used in several earlier designs.

https://devzone.nordicsemi.com/f/nordic-q-a/10899/does-nrf51-has-optional-internal-pullup-resistors-on-some-pins
https://devzone.nordicsemi.com/f/nordic-q-a/52455/does-nrf9160-have-internal-pull-up-resistors-for-i2c-twim
https://devzone.nordicsemi.com/f/nordic-q-a/81006/i2c-bus-spikes
https://devzone.nordicsemi.com/f/nordic-q-a/66459/external-pull-ups-i2c

The general impression I get is that the internal pullups are generally adequate for one or two devices, when running at 100kHz. But if you're using a higher I2C frequency, or there are multiple devices on a bus, then it might be marginal.

I don't know what frequency you're using, but if the hardware designers made that decision, the sensors are not spec'd for lower-value pull-ups, and assuming that the design was tested, then it's probably all right.

scath999, 2021-12-16:
I've managed to get a version that incorporates all the changes suggested by @dhalbert and @microDev1 and updated my draft PR - it's going through checks right now.

Testing it locally, the new method signature works and I'm able to get an `i2c` interface using `busio.i2c(scl, sda, internal_pullup = True)`

However, when I run `i2c.scan()` I get an empty array back. Not sure what I've missed here, but pretty darn skippy I should get something back from that call.

Thoughts (when you have time)?

dhalbert, 2021-12-16:
I haven't looked at the code yet, but do you measure 3.3V at the pins, indicating that they are indeed being pulled up? Have you looked at the waveforms?

scath999, 2021-12-16:
@dhalbert Short answer: yes, and I get a voltage but no device addresses.

The steps required are as follows:

1) Set the sensor Power Pin to 1: When I measure voltage on the declared I2C pins I get ~3.3V
2) Create an I2C interface and look for device addresses using the following code

```
scl = microcontroller.pin.P0_21
sda = microcontroller.pin.P0_20
i2c = busio.I2C(scl, sda, internal_pullups = True)

while not i2c.try_lock():
    pass

try:
    while True:
        print(
            "I2C addresses found:",
            [hex(device_address) for device_address in i2c.scan()],
        )
        time.sleep(2)

finally:  # unlock the i2c bus when ctrl-c'ing out of the loop
    i2c.unlock()
```

The output always `I2C addresses found: []`

Note: I have all the edits you suggested in the draft PR.

dhalbert, 2023-10-28:
Closing this because no further work has been done, and the internal pullups are too weak for general use.

closed 2023-10-28 by dhalbert