adafruit/circuitpython #5703: Todays (2021-12-10) build of CP for Raspberry Pi 4B does not work

dronecz, 2021-12-10:
### CircuitPython version

```python
latest CircuitPython 7.1.0-beta.2
```

### Code/REPL

```python
none
```

### Behavior

Pi does not boot up

### Description

I download latest build from [web](https://circuitpython.org/board/raspberrypi_pi4b/) and when I flash it to the SD card, I can see these files: 

[image]

When I try older build (7.2.0-alpha.0-102-gb50443240 on 2021-12-10), Pi works OK and I can see these files at my SD card. 

[image]

I hope this info help. Thanks for great work!

jerryneedell, 2021-12-10:
You probably don't want a build from the 7.1.x branch. Try the latest build from 7.2.0-alpha. It should be the previous build to the beta version in the list at the S3 site.

jerryneedell, 2021-12-12:
FYI -- I just tried the latest build and it does boot for me -- the Mass Stage Drive (CIRCUITPY) is still not working, but I get a REPL on the UART.

tannewt, 2021-12-13:
@jerryneedell is correct. The `.dtb` files were added after the 7.1.0 branch so please use the absolute newest builds. (We'll do a 7.2.0 unstable release once 7.1.0 is stable.)

closed 2021-12-13 by tannewt