adafruit/circuitpython #5704: traceback.print_exception causes crash on user-defined exceptions

timhawes, 2021-12-11:
### CircuitPython version

```python
Adafruit CircuitPython 7.0.0 on 2021-09-20; FeatherS2 with ESP32S2
Adafruit CircuitPython 7.1.0-beta.2 on 2021-12-10; FeatherS2 with ESP32S2
```

### Code/REPL

```python
import traceback

class UserDefinedException(Exception):
    pass

try:
    raise UserDefinedException("test")
except Exception as e:
    traceback.print_exception(None, e, e.__traceback__)
```

### Behavior

CircuitPython crashes and reboots on the traceback.print_exception call.

### Description

traceback.print_exception causes a crash when called on user-defined exceptions. Builtin exceptions are handled as expected.

### Additional information

Tested on ESP32-S2, was not able to replicate it on RP2040.

tannewt, 2021-12-13:
@microDev1 Any idea? You've been in the exception code most recently.

timhawes, 2021-12-13:
I did some more debugging on this. It also affects RP2040 (and maybe others) but only caused a hard crash on espressif. On RP2040 it finds that exc->traceback is NULL (or corrupt) and only fails to print the traceback.

I believe the cause is related to the way that mp_obj_exception_add_traceback and mp_obj_exception_get_traceback both operate on the native exception object (by calling get_native_exception) but the traceback module attaches the traceback directly to the exception it is given as parameter.

microDev1, 2021-12-14:
> @microDev1 Any idea? You've been in the exception code most recently.

I took a look at this but I have no idea what is going on here.

On `rp2040`:
- `UserDefinedException: test` is printed in REPL and CPY doesn't crash.

On `espressif`:
- In normal build, CPY crashes and reboots.
- In debug build, no backtrace is available and `<traceback object at 0x...>` is printed in REPL and CPY doesn't crash.

timhawes, 2021-12-14:
I think traceback_exception_common (in shared-bindings/traceback/__init__.c) has a broken pointer. I have a working fix and will make a PR.

closed 2021-12-15 by timhawes