adafruit/circuitpython #5705: DisplayIO not drawing on screen in 7.1.0 Beta 2

FoamyGuy, 2021-12-11:
### CircuitPython version

```python
Adafruit CircuitPython 7.1.0-beta.2 on 2021-12-10; Adafruit PyPortal with samd51j20
Board ID:pyportal
```

### Code/REPL

```python
All displayio example code that I have tried does not show anything on the screen.

label simpletest:

import board
import terminalio
from adafruit_display_text import label

text = "Hello world"
text_area = label.Label(terminalio.FONT, text=text)
text_area.x = 10
text_area.y = 10
board.DISPLAY.show(text_area)
while True:
    pass
```

ondiskbitmap example from learn:
```
import board
import displayio

display = board.DISPLAY

# Current method valid for CircuitPython 6 & 7

# Open the file
with open("/purple.bmp", "rb") as bitmap_file:

    # Setup the file as the bitmap data source
    bitmap = displayio.OnDiskBitmap(bitmap_file)

    # Create a TileGrid to hold the bitmap
    tile_grid = displayio.TileGrid(
        bitmap,
        pixel_shader=getattr(bitmap, 'pixel_shader', displayio.ColorConverter())
    )

    # Create a Group to hold the TileGrid
    group = displayio.Group()

    # Add the TileGrid to the Group
    group.append(tile_grid)

    # Add the Group to the Display
    display.show(group)

    # Loop forever so you can enjoy your image
    while True:
        pass
```

vectorio example code from readthedocs:
```
import displayio
import vectorio
import board

display = board.DISPLAY

group = displayio.Group()

palette = displayio.Palette(1)
palette[0] = 0x125690

circle = vectorio.Circle(pixel_shader=palette, radius=25, x=70, y=40)
group.append(circle)

rectangle = vectorio.Rectangle(pixel_shader=palette, width=40, height=30, x=55, y=45)
group.append(rectangle)

points=[(5, 5), (100, 20), (20, 20), (20, 100)]
polygon = vectorio.Polygon(pixel_shader=palette, points=points, x=0, y=0)
group.append(polygon)

display.show(group)
while True:
    pass
```
```

### Behavior

The display seems to "freeze" showing the last of the serial output from before the script runs. The old serial output remains visible on the display, but the screen is never taken over with the displayio objects and any further prints to serial do not get shown, which does make some sense because it's presumably trying to show displayio things not the serial output any more.

### Description

So far every displayio example that I have tried has had the same behavior with "freezing" the serial output on the screen and not displaying any displayio objects. I have not tried 100% of everything but I suspect this probably does affect anything that can be shown with displayio.

I've also noticed if you cold boot with an empty code.py file and open the serial connection. The first time you press ctrl-C you will see the normal stuff get printed on the display:
![image](https://user-images.githubusercontent.com/2406189/145688052-c5649307-6468-4053-95a3-eafedebd3130.png)

But from that point it gets "frozen" nothing else will ever change the display. Pressing ctrl-C more times will not print any additional `>>>` and running something like `print("hello") from the repl will not show "Hello" get printed on the display. All of these outputs do appear normally in the serial communication channel so this seems to effect only display output. The actual serial communication does appear to be working properly.

### Additional information

displayio examples do work as expected on:
```
Adafruit CircuitPython 7.1.0-beta.1 on 2021-11-30; Adafruit PyPortal with samd51j20
Board ID:pyportal
```
So something between `beta.1` and `beta.2` seems to have caused this issue.

FoamyGuy, 2021-12-11:
I have done some bisecting and determined that this is the most recent build which works correctly:
```
Adafruit CircuitPython 7.1.0-beta.1-14-gfea0103ff on 2021-12-01; Adafruit PyPortal with samd51j20
Board ID:pyportal
```

And this is the first build which starts failing as described above:
```
Adafruit CircuitPython 7.1.0-beta.1-17-g8cf269d2a on 2021-12-01; Adafruit PyPortal with samd51j20
Board ID:pyportal
```

At present time all builds above `8cf269d.uf2` in the PyPortal US S3 list are failing as described above.

rsbohn, 2021-12-11:
I tried it on the pybadge, wasn't able to reproduce. My display_shapes and vectorio scripts both work.
On the other hand, auto-reload doesn't seem to work after `7.1.0-beta.1-14-gfea0103ff on 2021-12-01`.

I first noticed the auto-reload problem with 7.1.0-beta.2 on the feather-M4.

FoamyGuy, 2021-12-11:
This is the commit where the behavior changes I think:
https://github.com/adafruit/circuitpython/commit/8cf269d2a071a59314a6f07bf497b444af2264e5

Of the changes in there my guess was that the high level switch here was related:
https://github.com/adafruit/circuitpython/blob/b50443240f762b84616eed77d928fe26b056638a/ports/atmel-samd/mpconfigport.mk#L123

Setting that back to `0` and re-building does seem to make the display go back to behaving normally. The REPL does keep printing new things and the displayio examples above render properly.

It will probably be better to try to understand what about the ALARM is causing the issues to fix it there then to simply disable it like this though.

dhalbert, 2021-12-12:
I narrowed this down to https://github.com/adafruit/circuitpython/blob/f695b14e932310b44feda20653c3574bba514071/main.c#L173.
Commenting that out makes `vectorio` work again, the terminal show up, etc.

Could be a gc-marking issue. Will investigate further.

FoamyGuy, 2021-12-13:
I confirmed that #5708 resolves this issue on PyPortal, both the display auto-reload are behaving normally using a build from this branch.

@rsbohn if you have a chance to try on your device to see if the behavior you noticed with auto-reload is fixed using this branch from #5708. If you're still seeing that we can open a separate issue with the description of it.

closed 2021-12-13 by FoamyGuy

rsbohn, 2021-12-13:
It is fixed in 7.1.0 beta.3. Changes to the filesystem trigger auto-reload as expected. Thanks!

PR #5708 (2021-12-13): Clear SAMD TAMPER interrupt in pinalarm properly
- Fixes #5705.

- In SAMD, `alarm_pin_pinalarm_reset()` was clearing the TAMPER interrupt flag with a bitwise operation, instead of just writing a 1 to the right bit, with the rest zeros, using a full register write. The bitwise operation apparently caused the other RTC interrupts to be messed up in some way, and then caused problems with other timing-based operations (probably background task handling), and showed up as `displayio` problems.

- Based on the error above, looked for and fixed a few other `INTENCLR.bit` and `INTENSET.bit` assignme [... cut]