adafruit/circuitpython #5707: synthio.MidiTrack produces single note/beep only

bludin, 2021-12-12:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0-alpha.0-107-g05e073415 on 2021-12-12; Adafruit PyPortal Titano with samd51j20
```

### Code/REPL

```python
import audioio
import board
import synthio
import digitalio

en = digitalio.DigitalInOut(board.SPEAKER_ENABLE)
en.switch_to_output(True)
dac = audioio.AudioOut(board.SPEAKER)
melody = synthio.MidiTrack(b"\0\x90H\0*\x80H\0\6\x90J\0*\x80J\0\6\x90L\0*\x80L\0\6\x90J\0" +
                           b"*\x80J\0\6\x90H\0*\x80H\0\6\x90J\0*\x80J\0\6\x90L\0T\x80L\0" +
                           b"\x0c\x90H\0T\x80H\0\x0c\x90H\0T\x80H\0", tempo=640)
dac.play(melody)
print("playing")
while dac.playing:
  pass
print("stopped")
```

### Behavior

The example from [https://circuitpython.readthedocs.io/en/latest/shared-bindings/synthio/index.html](https://circuitpython.readthedocs.io/en/latest/shared-bindings/synthio/index.html) (with SPEAKER_ENABLE added) plays a single note/beep only and never finishes. 

### Description

Same behaviour in 7.0.0, 7.1.0-beta.1 and 7.1.0-beta.2

rsbohn, 2021-12-12:
I get the same result on the pybadge. One squawk, then silence. Dac is still playing but no sound is produced.

tannewt, 2021-12-13:
Does it work ok in 6.3.0? It'd be helpful to determine what unstable 7.0.0 release broke it.

bludin, 2021-12-14:
synthio didn't exist in 6.3.0

rsbohn, 2021-12-14:
Not in the pybadge 6.3.0 release firmware either. I did find it in the 7.0.0-alpha.2 firmware.
BTW, what sound should it make? Is tempo=640 too high?

dhalbert, 2021-12-14:
Pinging @tyomitch, who contributed `synthio`.

tyomitch, 2021-12-17:
> Pinging @tyomitch, who contributed `synthio`.

On my Meowbit (which has `audiopwmio` but not a DAC-based `audioio`) with 7.1.0-beta.3, the example from the docs (with `board.BUZZ.play(melody)`) works with no problem.

I'd be happy to help, but I don't have a pybadge to test on.

> BTW, what sound should it make? Is tempo=640 too high?

http://atari.ruvds.com/level.mp3 (the start-level sound from QBasic Nibbles)

The `tempo` affects only the interpretation of the MIDI stream; as opposed to `sample_rate` which affects how frequently the [PWM]AudioOut state is changed.

rsbohn, 2021-12-17:
I expected this to play two notes, then quit.
Instead it plays a warbling sound that only ends
when I call `dac.stop()`.

```python
# synthio test
# Raspbery Pi Pico RP2040
# Adafruit CircuitPython 7.1.0-beta.3 on 2021-12-13;
import time
import audiocore
import audiopwmio
import board
import synthio

dac = audiopwmio.PWMAudioOut(board.GP26)

G2 = 43
C3 = 60
midi = bytearray([
    0, 0x90, C3, 0,
    25, 0x80, C3, 0,
    50, 0x90, G2, 0,
    75, 0x80, G2, 0
])
audio = synthio.MidiTrack(midi, tempo=640)

dac.play(audio)
for _ in range(5):
    if not dac.playing: break
    time.sleep(0.25)
if dac.playing:
    print("hard stop.")
    dac.stop()
```

tyomitch, 2021-12-18:
> I expected this to play two notes, then quit.

It does play two notes, then quit, on my Meowbit

Does adjusting `sample_rate` have any effect on your board(s)?

rsbohn, 2021-12-18:
Doubling the tempo does make it play at twice the speed.

rsbohn, 2021-12-18:
With 7.1.0 beta3 I get one note on the pybadge, followed by silence until I call dac.stop(). On the pi pico it loops the one note until I call dac.stop(). 

Maybe someone with a different STM32F401 board could give it a try. All my boards are ATSAMD51 or RP2040 boards.

jepler, 2023-05-03:
If you're still having a problem with this, I recommend trying to play the MidiTrack through an AudioMixer.

The reason is that the MidiTrack produces audio buffers of varying sizes, including very small sizes for short notes. (in older versions it also produced extremely large buffers for long-held notes, but now it has a certain upper limit buffer size)

Some of our audio out implementations handle this small audio buffers poorly and get confused because with a small buffer the next audio interrupt comes sooner, and the next audio data may not be prepared yet. samd and rp2040 fall into this category.

AudioMixer always creates output blocks of a consistent size by fetching audio data from the underlying source multiple times if needed, so it can help work around this problem.

jepler, 2023-05-04:
Oh, MidiTrack _was_ completely broken on samd and rp2040 until 04f4092e110b50fc03dd55e8a5f1c2151c5d0035 -- closing because I think we did fix the underlying cause. Feel free to open a fresh bug report if this persists in 8.1.0 and later.

closed 2023-05-04 by jepler