adafruit/circuitpython #5722: Typo in OnDiskBitmap documentation: shadder

rsbohn, 2021-12-14:
### CircuitPython version

```python
https://circuitpython.readthedocs.io/en/7.0.x/shared-bindings/displayio/index.html
```

### Code/REPL

```python
from displayio import OnDiskBitmap
```

### Behavior

The image’s pixel_shader. The type depends on the underlying bitmap’s structure. The pixel shadder can be modified (e.g., to set the transparent pixel or, for paletted images, to update the palette

### Description

- "shadder"
- Need punctuation at the end of the sentence.

### Additional information

Not present in 6.3.x docs.

tannewt, 2021-12-15:
Would you mind making a PR to fix it? The source is here: https://github.com/adafruit/circuitpython/blob/main/shared-bindings/displayio/OnDiskBitmap.c#L141

rsbohn, 2021-12-15:
https://github.com/adafruit/circuitpython/pull/5731 -- fix this, also one line in BUILDING.md.

dhalbert, 2021-12-16:
Fixed by #5731.

closed 2021-12-16 by dhalbert