adafruit/circuitpython #5733: ESP32-C3 timer period doubling

dhalbert, 2021-12-16:
Track this apparent bug for our own ESP32-C3 port.
- https://github.com/micropython/micropython/issues/8084

tannewt, 2022-08-09:
How would this impact CircuitPython? I don't think this issue is actionable as-is.

dhalbert, 2022-08-10:
I don't know; I just wanted to see whether we might suffer from the same issue. I didn't have time to research it and just wanted to "bookmark" it for further study.

dhalbert, 2022-10-05:
I only see that we use timers in a similar way in `frequencyio.FrequencyIn`. So I'll test whether it's accurate or off by a factor of two.

dhalbert, 2022-10-05:
Pushed to 8.x.x because of #6998.

dhalbert, 2022-11-19:
MicroPython followup here: https://github.com/micropython/micropython/issues/8084#issuecomment-1319939405 and https://github.com/micropython/micropython/pull/10014